ESPresense/ESPresense #791: Device not showing in Mqtt explorer

mmitch2k, 2023-01-21:
### Describe the bug

This device shows up as detected via esp32 terminal but is not showing up in Mqtt Explorer. Any ideas?

### Version

ESPresense Version: v3.1.1

### Device (be specific)

Pixel Watch

### Screenshots

[image]

rouloy, 2023-02-09:
Got the same, but only with a specific ESP32, the one I had laying around works 100%, then bought a new one for another room and this one only publishes every 10-15 minutes even though the beacon sits right next to it.

mmitch2k, 2023-02-09:
yea i have 4 beacons, 2 from the exact same company and one shows the other does not. quite surprising and frustrating that any of this stuff even works half the time

rouloy, 2023-02-10:
After a power cycle it seems to work for 10-15 minutes again.

tenly2000, 2023-03-07:
I had a similar problem today.  For me, it was because I was using the FILTER setting on the ESP32 to only send information about known devices.  But apparently the FILTER comparison is case-sensitive and I had a lowercase J instead of an uppercase one in my entry.  I corrected it and restarted the ESP32 and the device immediately appeared.