ESPresense/ESPresense #795: Implement ESPresense fuctionality in Home Assistant taking advantage of ble_proxy of ESPHome

danps1, 2023-01-27:
### Is your feature request related to a problem? Please describe.

I (and seemingly many others) love the functionality of ESPresense, and also the ble-proxy functionality of ESPHome. Both have different use-cases, and work excellently. However, this leads to a simple issue - having to have 2x ESP32 devices in each room.

### Describe the solution you'd like

I have made a similar feature request in Home Assistant at https://community.home-assistant.io/t/apple-devices-presence-using-esphome-ble-proxy-espresense-alternative/524019 and after considering it some more I think there are some advantages to this project in considering it.

I can see in the ble_proxy logging in ESPHome that all BLE advertisement packets are already being sent to Home Assistant along with RSSI information...
```
[10:06:55][V][bluetooth_proxy:033]: Proxying packet from  - AA:AA:AA:AA:AA:AA (Apple iPhone). RSSI: -69 dB
[10:06:56][V][bluetooth_proxy:033]: Proxying packet from  - AA:AA:AA:AA:AA:AA (Apple iPhone). RSSI: -69 dB
[10:06:57][V][bluetooth_proxy:033]: Proxying packet from  - AA:AA:AA:AA:AA:AA (Apple iPhone). RSSI: -70 dB
[10:06:57][V][bluetooth_proxy:033]: Proxying packet from  - BB:BB:BB:BB:BB:BB (Apple iPad). RSSI: -85 dB
[10:06:57][V][bluetooth_proxy:033]: Proxying packet from  - CC:CC:CC:CC:CC:CC (Apple Watch). RSSI: -75 dB
[10:06:57][V][bluetooth_proxy:033]: Proxying packet from  - AA:AA:AA:AA:AA:AA (Apple iPhone). RSSI: -69 dB
[10:06:58][V][bluetooth_proxy:033]: Proxying packet from  - AA:AA:AA:AA:AA:AA (Apple iPhone). RSSI: -69 dB
[10:06:59][V][bluetooth_proxy:033]: Proxying packet from  - CC:CC:CC:CC:CC:CC (Apple Watch). RSSI: -77 dB
[10:06:59][V][bluetooth_proxy:033]: Proxying packet from  - AA:AA:AA:AA:AA:AA (Apple iPhone). RSSI: -70 dB
[10:06:59][V][bluetooth_proxy:033]: Proxying packet from  - BB:BB:BB:BB:BB:BB (Apple iPad). RSSI: -86 dB
[10:07:00][V][bluetooth_proxy:033]: Proxying packet from  - CC:CC:CC:CC:CC:CC (Apple Watch). RSSI: -75 dB
[10:07:00][V][bluetooth_proxy:033]: Proxying packet from  - AA:AA:AA:AA:AA:AA (Apple iPhone). RSSI: -71 dB
[10:07:00][V][bluetooth_proxy:033]: Proxying packet from  - BB:BB:BB:BB:BB:BB (Apple iPad). RSSI: -84 dB
[10:07:01][V][bluetooth_proxy:033]: Proxying packet from  - BB:BB:BB:BB:BB:BB (Apple iPad). RSSI: -87 dB
[10:07:03][V][bluetooth_proxy:033]: Proxying packet from  - AA:AA:AA:AA:AA:AA (Apple iPhone). RSSI: -71 dB
[10:07:03][V][bluetooth_proxy:033]: Proxying packet from  - BB:BB:BB:BB:BB:BB (Apple iPad). RSSI: -81 dB
[10:07:03][V][bluetooth_proxy:033]: Proxying packet from  - CC:CC:CC:CC:CC:CC (Apple Watch). RSSI: -79 dB
[10:07:04][V][bluetooth_proxy:033]: Proxying packet from  - AA:AA:AA:AA:AA:AA (Apple iPhone). RSSI: -77 dB
[10:07:04][V][bluetooth_proxy:033]: Proxying packet from  - CC:CC:CC:CC:CC:CC (Apple Watch). RSSI: -78 dB
```
Presumably at Home Assistant these advertisements are just ignored, as no component uses them. Obviously the difference with ESPresense is that a whole load of logic, such as calculating the distance, and "decoding" the IRK is performed on the ESP before sending the mqtt packet.

My feature request is essentially to shift this logic from the ESP and onto Home Assistant instead. I think that all the information needed to perform this logic on HA is already being forwarded by the proxy, so no additional coding of ESPHome would be needed.

Other than reducing the number of ESP32 devices needed per room, a couple of other advantages to this approach spring to mind, and there may be others that I haven't thought of. Firstly, it would offload the bulk of the processing from the ESP to HA, where presumably there is more processing power available. Secondly, centralising the processing would allow for more enhanced functionality such as trilateralization, etc, which I can see is already being worked on with the ad-espresense-ips and ESPresense-Companion projects.

### Additional context

I can certainly appreciate that this request is suggesting a fairly radical pivot to how ESPresense works, however I do think that there are also some long-term advantages that could be considered.

csw3190, 2023-01-27:
I think this has been posted before. See: https://github.com/ESPresense/ESPresense/issues/652

danps1, 2023-01-27:
Hi @csw3190 - yes agreed this looks similar, but what I'm proposing is not to implement both proxy and ESPresense on the same device, but rather to fundamentally shift where ESPresense operates from being on the ESP to being on Home Assistant.

martinw72, 2023-02-22:
I like the idea, too! My biggest problem ist that I need a punch of ESP32 devices per room to get all the functionality I would like to have. So a "featured" proxy would be very helpful. But I think in my case it would need additional functionality for the proxy, too. It would be necessary that the BLE Proxy adds device name (room name) with the data, to identify were the info is coming from. Another option would be to add esppresense functionality as a library to ESPHome.

Inrego, 2023-03-02:
Just want to chime in. I really want to use ESPresense. But I'm not going to deploy a bunch of ESP32 just for that. I want to make use of devices already deployed in my home (Shelly devices work as BLE proxy as well for HA).

CumpsD, 2023-06-28:
Would it be possible to run ESPresence logic in a Docker container? We could then always forward the noise from ESPHome to the container, which in turn broadcasts MQTT messages (and then ESPresense is not tied to HA)

codyc1515, 2023-09-27:
Still current.

codyc1515, 2023-10-04:
Functionality has been implemented - https://www.home-assistant.io/integrations/private_ble_device/

Jc2k, 2023-10-04:
(That isn't a direct and full replacement for espresense. It should be fine for answering "is my phone at home" with device trackers, but it's not enough to do room presence).

DTTerastar, 2023-11-05:
Apple Watches (my primary use case) advertise every 500ms.  There is no way that HA will ever proxy every single packet.  You need to do an intelligent rollup of the information especially since on an ESP32 the antenna sharing makes you not get bluetooth while you're broadcasting WiFi.  That makes the proposed approach not as accurate.  And we all know we need to improve the accuracy, not go the other direction...

gunnarbeutner, 2023-11-06:
I've spent the past few days trying to get this to work using two different approaches:

1. Using BleFingerprint as a library with ESPHome: This sort of works but the resulting firmware image is large enough that you'd lose OTA capabilities once you add pretty much _any_ other modules because you have to use the huge_app partition schema. Also, RAM usage is a major problem with BleFingerprint and ESPHome competing for whatever tiny amount of free RAM is left after startup.

2. Using ESPHome to proxy raw BLE advertisements to an MQTT server and then running ESPresense somewhere else: This works fairly decently, but there's a _lot_ of network traffic (around 750 packets/s at 1-2Mbit/s for my 20 nodes) and you lose a significant amount of radio time that could be spent scanning for advertisements.

[image]

Also, running ESPresense "elsewhere" is far from trivial. Linux and Arduino/ESP32 are obviously very different runtime environments. I managed to build a mostly unmodified ESPresense for Linux with a central process that discovers ESPHome nodes and spawns processes for each of them:

```
3069193 pts/3    Ss     0:00  \_ bash
2596803 pts/3    Sl+    0:16      \_ /app/espresense
2596807 pts/3    Sl+    0:25          \_ /app/espresense esproxy-0386ec ec8603
2596808 pts/3    Sl+    0:25          \_ /app/espresense esproxy-041568 681504
2596809 pts/3    Sl+    0:28          \_ /app/espresense esproxy-066d48 486d06
2596810 pts/3    Sl+    0:25          \_ /app/espresense esproxy-033e60 603e03
2596811 pts/3    Sl+    0:26          \_ /app/espresense esproxy-0409dc dc0904
2596812 pts/3    Sl+    0:25          \_ /app/espresense esproxy-02f970 70f902
2596813 pts/3    Sl+    0:22          \_ /app/espresense esproxy-01386c 6c3801
2596814 pts/3    Sl+    0:22          \_ /app/espresense esproxy-01e358 58e301
2596815 pts/3    Sl+    0:21          \_ /app/espresense esproxy-035874 745803
2596816 pts/3    Sl+    0:25          \_ /app/espresense esproxy-033adc dc3a03
2596817 pts/3    Sl+    0:23          \_ /app/espresense esproxy-040e3c 3c0e04
2596818 pts/3    Sl+    0:23          \_ /app/espresense esproxy-074500 004507
2596819 pts/3    Sl+    0:25          \_ /app/espresense esproxy-0363ac ac6303
2596820 pts/3    Sl+    0:23          \_ /app/espresense esproxy-030bc8 c80b03
2596821 pts/3    Sl+    0:25          \_ /app/espresense esproxy-01ef74 74ef01
2596822 pts/3    Sl+    0:27          \_ /app/espresense esproxy-033ec0 c03e03
2596823 pts/3    Sl+    0:25          \_ /app/espresense esproxy-040d84 840d04
2596824 pts/3    Sl+    0:18          \_ /app/espresense esproxy-0382bc bc8203
2596825 pts/3    Sl+    0:26          \_ /app/espresense esproxy-02f7d0 d0f702
2596830 pts/3    Sl+    0:28          \_ /app/espresense esproxy-037d40 407d03
```

One major benefit is that you're able to "update the firmware" much more quickly, e.g. when playing around with filter parameters or trying to add new stuff. But that's about it.

Some of the problems I ran into so far:

* Significant network stability issues due to the huge amount of WiFi traffic. At a minimum you'd need some client-side filtering (at which point you'd need IRK resolving/fingerprint support on the nodes, i.e. you might as well just run ESPresense locally on the nodes).
* ESPHome either supports scanning for BLE advertisements or sending iBeacon frames but doesn't currently do both at the same time (https://github.com/esphome/esphome/blob/dev/esphome/components/esp32_ble_beacon/__init__.py#L9).
* ESPHome's support for managing a fleet of devices with the same firmware image isn't great (https://github.com/esphome/feature-requests/issues/1762).
* Jitter for MQTT messages: ESPresense assumes that advertisements were received at the exact time when `Seen()` is called. MQTT introduces a bunch of inter-frame jitter though. That could be worked around by adding a timestamp in those MQTT messages but I'm close to pulling the plug on this experiment.

DTTerastar, 2023-11-06:
Wow!  That's a crazy amount of work!  

ESP32s are awesome, but they are equivalent to 1 unix process.  It's terrible that you need to run 2 of them if you need to do 2 processes, but at the same time it's kinda cool since you can locate them physically next to whatever you're trying to do.  People need to get over the we should put EVERYTHING into one poor esp32.

If you wanted to take the above experiment to the max you should try doing it POE ethernet nodes.  That would fix the antenna sharing.

marssl, 2025-06-27:
This is absolutely a very good idea, especially for users like me who connect the ESP32 directly to the main network in POE mode for stability.
If ble_proxy and ESPresense can be implemented on a single ESP32, it would be simply perfect.

tabrisnet, 2025-06-28:
a) check out the Bermuda Trilateration integrationagittins/bermuda: Bermuda Bluetooth/BLE Triangulation / Trilateration for HomeAssistantgithub.comb) the ESPresense devs have said it is not possible to cram both functionalities with ESPhome into the limited flash & RAM on an ESP32.c) ESPresense does all of the processing on the ESP32.Bermuda makes HA do it, but to do so it has to send back a lot more data, making WiFi less suitable to the purpose [that is, for BT proxies, Ethernet is advised]. This then reduces the places you can put it.Sent from myPhone.