ESPresense/ESPresense #800: iBeacon UUID not recognized (reported) whenn mac is unchanged

gnome4711, 2023-02-09:
### Describe the bug

I made an app for android smartphones. It changes the uuid of an iBeacon periodically.
 When changing the UUID the advertised mac is unchanged. in this case Espresense does not report the new UUID. But I can scan see the changing by scanning with another smartphone. The old UUID stops advertising and the new takes over. Espresense still reports the old UUID.

### Version

3.1.1

### Active scan enabled

no

### Include filter

iBeacon:

### Exclude filter

a few

gnome4711, 2023-02-22:
@dspen 
yes I tried that workaround   with "Forget beacon if not seen for (in milliseconds):" = 50. that works while my beacons dsends every 100ms. But that causes a lot of traffic with 8 sensors in my house.

dspen, 2023-02-24:
Could you try it with a advertising period greater than 100ms and see if it works still? That would reduce load, and the keys to the linked list should still be generated properly.

gnome4711, 2023-02-25:
@dspen 
No it doesn't. Thats why I went below the advertising intervall of my beacons wich is 100ms.

gnome4711, 2023-02-27:
@dspen 
Sorry, I got your post wrong. 100ms is necessary. When increasing to 250ms, espresense isn't always detecting the beacon and I have to wait in front of the door a few seconds. I guess the beacon isn't recognized for a few scan cycles: i think they "miss" each other for few ms (scanning/advertising).

Reporting a  not longer advertised UUID:
I'm not quite sure if this isn't even a problem for the functionallity. The Sensor at the front door reports the current used UUID by walking towards the door and when it is in range the beacon will be reported. Even if the UUID changes during the approach (the rssi gets better), the infos inside the UUID should still be valid to open the door based on the konfiguration of my system in the background.
But I'd feel better knowing the sensor reports the right UUID.

Insinde the house it doesn't really matter because the beacon needs to be updated within 10 minutes to be recognized as present.

closed 2023-06-12 by stale[bot]