ESPresense/ESPresense #802: MQTT Device tracker

addelovein, 2023-02-16:
### Is your feature request related to a problem? Please describe.

Today we add home and away using mqtt_room component which creates sensors.

Since sensors cant be associated with person object in Homeassistant a solution would be to use mqtt device tracker. 

Today we have to create an automation that first looks at the sensor and then sets the device_tracker. Kind of messy...

### Describe the solution you'd like

For us to do this properly s solution would be to have a device_tracker topic where all devices that are seen within x scope are located.

In that way we could even do a autodiscovery on device_tracker items if nedded.

### Describe alternatives you've considered

I think i got everything above

micdis, 2023-02-18:
isn't the MQTT device_tracker platform, what solves this!? 

https://www.home-assistant.io/integrations/device_tracker.mqtt/

addelovein, 2023-02-19:
Yes and no. 

"topic name format: <discovery_prefix>/device_tracker/[<node_id>/]<object_id>/config."

With this component it does not solve the problem well I can have missed something.  

Because if ESPresense loses a device and it is inactive that's when we want a device_tracker to not home.  And this we would want without an automation for each sensor from mqtt_room. 

I can't see the solution as it's setup today.

micdis, 2023-02-19:
yes you are right - not as easy as I thought it would be! :-)

addelovein, 2023-02-19:
From ESPresense perspective it not a challenge.  All we need is a new topic and place devices in it based on a timeout value.

labaland, 2023-06-07:
> From ESPresense perspective it not a challenge. All we need is a new topic and place devices in it based on a timeout value. Or even better have a state home/not_home.

+1 !!!

l-v-kuznetsov, 2023-06-27:
I second that question. Has anybody found some elegant way to create a device_tracker in HA from ESPresense?

RaoulSargent, 2023-07-26:
I am looking at this as well and would like to link espresense to home assistant “Person”.

ejpenney, 2023-07-26:
> I am looking at this as well and would like to link espresense to home assistant “Person”.

I personally handled this via an automation as described [here](https://github.com/ESPresense/ESPresense/discussions/144):

```
alias: "Fix: Create ESPresense Device Trackers"
description: From https://github.com/ESPresense/ESPresense/discussions/144
trigger:
  - platform: state
    id: update
    entity_id:
      - sensor.emory_s_lg_v60_bt
    from: null
    to: null
  - platform: homeassistant
    event: start
action:
  - data:
      dev_id: "{{ trigger.entity_id.replace(\"sensor.\", \"\") }}"
      location_name: >-
        {{ iif(not is_state(trigger.entity_id, "not_home"),
        "home", "not_home") }}
    service: device_tracker.see
mode: single
max_exceeded: silent
```

This is a redacted version of my automation, you can add ESPresense sensors to the trigger and it'll handle multiple sensors.  You can then add the resulting device_tracker entities to the person.

A bit hacky, but it works.  That said, I wouldn't mind a native device_tracker, even if you specify `gps/gps_accuracy` these device_trackers do NOT fire Zone triggers, which is quite frustrating.

mashedkeyboard, 2023-08-09:
Adding another vote to this - having this with the discoverable topic status would be super helpful!

DTTerastar, 2023-11-03:
Companion attempts to do this.  Please check it out: https://github.com/ESPresense/ESPresense-companion

closed 2023-11-05 by DTTerastar

bzumik1, 2024-04-10:
I still do not see any easy solution how to have ESPresense as device tracker. I am for reopening this issue.