ESPresense/ESPresense #805: Adding settings for Wifi power save mode

Langelus, 2023-02-18:
### Is your feature request related to a problem? Please describe.

Running three ESP32 devboards and all of them seems to go up and down on the network (which is otherwise solid with every other device, Unifi 6 accespoints).

One thing I've noted on my other devices running the same board but ESPHome on them, disabling the wifi power save mode seems to solve these issues. 

### Describe the solution you'd like

If we could get some more control over these settings like powersaving and perhaps "fast connect" mode that would be great.

Thanks for all the hard work with this awesome piece of software!

closed 2023-06-08 by stale[bot]

arcnmx, 2023-06-09:
mean, afaik this is still relevant :(

Langelus, 2023-06-10:
I do agree that although it might not be top priority it is still very much relevant

julianrinaldi, 2023-09-09:
I would like to see this implemented. I think it would solve some issues I'm having.

zwolfinger, 2023-10-20:
Bump for powersave settings

ryanrozich, 2024-08-26:
+1 for this

maxi1134, 2026-03-12:
+! for this. I have 20 Nodes and they drop often, according yo many sources the power saving might be the cause

Tagging @DTTerastar as this is an older issue that was marked as closed by the bot.

Is there any way to get this setting exposed? Or maybe turned off by default?