ESPresense/ESPresense #808: v3.1.1 Crashes on a ESP32-WROOM-32U with external antenna.  How do I best troubleshoot?

Huwee2, 2023-02-22:
### Describe the bug

Hi All,

I have a normal ESP32 running ESPresense in my driveway but recently I bought ESP32-WROOM-32U's from Aliexpress and I have fitted one of these ESP32 with an external antenna - for better coverage when I'm approaching home.

Both are ESP32's running simultaneously in the driveway with the same software v3.1.1 and same configuration (one IRK for my iPhone and filter to only report the iPhone and an ESPHome iBeacon back to MQTT)

The normal ESP32 runs successfully for days / weeks / months without problem.  The ESP32 with the external antenna crashes after 3-10 days.  My driveway gets a **lot** of passing traffic - per week: 40k cars/trucks, 6k cyclists, 4k pedestrians :-D

The last telemetry of the (crashing) ESP32-32U sent was:
```
{
   "ip":"192.168.x.x",
   "uptime":421,
   "firm":"esp32",
   "rssi":-77,
   "ver":"v3.1.1",
   "adverts":10032,
   "seen":8942,
   "reported":18,
   "freeHeap":27264,
   "maxAllocHeap":10484,
   "memFrag":61.5463615,
   "scanHighWater":1040,
   "reportHighWater":1048
}
```

Based on the troubleshooting of bug https://github.com/ESPresense/ESPresense/issues/211, I added some telemetry to HomeAssistant and we can see a crash at 10:42am:
[image]

"ESPresense Driveway" is the working ESP32
"ESPresense Driveway3" is the ESP32-32U that crashes
I've added the configuration to HomeAssistant:
```
mqtt:
  sensor:
  - name: 'ESPresense Driveway Memory'      
    state_topic: espresense/rooms/driveway/telemetry
    value_template: '{{ value_json.maxAllocHeap }}'
    icon: mdi:memory
  - name: 'ESPresense Driveway3 Memory'      
    state_topic: espresense/rooms/driveway3/telemetry
    value_template: '{{ value_json.maxAllocHeap }}'
    icon: mdi:memory
```

So my question is - what's the best way to troubleshoot this while running it remotely?  Any ideas for getting logs remotely? Is there a version I can recompile and push onto the ESP32?

Thanks for your time!

### Version

v3.1.1

### Active scan enabled

No

### Include filter

None

### Exclude filter

iBeacon:xxx iPhone:xxx

### Query filter

None

### Firmware flavor

Standard

### Device (be specific)

ESP32-WROOM-32U (XX0H65)

Huwee2, 2023-02-27:
In lieu of remote debugging, I've set up a second ESP32 3U with external antenna, a long USB cable and connected to my window - hopefully it will experience the same symptoms as the one in my driveway :-)

Then I've cloned the source locally, added `build_type = debug` under `[env:esp32]`, and rebuilt in Visual Code and attached the serial monitor.

Will update once I have some crash logs 👍

Huwee2, 2023-03-01:
My external ESP32 with antenna has crashed this morning, last telemetry:
```
{
  "ip": "192.168.x.x",
  "uptime": 28427,
  "firm": "esp32",
  "rssi": -78,
  "ver": "v3.1.1",
  "adverts": 171880,
  "seen": 167837,
  "reported": 71,
  "freeHeap": 83772,
  "maxAllocHeap": 31732,
  "memFrag": 62.12099508,
  "scanHighWater": 1048,
  "reportHighWater": 1048
}
```

Still waiting for my internal ESP32 with antenna to crash, once this happens I'll provide the crash report.

Thanks!

Huwee2, 2023-03-02:
Nice little crash on the internal ESP32 today (connected via USB), but it successfully rebooted 👍.  Still waiting for a crash with no reboot and then will post results here.

```
07:30:29.017 > 1 New   | MAC: e08553693d4a, ID: e08553693d4a                                              -63dBm 
07:30:31.185 > Disconnected from MQTT; reason 0
07:30:31.250 > 1 New   | MAC: 15c723ad0a2c, ID: exp:20                                                    -101dBm 
07:30:31.564 > 1 New   | MAC: 7c4394565e90, ID: iBeacon:xxxxxx-xxxx-xxxx-xxxx-xxxxxx-x-x          -103dBm 
07:30:32.634 > 1 New   | MAC: 5999376a1dd4, ID: apple:1006:10-7                                           -91dBm 2f1e9acd728e
07:30:34.185 > 0 Reconnect timer
07:30:34.187 > 0 Reconnecting to Network...
07:30:34.197 > Connecting to WiFi SSID 'Patricia's House'...........................................................Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1).
07:30:37.170 >
07:30:37.170 > Core  1 register dump:
07:30:37.172 > PC      : 0x4000d060  PS      : 0x00060035  A0      : 0x80115149  A1      : 0x3ffd1050  
07:30:37.241 > => 0x4000d060: ?? ??:0
07:30:37.241 > A2      : 0x00000000  A3      : 0x00000000  A4      : 0x0000fa00  A5      : 0x7c736c00  
07:30:37.534 > A6      : 0x00000003  A7      : 0x00060023  A8      : 0xfa000000  A9      : 0x00000013
07:30:37.534 > A10     : 0x7c730000  A11     : 0x6c000000  A12     : 0x00000000  A13     : 0x3ffea950
07:30:37.534 > A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x0000000a  EXCCAUSE: 0x00000006
07:30:37.534 > EXCVADDR: 0x00000000  LBEG    : 0x40085925  LEND    : 0x4008592d  LCOUNT  : 0x00000027
07:30:37.534 > => 0x40085925: esp_timer_impl_get_counter_reg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer_impl_lac.c:109
07:30:37.534 > => 0x4008592d: esp_timer_impl_get_counter_reg at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_timer/src/esp_timer_impl_lac.c:111
07:30:37.534 >
07:30:37.534 >
07:30:37.534 > Backtrace:0x4000d05d:0x3ffd10500x40115146:0x3ffd1070 0x400dd473:0x3ffd1090 0x401157f5:0x3ffd10b0
07:30:37.534 > => 0x4000d05d: ?? ??:0
07:30:37.534 > => 0x40115146: AsyncMqttClientInternals::UnsubAckPacket::UnsubAckPacket(AsyncMqttClientInternals::ParsingInformation*, std::function<void (unsigned short)>) at .pio/libdeps/esp32/AsyncMqttClient/src/AsyncMqttClient/Packets/UnsubAckPacket.cpp:5
07:30:37.534 > => 0x400dd473: ArduinoJson6200_F1::CountingDecorator<ArduinoJson6200_F1::DummyWriter>::write(unsigned char const*, unsigned int) at .pio/libdeps/esp32/ArduinoJson/src/ArduinoJson/Serialization/CountingDecorator.hpp:21
07:30:37.534 >       (inlined by) ArduinoJson6200_F1::TextFormatter<ArduinoJson6200_F1::DummyWriter>::writeRaw(char const*, char const*) at .pio/libdeps/esp32/ArduinoJson/src/ArduinoJson/Json/TextFormatter.hpp:155
07:30:37.534 >       (inlined by) ArduinoJson6200_F1::enable_if<ArduinoJson6200_F1::is_unsigned<unsigned short>::value, void>::type ArduinoJson6200_F1::TextFormatter<ArduinoJson6200_F1::DummyWriter>::writeInteger<unsigned short>(unsigned short) at .pio/libdeps/esp32/ArduinoJson/src/ArduinoJson/Json/TextFormatter.hpp:126
07:30:37.534 >       (inlined by) ArduinoJson6200_F1::enable_if<ArduinoJson6200_F1::is_signed<short>::value, void>::type ArduinoJson6200_F1::TextFormatter<ArduinoJson6200_F1::DummyWriter>::writeInteger<short>(short) at .pio/libdeps/esp32/ArduinoJson/src/ArduinoJson/Json/TextFormatter.hpp:110
07:30:37.534 >       (inlined by) void ArduinoJson6200_F1::TextFormatter<ArduinoJson6200_F1::DummyWriter>::writeFloat<double>(double) at .pio/libdeps/esp32/ArduinoJson/src/ArduinoJson/Json/TextFormatter.hpp:96
07:30:37.534 >       (inlined by) ArduinoJson6200_F1::JsonSerializer<ArduinoJson6200_F1::DummyWriter>::visitFloat(double) at .pio/libdeps/esp32/ArduinoJson/src/ArduinoJson/Json/JsonSerializer.hpp:62
07:30:37.534 >       (inlined by) ArduinoJson6200_F1::JsonSerializer<ArduinoJson6200_F1::DummyWriter>::result_type ArduinoJson6200_F1::VariantData::accept<ArduinoJson6200_F1::JsonSerializer<ArduinoJson6200_F1::DummyWriter> >(ArduinoJson6200_F1::JsonSerializer<ArduinoJson6200_F1::DummyWriter>&) const at .pio/libdeps/esp32/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:49
07:30:37.534 >       (inlined by) ArduinoJson6200_F1::JsonSerializer<ArduinoJson6200_F1::DummyWriter>::visitObject(ArduinoJson6200_F1::CollectionData const&) at .pio/libdeps/esp32/ArduinoJson/src/ArduinoJson/Json/JsonSerializer.hpp:48
07:30:37.534 >       (inlined by) ArduinoJson6200_F1::JsonSerializer<ArduinoJson6200_F1::DummyWriter>::result_type ArduinoJson6200_F1::VariantData::accept<ArduinoJson6200_F1::JsonSerializer<ArduinoJson6200_F1::DummyWriter> >(ArduinoJson6200_F1::JsonSerializer<ArduinoJson6200_F1::DummyWriter>&) const at .pio/libdeps/esp32/ArduinoJson/src/ArduinoJson/Variant/VariantData.hpp:55
07:30:37.534 >       (inlined by) ArduinoJson6200_F1::JsonSerializer<ArduinoJson6200_F1::DummyWriter>::result_type ArduinoJson6200_F1::variantAccept<ArduinoJson6200_F1::JsonSerializer<ArduinoJson6200_F1::DummyWriter> >(ArduinoJson6200_F1::VariantData const*, ArduinoJson6200_F1::JsonSerializer<ArduinoJson6200_F1::DummyWriter>&) at .pio/libdeps/esp32/ArduinoJson/src/ArduinoJson/Variant/VariantFunctions.hpp:18
[... cut ...]
07:30:37.893 > => 0x4011f415: BleFingerprintCollection::Count(BleFingerprint*, bool) at lib/BleFingerprint/BleFingerprintCollection.cpp:36 (discriminator 1)
07:30:37.893 > => 0x40097220: prvCheckTasksWaitingTermination at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:4227
07:30:37.893 >       (inlined by) prvIdleTask at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/tasks.c:3930
07:30:37.893 >
07:30:37.893 >
07:30:37.893 >
07:30:37.893 >
07:30:37.893 > ELF file SHA256: 0000000000000000
07:30:37.893 >
07:30:37.893 > Rebooting...
```

closed 2023-06-14 by stale[bot]

Telectroboy, 2023-06-26:
Hello,

I have the same problem and going in the same direction. Memory allocation problem when too much bluetooth devices received. It comes back online alone when the amount of devices goes down! I need to find a solution for this issue. 3.3.0b0

Huwee2, 2023-06-26:
Hi,
My symptoms differ slightly - as it crashes and never comes back.  I have to power cycle (reboot) the ESP32.
When I have some free time I'll try debug again

Telectroboy, 2023-06-26:
Hi,

[image]

Here is the way mines react. And I have 10. They react the same way according to their positions. Some others far away doesn't have the problem.

Telectroboy, 2023-06-27:
As an idea :
I tried to remove print.serial to my fork.
https://github.com/Telectroboy/ESPresense

Didn't had time to compile and try...

Telectroboy, 2023-06-28:
Hello,

I've tried to build a release here https://github.com/Telectroboy/ESPresense/releases/tag/3.3.0b1.1
It's a test.
I don't meet the conditions anymore. I will try to go in a supermarket or concert to try it.
If you have the possibility to try it, please tell me if it works.

The code is based on version 3.3.0b1

Mikecodex, 2024-04-17:
Fast, easy question: When you installed the antenna, did you need to activate it in software, or was it automatic?

Telectroboy, 2024-04-17:
> Fast, easy question: When you installed the antenna, did you need to activate it in software, or was it automatic?

Hello, No you need to change the jumper from internal antenna to external one. No software mod. Not automatic too !

Telectroboy, 2024-04-17:
This one?
[image]

No internal antenna ! So it's mandatory to put one!