ESPresense/ESPresense #809: Which firmware to use for M5Stamp C3 Mate?

TaakoMagnusen, 2023-02-23:
### Describe the bug

The firmware flasher page has a drop down for Standard, Verbose, M5Atom, M5Stick, M5StickC-Plus, and Macchina.

If i ordered an M5Stamp C3 Mate which one of these do i use? Standard? Or does one of the M5 ones apply to the M5Stamp C3 Mate?

oliverboermel, 2023-02-25:
I have the same question i tried every .bin for the side nothing is working ??????

TaakoMagnusen, 2023-02-25:
@oliverboermel I actually flashed mine yesterday, my M5stamp C3 Mate flashed fine with the Standard firmware. But you have to unscrew the protective cover and hold the black button in the center of the PCB (not the button that's accessible when the case is on) while plugging the device into power in order to have it appear in the chrome device list, then you have you have to hold that Button again while clicking install, the let go for erasing.

TaakoMagnusen, 2023-02-25:
One thing to note is that it seems that you need to select Addressable GRB ( not RGB) for the m5stamp c3 mate, even though the documentation says it uses an RGB ordering

closed 2023-02-25 by TaakoMagnusen

oliverboermel, 2023-02-25:
I am sorry from the espresense webside ist not possible to flash anything i tried everything
and i have ot a problme of flashing after flashing ist makes only errors
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x7 (TG0WDT_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40049a42
SPIWP:0xee
mode:QIO, clock div:1
load:0x3c100020,len:0x29578
ets_loader.c 78

oliverboermel, 2023-02-25:
One thing to note is that it seems that you need to select Addressable GRB ( not RGB) for the m5stamp c3 mate, even though the documentation says it uses an RGB ordering
i dont understand that what do you mean how do i do that?

TaakoMagnusen, 2023-02-25:
> One thing to note is that it seems that you need to select Addressable GRB ( not RGB) for the m5stamp c3 mate, even though the documentation says it uses an RGB ordering i dont understand that what do you mean how do i do that?

You do it after you successfully flash, when you're setting up the device.

I don't know how to help you. The webflasher worked for me.

oliverboermel, 2023-02-25:
i tried 3 diffrent computer no one was able to flash from the webside on esphome i can flash without a problem
i even can flash alle the bin's but on espresense side 
Failed to initialize. Try resetting your device or holding the BOOT button while clicking INSTALL.
and why take of the cover you can press the button even with Cover

TaakoMagnusen, 2023-02-25:
The button accessible from the cover is not the right one.... Like I explained earlier you have to take off the cover which reveals another Hutton that the cover normal hides. That's the button you need to use.

oliverboermel, 2023-02-25:
yes the one close to the led i tried it too ut why take of the cover you can press it anywhay with the cover bt i took it of and tried it too

oliverboermel, 2023-02-25:
and from esphome i can flash without pressing a button

TaakoMagnusen, 2023-02-25:
Ah I didn't realize the slatted area in the center was used for the flash button.

In either case idk what to tell you, I was able to flash jus r fine

Hold center button while plugging the c3 mate into usbC, then release, select the device on webflasher, then before you hit install you press and hold the same button, after you click install you can release

It worked for me

oliverboermel, 2023-02-25:
i tried to flash with the esphome flasher esp32c3.bin the process is fine but after flashing the boot error

oliverboermel, 2023-02-25:
i try theat with the button million times always same error tried already 3 diffrent m5stamps

TaakoMagnusen, 2023-02-25:
Idk man, probably open a new issue as I the standards firmware worked for me. Maybe you need different drivers or a different USB cable

oliverboermel, 2023-02-25:
i tried 5 diffrent cable 3 diffrent computer wich binary is the standard for that device

TaakoMagnusen, 2023-02-25:
It's the one that says "Standard" on the webflasher drop-down menu. Idk where to download it since I just used the webflasher

oliverboermel, 2023-02-25:
i did it is not working

TaakoMagnusen, 2023-02-25:
Open a new issuethen?

oliverboermel, 2023-02-25:
I did already Gesendet mit der mobilen Mail AppAm 25.02.23 um 19:05 schrieb TaakoMagnusen
                
            
            
                
                    Von: "TaakoMagnusen" ***@***.***>Datum: 25. Februar 2023An: "ESPresense/ESPresense" ***@***.***>Cc: "Mention" ***@***.***>,"oliverboermel" ***@***.***>Betreff: Re: [ESPresense/ESPresense] Which firmware to use for M5Stamp C3 Mate? (Issue #809)
                
                
Open a new issuethen?

TaakoMagnusen, 2023-02-25:
Good luck!