ESPresense/ESPresense #833: ESPresence & Adafruit QTPY ESP32-C3 - Will not flash on new QTPY

ekramlich, 2023-03-10:
Can you help? 

I am having an issue with 4 new Adafruit QTPY ESP32-C3 that I purchased in Mar 2023. Previously, I purchased and install/flashed ESPresence on 2 Adafruit QTPY ESP32-C3 that I purchased in April 2022 with no issues. The QTPY (2022) work perfectly and I have reflashed them from Espresence to Esphome several times. 

With no clear reason, I have try several times to flash the new QTPY ESP32-C3 (2023) with ESPresense and it has failed. The captive wifi portal will not show up. Ironically, I have flashed the same QTPY (2023) with ESPHome with no isssues. I can not figure out why QTPY (2023) will not accept the flashing but the QTPY (2022) version will. It is the same chipset. 

Any ideas on the issue? It is either ESPresence or the Adafruit QTPY.

DTTerastar, 2023-03-11:
Do you have a log of the console?

ekramlich, 2023-03-11:
Several times, there was a repeated invalid header: 0xffffffff error.

I am read several forums, but can isolated the reason. Any ideas?

DTTerastar, 2023-03-11:
That usually means your flashing a esp32 firmware to another type of esp32 (a c3 or an s3).  Are you using [espresense.com/install](https://espresense.com/firmware)?

ekramlich, 2023-03-12:
Initially, yes - I used espresense.com/install.  After several unsuccessful flashings, I used Esphome and Adafruit ESP web install tool with the same result. I also tried Expressif's command line esptool.  None were able to install and launch the captive portal to configure.   

I am still trying to figure out the issue.

ArtJames, 2023-03-13:
Similar problem with ESP32-Wroom-32 from AZ-Deilvery.
After flashing I can see the device scanning for BT devices, but no way to reach the webinterface.
3.1.1 works fine, so I stay with it.

Wipnose, 2023-03-21:
This afternoon, had the same problems, even with v3.1.1, but with lots of try and error got finally Adafruit QTPY ESP32-C3 working with v3.2.2 !. 
Of course, and you probably have done earlier, we first had to manually install the CP210x driver in windows followed with a computer reset. Then verified that the ESP32 is connected with a proper data-USB cable, then we figured out to press the 'boost' button and reset in order to connect this ESP32 to computer and started flashing.  
Then we both experienced that captive browsers did not start :(  ... 
So, like you, I tried unsuccesfully various combinations: resets, new flashing of ESP, my laptop (wifi) and my android phone, and....got it working somehow with an Iphone: 
Made sure that Iphone's wifi settings were cleared by 'forgeting the [previous] ESPresence wifinetwork', disabled the wifi on the Iphone, then did a clean flash on the troubled ESP, disconnected the ESP from computer, connected the ESP to a seperate USB power socket (wanted to make sure the data or USB connection did not distort the communication with other devices). Then turned on wifi at the Iphone and manually (!) connected wifi to the ESP (and yes, give the ESP-AP a minute to broadcast its ID and also a minute to establish wif connection with the phone) and then surprice surprice the captive browser popped up !!! 
By the way I noticed that the page for the initial configuration is defaulted at 192.168.4.1  . 
The 'not-popping-up' of the captive browser (be it on my other Android labtop etc) has probably something to do with authentication during wifi setup.  Am assuming that other responsive people may have better solutions, but I wanted to let you know that it worked at my place. Good luck.

closed 2023-07-09 by stale[bot]