ESPresense/ESPresense #860: LD2410 BLE

rbjoensen, 2023-03-22:
### Is your feature request related to a problem? Please describe.

Hi and thank you for your great effort in making the best BLE tracker. Is it possible to use the LD2410 BLE as a radar sensor 
Sincerely

DTTerastar, 2023-03-22:
I have some of those ready to go.  I certainly think it'd be cool to do.

Wheemer, 2023-03-22:
I have a couple of those on order. Are they not bluetooth on their own? What would need to be supported in espresense?

rbjoensen, 2023-03-24:
Hi, the radar sensor works on D25, probably others too. The first one i tried didn't work, probably a faulty board

realroywalker, 2023-04-07:
@DTTerastar are you specifically referring to the BLE model of the LD2410? - I have a bunch of the non BLE versions and they work great with espresense via the 'OUT' pin for presence, but currently the only way to configure the board is to flash something like esphome first. It would be great for the board config to be drivable via espresense too!

creuzerm, 2023-05-26:
I have one that I use the motion pin for presence detection and use the phone app to configure over ble. Got a second one coming for the other side of the garage. 

I'd like to think that the extra data coming from the uart lines would be useful.  I can turn on a near light or a far light instead of turning on all the lights. 350 Watts of light vs 250. It would add up over time on the electric bill. 

I am not ready yet to give the uart interface a try.  Only a few months in on home assistant.

aidanbha79, 2023-11-08:
I have set up the LD2410 C with a M5stamp C and got the radar to work but it seems to stop the Bluetooth from working is there anything I can do to get this working with both in HA?

DTTerastar, 2026-11-04:
#2611 adds read-only UART support for the LD2410 / LD2410B / LD2410C (the module's TX to a node RX pin). It publishes presence, moving/still, distances and energies. It doesn't use the LD2410B's Bluetooth link. Testing with a real module would help.