ESPresense/ESPresense #861: Beacon dosent transmit on Oneplus 8 pro

Mindkey75, 2023-03-22:
### Describe the bug

I have a problem setting up Oneplus 8 pro to esperense with HA companion app. When I have installed Espresense on an ESP32 and listen for Beacon on Mqtt explorer or in web terminal I don't get any IBeacon. If, on the other hand, I use the program Beacon Scope to look for my beacon from HA, it finds it without problems.

### Version

Espresense version 3.2.2 HA version 2023.3.5

DTTerastar, 2023-03-22:
HA companion app is buggy, change all the settings a couple times and it'll eventually work.

Mindkey75, 2023-03-22:
No i have setup the sensor now for about 5 days and nothing happen. I can not see the beacon in mqtt explorer or in web terminal. When i try to search for the beacon in beacon scope app i see the beacon id from HA companion app. Its like the espresense not see the beacon from HA companion app. I Can olso see the beacon id in HA Beacon tracker integration. It is the first one. [image0.png]

 Mvh Morten Henriksen

DTTerastar, 2023-03-23:
What txPower are you using?  What rss@1m are you sending? Do you have max distance set to 0 on espresense?

Mindkey75, 2023-03-23:
> What txPower are you using? What ***@***.*** are you sending? Do you have max distance set to 0 on espresense?

Hi Darell

Where can i find this information in companion app or i the Espresense settings? I can see the max distence is set to 16 meter in esperense.

Mindkey75, 2023-03-23:
Hi again

If you mean Tx power i the same as transmit Power is this set to medium. The advertise mode is set to low power, and the measured power at 1 meter is set to -59.
Again the max distance is set to 16 meter in Espresense.

Mindkey75, 2023-03-23:
I forgot to tell that my Espresense work fine all ready with 2 other Android phones and 2 apple phones. The problem is only with this Oneplus 8 PRO. As i allerfattigste told is it possible to see the oneplus beacon i other beacon scanner apps, but mqtt explorer and in web terminal is it not possible to find the beacon from HA companion app. It is like espresense not see the beacon only from my oneplus. I just try to setup HA on a Huawei tablet, and i see the beacon from companion app perfect.

 Mvh Morten Henriksen

Mindkey75, 2023-03-25:
I use -59 ***@***.*** m and tx i set to wery low. I try medium and High olso. Max distance is set to 16 meters. I try change every settings Manu times now, and it still dosent work

 Mvh Morten Henriksen

tbclark3, 2023-04-03:
I am seeing a similar, possibly the same issue, with my Pixel 6 Pro.  Using either the HA beacon OR a beacon generated from BeaconScope, the espsense devices record anywhere from 1 to 3 packets received, then stop recording and stop sending the packets to MQTT.  Beacons from other entities, such as Tiles, are received normally.  I can use an instance of BeanScope on a second device to prove that the Pixel is still transmitting when espsense devices are not recording any activity.

tbclark3, 2023-04-05:
Please look at issue 881.  I think it's the same problem, and I think I figured out the underlying issue.

Mindkey75, 2023-04-05:
thanks. is there a solution or do the developers have to change something before it works

 Mvh Morten Henriksen

Foguet, 2023-04-06:
I have the same problem with a Samsung S23 Ultra; however Samsung Tab S7 works; my Galaxy Watch 4 also works. I use the Home Assistant BLE sensor;
My S23 is streaming because with other apps I can see the stream.
I have already restarted my S23 several times, tried different versions of espresense (up to 3.2.0) and it doesn't work.

closed 2023-07-23 by stale[bot]