ESPresense/ESPresense #864: Device disconnections

zombie241088, 2023-03-23:
Hi, after a period of several months, suddenly one of the devices started disconnecting very frequently. There is no certain time of day that this happens and also the amount of times a day or the frequency between each disconnection varies every day.
I have 4 more devices at home. In some there are no disconnections at all and in others there are occasional disconnections as I mentioned above but less frequently.
How can I find out what the problem is and fix it?

Device information:
ESP32-D0WDQ6
Firmware: v3.2.4
Is there any additional information you would need from me to try to resolve the issue?

JtwoA, 2023-04-07:
make it a threesome. I'm having this across the entire fleet of ten.  I reflashed 3.2.4 on them tonight and they're holding so far.  Will report back on how long it lasts.

JtwoA, 2023-07-09:
Really, no comments/feedback/nothing?  Just #wontfix

closed 2023-10-21 by stale[bot]

markmghali, 2023-12-20:
Also having this issue on 2 of my esp32s. They just disconnect from home assistant after a random amount of time. Super annoying