ESPresense/ESPresense #872: Ability to set a filter to only known IRK's

drschnalli, 2023-03-29:
### Is your feature request related to a problem? Please describe.

im always frustrated looking into the mqtt browser/monitor and it gets flooded with devices i dont care about.

### Describe the solution you'd like

I would like to filter the mqtt output to the known IRK's instead of manually filtering out all other beacons and stuff to get rid of many mqtt traffic. I actually 

### Describe alternatives you've considered

i tried setting the maximum report range to 5 meters to get rid of some reports i dont need but that did not fix my problem

DTTerastar, 2023-03-29:
include irk: should work fine

drschnalli, 2023-03-29:
Can you explain this?

DTTerastar, 2023-05-09:
I use a naming convention for the ids like phone:lindsey and phone:derek and then I can add phone: to include and to count.

ShonP40, 2023-06-07:
Doesn't seem to work with `irk:<id>`

DTTerastar, 2023-09-07:
Can anyone confirm this is an issue for them?

ShonP40, 2023-09-07:
I’ll test it again when I get back home later today

ShonP40, 2023-10-01:
Kinda forgot about this issue, but yea, its still only sending iBeacons (and not IRKs)