ESPresense/ESPresense #875: Also booting loop - on ESP-Wroom-32

tbals, 2023-03-30:
### Describe the bug

The ESP-Board was flashed on a Windows 10 with Edge and ESP Web Tools. The flashing procedure was fine, no problems.
After first boot I set up my WLAN and MQTT Broker and thats it. No more settings.
But WEB UI was laggy and ping not really works, so I just connect to Arduino IDE and watch the serial output. The ESP-Boards boots every 2-5 seconds

### Version

3.2.4

### Logs

```text
ELF file SHA256: 0000000000000000

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Pre-Setup Free Mem: 206580
Setup network
Connecting to WiFi SSID 'foobar'......................................................................172.20.1.163
Firmware:     esp32
Version:      v3.2.4
WiFi BSSID:   C4:01:7C:33:67:C8 (channel=9 rssi=-59)
IP address:   172.20.1.163
DNS address:  172.20.1.53
Hostname:     espresense-home
Room:         home
Mqtt server:  172.20.1.5:1883
Max Distance: 16.00
PIR Sensor:   disabled
Radar Sensor: disabled
I2C Bus 1:    sda=21 scl=22
Query:        
Include:      
Exclude:      
Known Macs:   
Count Ids:    
Post-Setup Free Mem: 139472

1 Motion | Pir: no Radar: no
1 Advert | iBeacon
1 Alias  | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-12-58439 to node:home
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-12-58439 to {"id":"node:home","name":"home"}
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-12-58439 to {"id":"node:home","name":"home"}
1 New    | 6ccc89b1ef5a | apple:1005:9-12                                           -85dBm 0a187c5136
1 New    | 5f4eb6db2bd8 | apple:1006:10-7                                           -71dBm 371a176fc152
1 New    | 2162db96d305 | exp:20                                                    -71dBm 
1 New    | dd10df504816 | dd10df504816                                              -88dBm 
1 New    | 0b37fd6c1c47 | msft:cdp:0902                                             -48dBm 
1 New    | a4c1384a0464 | a4c1384a0464                                              -68dBm 
1 New    | d21510b88c0a | d21510b88c0a                                              -55dBm 
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400859b0: 04928709 1d0020c0 06082df0
Core  1 register dump:
PC      : 0x400859b7  PS      : 0x00060330  A0      : 0xc0050c00  A1      : 0x3ffcfc30  
A2      : 0x00bcd16b  A3      : 0x00000000  A4      : 0x80084317  A5      : 0x3ffcfc00  
A6      : 0x00000001  A7      : 0x3ffce5c0  A8      : 0x00bcd16b  A9      : 0x3ff5f078  
A10     : 0x3ff5f07c  A11     : 0x00000000  A12     : 0x80000020  A13     : 0x00000007  
A14     : 0x00000005  A15     : 0x00000001  SAR     : 0x0000001b  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x40085999  LEND    : 0x400859a1  LCOUNT  : 0x00000000  

Backtrace: 0x400859b4:0x3ffcfc30 0x40050bfd:0x3ffcfc50 0x40116982:0x3ffcfc70 0x40107b5b:0x3ffcfc90 0x40107c2e:0x3ffcfcb0 0x400dfaf1:0x3ffcfcd0 0x400dfb7f:0x3ffcfd00 0x401171a5:0x3ffcfd20
```

thenoid, 2023-05-27:
I think i'm also being hit by this bug.  When connected to my laptop it appears to work fine.  When connected to a wall wart though  it just loops

DanielTerletzkiy, 2023-07-14:
same issue :/
i am using a seeed studio esp32, but after flashing on my windows 11 pc, it only says "device not recognized" every 4 seconds

closed 2023-10-28 by stale[bot]