ESPresense/ESPresense #879: PM1006 and/or LD2410 sensors can be added?

iMike78, 2023-04-02:
### Describe the solution you'd like

Hi lads.

I'm just wondering if there is any chance to add these sensors to the project?

The PM1006 would be great, as this PM2.5 sensor is in the IKEA new dumb air quality sensor device, and while we make it smart with an esp32 D1 mini, we could hide the node nicely inside the VINDRIKTNING case. 2 in 1 solution. :)

OR

The LD2410 sensor is an mmWave presence detector, which could detect anyone even without the phone/BL tag, I think it's way better than a PIR sensor, and still cheap. I found a nice [3D printable case](https://www.thingiverse.com/thing:5631878) for this.

The cherry on the cake would be to add both into a VINDRIKTNING case :)

Sorry if my English is not perfect.

creuzerm, 2023-04-28:
It's a PMS5003 for me and the LD2410 which I want to overlap with espresense. I am looking at having pairs of ESPs scattered across the house and garage at the moment.

Or it's trying to get the presence data collection running in ESPhome.

iMike78, 2023-04-29:
Now I'm using an LD2410B with espresence attached as a radar, so I can get the presence even if its still, and I can configure the sensor trough Bluetooth.

creuzerm, 2023-04-29:
Brilliant Idea on the LD2410B. I just did that. Those tiny headers are a PAIN to solder onto aren't they?
I now have 'radar motion' which looks to be both movement and standing still. This just worked.  Perfection!

There is a lot of capability to this sensor left unused if we communicate via the UART interface.
There are several ESPHome implementations for this such as https://github.com/rain931215/ESPHome-LD2410 which may jump-start bringing the capability here.

creuzerm, 2023-05-27:
@iMike78 I fleshed out the 'sensors' section of the documentation as it was empty and included your ideas on how to use the LD2410b. https://espresense.com/sensors Thank you for your help to me for setting it up. I got a LD2410c just today for the opposite corner of my garage. Such a NICE little device.

As for the PM1006, I doubt we will be able to convince the maintainer to support that as it's a hard argument that this is a presence detector. Yes, kicking up dust mucking about in a basement or garage, but too many false positives. We may need to live with ESPHome for this device.

rojoricardo, 2023-06-09:
> Now I'm using an LD2410B with espresence attached as a radar, so I can get the presence even if its still, and I can configure the sensor trough Bluetooth.

Which GPIO did you use? i connected to GPIO13 and selected 13 in the ESPresence interface, but it always says detected, no change whatsover

creuzerm, 2023-06-09:
@rojoricardo  That will depend on your specific ESP32 dev board.  

Try setting the motion pin type to "Pulldown" as this will allow everything to 'default' to off and when the sensor triggers and goes hi, the ESP can see that change.  Without a Pulldown, the pin isn't really on, but could have some amount of voltage on it and thus appear to be on.

[image]

closed 2023-09-21 by stale[bot]

valentinmirchev, 2024-10-25:
I am having issues on detecting motion even after I have set the radar values exactly as explained above. I use the 3.3V and ground pins, plus the OUT pin of the LD2410b to connect to pin4 on esp32. In the esp32 config I set the pin correctly and tried both Pullup and pulldown but no luck with either of them.

creuzerm, 2024-10-25:
> I am having issues on detecting motion even after I have set the radar values exactly as explained above. I use the 3.3V and ground pins, plus the OUT pin of the LD2410b to connect to pin4 on esp32. [...]

You look to be using the Bluetooth version. Have you confirmed via a Bluetooth app that it's working and detecting and such? 

The next step is to use a tool such as MQTT-Explorer (or a device debug log) to see that the detection is getting emitted to MQTT correctly.

Thirdly is the Home Assistant configuration and such.