ESPresense/ESPresense #881: ESPresense not seeing beacons from Pixel 6 Pro

tbclark3, 2023-04-04:
### Describe the bug

I have esp32 devices in each room, all reporting to mqtt, and the setup works beautifully with Tile transmitters. I am also trying to detect my Android phones but on MQTT Explorer they consistently show 1 or 2 packets received, and then nothing more.  Stopping and starting the Pixel 6 transmitter usually causes the receipt of 1 additional packet, but not always.  My Pixel 6 Pro is running Android 13 with the March 5 2023 update.

I normally use the Home Assistant BLE sensor as the transmitter on the Pixel, but as a test, I installed BeaconScope on the Pixel 6 and on my Chromebook.  With the esp32 adjacent to the Chromebook,  the Chromebook consistently shows packets received as expected, while MQTT shows nothing received from the Pixel although it continues to show packets normally received from 4 Tile devices and others.  I used the monitor, but it is hard to tell when packets are received because it only shows devices that appear and leave.  However, it does show that the Pixel is detected, and 2 minutes later, it leaves, never to return.

There is a bug report in a different thread (#861) from the owner of a OnePlus 8 Pro who may be describing the same issue.

### Version

3.2.4

### Device (be specific)

‎ESP32 D1 mini

### Logs

```text
No relevant logs
```

radkokeves, 2023-04-04:
as pixel 6 user, i have to say, in my case, i have to restart phone regularly for beacons to work properly... a lot of things can be solved by restarting pixel...

tbclark3, 2023-04-05:
I've figured it out, and it is definitely an espresense bug.  It is Android-related, but not Pixel per se.  Espresense correctly logs (and sends to MQTT) the first Android iBeacon, and any subsequent beacons **that use the same MAC address.**  However, if the MAC address changes, no subsequent iBeacons are logged for that iBeacon.  That explains why "1 or 2 packets" might initially be received.  Sometimes Android sends the first 2 packets before it changes the MAC address.

Foguet, 2023-04-06:
I have the same problem with a Samsung S23 Ultra; however Samsung Tab S7 works; my Galaxy Watch 4 also works. I use the Home Assistant BLE sensor;
My S23 is streaming because with other apps I can see the stream.
I have already restarted my S23 several times, tried different versions of espresense (up to 3.2.0) and it doesn't work.

pimpinpanther96, 2023-04-17:
I had the same issue on my pixel 6 for months. Change the "measured power" setting in the HA companion app to -79. I randomly tried it one day after I thought I had tried everything and it's been working perfectly ever since.

Wheemer, 2023-07-01:
> I had the same issue on my pixel 6 for months. Change the "measured power" setting in the HA companion app to -79. I randomly tried it one day after I thought I had tried everything and it's been working perfectly ever since.

I tried this and it did not help. I have an automation in my garage for the lights based on presense. Every couple of days I am left in the dark suddenly as my pixel 4 stops being detected. Only a reboot and unlock solves for another couple days.

closed 2023-10-18 by stale[bot]