ESPresense/ESPresense #888: GL-iNet S10 constantly drops POE connection

uspino2, 2023-04-19:
### Describe the bug

Installed ESPresense in GL-iNet S10 (v2.1), connected to the AP portal and selected GL-iNET 2.1 POE power in drop-down menu. Reboot. Connected to POE, it gets an IP address and seems to start up, but drops connection every 2 seconds. Might be related to [this](https://github.com/gl-inet/gl-s10-idf-patch)?

### Version

3.2.4

### Firmware flavor

Standard

### Device (be specific)

GL-iNet S10 (v2.1)

DTTerastar, 2023-04-19:
That link is the idf way to solve.  We need to find the Arduino way of solving it.

Matsuo3rd, 2023-05-14:
Got this fixed on my setup.
See https://github.com/ESPresense/ESPresense/pull/905

uspino2, 2023-05-14:
> Got this fixed on my setup.
> See #905

Great! Hoping a new release with the fix will come out soon...

DTTerastar, 2023-05-25:
Released a beta now, please check!

uspino2, 2023-05-26:
> Released a beta now, please check!

Thank you! I'm out of town for a week, but I'll check and report back.

uspino2, 2023-07-03:
Just wanted to report that GL-iNet S10 has been working perfectly fine on v3.3.0b1. No disconnections whatsoever. Good range, runs cool, and POE turns it into a great little device.

closed 2023-07-03 by DTTerastar