ESPresense/ESPresense #900: W1901400 switchbot can't save to flash

mogorman, 2023-05-08:
### Describe the bug

I flashed my switchbot with tasmota using instructions found here https://github.com/kendallgoto/switchbota
I then flash espresense over tasmota using its web interface which succeeded.  I used the esp32c3 binary.  Now whenever I try to save I get this error.

"Error while writing to flash filesystem."

### Version

3.2.4

### Device (be specific)

W1901400 switchbot relay

niceboygithub, 2023-05-09:
Add  the setting to platformio.ini then generate firmware, it may fix this issue
```
board_build.flash_mode = dio
```

rpiersall, 2023-05-25:
I'm trying to do the same (espresense on W1901400) and ran afoul of the exact same issue

I modified the esp32c3 config within platformio.ini file as follows based on @niceboygithub's suggestion

```
  [esp32c3]
  extends = common
  board = esp32-c3-devkitm-1
+ board_build.flash_mode = dio
  build_flags =
    -D ARDUINO_ARCH_ESP32
    -D ARDUINO_ARCH_ESP32C3
    -D CONFIG_IDF_TARGET_ESP32C3
    -D CO
    -D CONFIG_BT_NIMBLE_PINNED_TO_CORE=0
    -D REPORT_PINNED_TO_CORE=0
    -D ESP32C3
    ${common.build_flags}
```

Reran the build with `pio run -e esp32c3`

And now, having loaded that onto one of my _other_ switchbot minis, it seems to be able to write to flash (including WiFi configuration).

It's no help for the unit I already flashed, might have to crack open the sonic welds to reprogram that one, but at least now I can proceed with the others!

closed 2023-09-06 by stale[bot]

nwithan8, 2024-05-20:
@DTTerastar is there any rea

> I'm trying to do the same (espresense on W1901400) and ran afoul of the exact same issue
>
> I modified the esp32c3 config within platformio.ini file as follows based on @niceboygithub's suggestion [...]

@DTTerastar is there any reason why this can't be included permanently in the source code? Avoided bricking (most of) my devices during the initial setup process, only to brick them when I ran "Update with GitHub Release" from the ESPresence add-on in Home Assistant, which flashed the broken firmware.

chrismcneil, 2024-05-24:
Bah just bricked one of my plugs. :( Now I know to read closed issues......