ESPresense/ESPresense #904: Enable one or more GPIOs as switches

WheresWes, 2023-05-10:
### Is your feature request related to a problem? Please describe.

As I understand it, ESPresence is great for detecting a large variety of inputs, but rather limited in its ability to output states beyond wireless means.  This of course requires several steps in the process to be working reliably - solid wireless connection, good network, stable mqtt broker and HA instance etc.   Depending on the use application this may not always be guaranteed - for example the ESPresence base station for a gate opener sitting at the limits of its wireless range.

It would be desirable to enable one or more gpios on the base station as switches which could be set to turn on when a particular condition is met - for example a filtered ble beacon is detected.  

That would enable direct connection of the ESPresence to whatever control module was being used without being reliant on so many intermediary steps

### Additional context

I use an esp32 with esphome BLE presence detection to detect the presence of ble beacons in our cars and then command the electronic gate to open automatically.  But the esphome BLE presence performs poorly when detecting more than a couple of beacons.  Other than this the module works well and is independant of the wireless connection to HA.  I'd like to transfer the BLE duties over the ESPresence on a separate module, but want a physical connection between the two for reliability.

DTTerastar, 2023-05-26:
Test this out in the beta I just released.

WheresWes, 2023-05-26:
I've had a quick play..  The switches added under the gpio section. I have the impression that these function as inputs to espresence correct? I'm looking for a switch output from espresence.. Possibly this can be done with the LED outputs options but I'm don't fully understand the operation of the different LED modes.  Does the 'motion' mode apply only to use with pir sensors or would it also include ble beacons?

WheresWes, 2023-05-26:
Could one of the LED modes be used to indicate when a filtered beacon is detected?

creuzerm, 2023-05-27:
> I've had a quick play.. The switches added under the gpio section. I have the impression that these function as inputs to espresence correct? I'm looking for a switch output from espresence.. [...]

The buttons/switches are inputs into ESPresense. I added both as we could see having a light switch and a door open switch and also a press button, capacitive touch switch, light beam break etc. While they are all the same thing to the ESP32 hardware, they do present slightly differently in Home Assistant.

"Motion" is a GPIO pin getting set by a device which is interpreted as motion. A PIR, a RADAR, etc, will set a pin Hi to indicate motion is present, this gets passed along to home assistant. So not the BLE beacons moving - that gets calculated in ESPresense Companion.

I agree that having an OUT (or two) controlled by setting an MQTT topic would be really convenient and will reduce the need of having 2 ESP32s sitting next to each other. Being able to turn that device on/off with the ESPresense unit would be a great cost savings. I think it's a natural fit to have some basic level of action hubbed around the presence detection. 

I have a air purifier sitting next to me that never moves. It would be great to have that be an ESPresense node instead of an ESPHome node. A single button of concern to turn it on and off and cycle the modes.

WheresWes, 2023-05-27:
Thanks Mike.. That is what I figured.  Could you also describe the 'status' LED.   Literature states it is on when detecting ble advertisements.. Does that mean all of them or only filtered beacons?

Darnshelm, 2023-08-21:
Hi are that possible to get 4 button or option to choose how many?

DTTerastar, 2023-08-22:
> Thanks Mike.. That is what I figured. Could you also describe the 'status' LED. Literature states it is on when detecting ble advertisements.. Does that mean all of them or only filtered beacons?

The ones in the count option

nybdog, 2023-11-27:
I too would love to have a GPIO output option who's state changed with based on a configurable beacon and distance threshold.