ESPresense/ESPresense #916: schrodinger's ESP-WROOM-32  - Wifi Connection Failure

thenoid, 2023-05-27:
### Describe the bug

So here is the the weirdness.   I bought a bunch of these [ESP-WROOM-32](https://www.amazon.com/gp/product/B0B19DXFHX/ref=ppx_od_dt_b_asin_title_s00?ie=UTF8&psc=1) boards.    They were working flawlessly for the longest time, but auto-update firmware was turned on, and at some point they stopped

When connected to my laptop, with the serial connection open either viea `screen` or the `web terminal` they boot up and work fine.  

When plugged into a wall wart (tried dozens from years of nerd hoarding) it just boot loops connecting to the wifi.
When plugged into my laptop, without the serial connection open, it also boot loops trying to connect to the wifi.

If I open a serial connection during the middle of the boot loop, it will:
* Fail the current attempt
* Start working on the next attempt.

It's like the act of observing it is changing the outcome.  Tried numerous verions with no luck

### Version

3.2.4, 3.1.1, 3.3.0.b0

### Device (be specific)

standard

### Logs

```text
...................................................................................................................................................... failed (status=6).
                                                                                                                                                                          Starting access point for configurat
ion portal.
SSID: 'espresense-papas-office'
                               IP: 192.168.4.1
                                              ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Setup network
Connecting to WiFi SSID 'slack-iot'........................................................................192.168.3.57
Firmware:     esp32
Version:      v3.2.0b12
WiFi BSSID:   6A:22:32:1D:64:7E (channel=6 rssi=-52)
                                                    IP address:   192.168.3.57
DNS address:  192.168.0.31
Hostname:     espresense-papas-office
Room:         Papas Office
Mqtt server:  192.168.0.25:1883
                                 Max Distance: 16.00
                                                    Init Free Mem:139576
                                                                        PIR Sensor:   disabled
Radar Sensor: disabled
I2C Bus 1:    sda=21 scl=22
Query:        
Include:      
Exclude:      
Known Macs:   
Count Ids:    
```
```

thenoid, 2023-05-28:
Seems like this: https://github.com/ESPresense/ESPresense/issues/628 might also be in play.  When connected to my laptop + serial console open, it doesn't care which channel it's on.

When connected to a power plug, if I set all my APs to channel 11 it connects up fine.  Any other channel, not so much.

zhp46, 2023-08-16:
I was having the same issues. With the serial monitor connected, I was able to get a wifi connection from a couple rooms away, but once I switched to a 5v power brick, the wifi connection would fail. However I was able to get a connection if I moved it closer to the AP. 

It turns out there is an issue with the CH340C USB serial chip, simply desoldering its vcc pin (pin 16, top right of chip) brings the wifi performance back to normal. 

[image]

Note that with this hack you will not be able to flash over usb without resoldering this pin.

closed 2023-11-05 by DTTerastar

Aziona, 2025-01-12:
Purchased five ESP32's recently with a 2102 serial converter (instead of 340). All five of them work outed of the box, no problems anymore. Maybe there was large batch with a faulty ch340 desgin on the market.