ESPresense/ESPresense #931: Beacon isn't showing "not_home"

LuxStatera, 2023-06-15:
I use ESPresense with Home Assistant to track the room location of my cat. If my cat leaves the house through the kitchen door then the tracker card still shows my cat is in the kitchen. Normally the card updates every second but if my cat were to exit the house for 10 mins then the tracker card will say "updated 10 mins ago". So the card will say "kitchen" and "updated 10 mins ago" instead of "not_home". How do I adjust the settings so I can see when my cat is no longer near any of the zones? I'm trying to setup an automation so I can get a notification if my cat gets out of the house

scaytrase, 2023-06-26:
Are you sure your cat isn't still within max configured distance of range?

LuxStatera, 2023-06-26:
I'm positive

Kummallinen, 2023-10-16:
I'm having the same issue with an iPhone. Other devices show not_home correctly but that iPhone just shows its last seen room even when hundreds of miles away.

The phone in question was paired with presence to set up ikr.

The node in question doesn't show the phone on the list of visible devices.

DTTerastar, 2023-11-05:
Is this for companion, or just using espresense w/ mqtt_room? mqtt_room will set you as not_home after the away timeout.  If it's for espresense companion I think this is a dup of this one: https://github.com/ESPresense/ESPresense-companion/issues/336

Kummallinen, 2023-11-25:
In my case it turned out the issue was a copy + paste error in the config for that device. I think I removed the away timeout property for that device by accident when adding additional ones.

closed 2023-11-30 by DTTerastar