ESPresense/ESPresense #932: ESPresence looses configuration (randomly?)

poweredgenl, 2023-06-15:
### Describe the bug

With the danger the issue cannot beeing solved. im noticing some werd behaviour with espresence.

Im running a few smalll wifi connected boards / regular esp32 dev boards - and they vary a bit in firmware...either versioen 3.1 or 3.2 ... 

My issue is this: at random times / but as far as i can see - it happens on host reboots of homeassistant (which esprecne is integrated in)... it looses 2 important config variables:

- the room name
- the IP of the mqtt server

Its hard for me to really simulate it - but this happens quite a few times rendering esprence unusable.

Workarounds: when i log in back to the ip - i can add the roomname and IP of mqtt - and it reconnects properly. The weird part is - the username/password IS retained.

Did anyone come across this issue? 

### Version

various

### Active scan enabled

yes

### Include filter

no

### Exclude filter

no

### Query filter

no

### Firmware flavor

3.X

### Device (be specific)

esp32/8266

goedh452, 2023-06-24:
I have exactly the same thing. For mee it seems to be related to a crash of home assistant. I run HA on an Intel NUC in docker and sometimes it becomes unresponsive, mostly related to a custom plugin. When rebooting the PC the config of ESPresense is wiped. I have 7 ESP32-dev boards.

gunnarbeutner, 2023-06-29:
Something similar happened here when I tried to reboot my ESPresense nodes: Instead of sending a request to `http://NODE/restart` I accidentally sent a POST request to `http://NODE/` (i.e. via `curl -X POST http://NODE/`) which wiped all configuration variables except for the MQTT user and password.

Maybe something is sending HTTP requests to your nodes and ESPresense _somehow_ ends up interpreting the request for the wrong URL path?

Perhaps a hidden required field should be added to the configuration form to make sure we don't accidentally overwrite the config.

closed 2023-10-11 by stale[bot]

gunnarbeutner, 2023-10-12:
This is still an issue.

axelct, 2024-02-19:
I can confirm the issue is still there with 3.3.2. Other device with 3.1.1 hasn't the issue.

Benniepie, 2024-05-23:
I have just discovered this, ESPresense Version: v3.3.2b3 - around 20 devices (all of them). 
The mqtt host and room name was missing. On my devices the mqtt host changed to "mqtt.z13.org" which I assume is some sort of default (not entered by me). I happened to be setting up NextDNS and noticed a ridiciulous amount of DNS A lookups for mqtt.z13.org (around 22,000 in a couple of hours) and traced this back to Espresence. Adding the correct mqtt host back into every device and then save/reboot has stopped it, although I don't know for how long.

axelct, 2024-05-23:
Waiting for a new release that fix the problem I solved settiing automation on homeassistant that execute, at a specified time, the following REST (POST method):
http://<espresence_node_ip>/?room=Salone&mqtt_host=<mqtt_host_ip>&pub_tele=1&discovery=1&pub_rooms_dep=1&pub_devices=1

With the above url is possible to reconfigure the missing parameters. Hope this help.

poweredgenl, 2024-05-23:
thats an interesting solution - at least one to mitigate possible issues ... you could enhance this with a sensor trigger - im monitoring the connection state of the ESPresnce devices and you could say "if == disconnect" "push [your script/url]"

axelct, 2024-05-23:
> thats an interesting solution - at least one to mitigate possible issues ... you could enhance this with a sensor trigger - im monitoring the connection state of the ESPresnce devices and you could say "if == disconnect" "push [your script/url]"

I had thought about connecting a sensor but the problem (in my case) is that the node is connected (ping response is ok) but the mqtt configuration is lost

poweredgenl, 2024-05-23:
what happens at my install is the binary sensor goes to 'diconnected' while ping is still OK - so im going to use that trigger for executing the curl script.