ESPresense/ESPresense #940: ESPresense only connects to network when connected to ethernet-connected PC

lostkoz4, 2023-06-22:
### Describe the bug

My ESP32 is not connecting to wifi during setup, timing out 100% of the time. The sensor is only available to home assistant if the device is plugged in to my desktop PC, which is connected to ethernet. When not plugged in to the PC the device will continually reset and launch its captive network, which I can sometimes connect to. When I do, the wifi settings are correct. I have other ESP32s connected to this same network running ESPHome that connect consistently and easily

Network credentials are correct. Network is 2.4 GHz, using WPA2-PSK [AES] security settings.

Logs (available when connected to PC) do not indicate any issues, as the issue is not occurring when connected to the PC

### Version

3.2.4

### Device (be specific)

https://www.amazon.com/dp/B0B19KRPRC?psc=1&ref=ppx_yo2ov_dt_b_product_details

closed 2023-10-05 by stale[bot]

KevinBalkoski, 2023-11-30:
Did you ever figure this out? I am having the same issue.

lostkoz4, 2023-11-30:
I did, the issue was the esp boards I had. They were USB-C powered and weren’t getting enough power to fully run their WiFi connection. I switched to microUSB powered boards and everything immediately worked flawlessly