ESPresense/ESPresense #944: Bug when too much Bluetooth devices in the area

Telectroboy, 2023-06-24:
### Describe the bug

Hello, 
I use ESPresense on ESP32 Cam.
They completely fail when there are too much Bluetooth devices in the area. They are still connected to wifi.
But don't answer to ping anymore or just sometimes but very random.
Impossible to access to webpage or UI.
And no datas on MQTT anymore too.

The issue disappears when devices get away.
I use filters to receive only my itag:

We speak about hundreds of devices in the same area.
And I'm with 3.3.0b0

### Version

3.3.0b0

### Include filter

itag:

### Exclude filter

nono

### Device (be specific)

ESP32 CAM

### Logs

```text
Écouter
espresense/rooms/fagtc2/#
QoS
0
Message 60 reçu sur espresense/rooms/fagtc2/switch à 12:13 :
OFF
QoS: 0 - Retain: true
Message 59 reçu sur espresense/rooms/fagtc2/led_1 à 12:13 :
{"state":"OFF"}
QoS: 0 - Retain: true
Message 58 reçu sur espresense/rooms/fagtc2/button_2_timeout à 12:13 :
0.50
QoS: 0 - Retain: true
Message 57 reçu sur espresense/rooms/fagtc2/button_1_timeout à 12:13 :
0.50
QoS: 0 - Retain: true
Message 56 reçu sur espresense/rooms/fagtc2/switch_2_timeout à 12:13 :
0.50
QoS: 0 - Retain: true
Message 55 reçu sur espresense/rooms/fagtc2/switch_1_timeout à 12:13 :
0.50
QoS: 0 - Retain: true
Message 54 reçu sur espresense/rooms/fagtc2/radar_timeout à 12:13 :
0.50
QoS: 0 - Retain: true
Message 53 reçu sur espresense/rooms/fagtc2/pir_timeout à 12:13 :
0.50
QoS: 0 - Retain: true
Message 52 reçu sur espresense/rooms/fagtc2/prerelease à 12:13 :
OFF
QoS: 0 - Retain: true
Message 51 reçu sur espresense/rooms/fagtc2/auto_update à 12:13 :
OFF
QoS: 0 - Retain: true
Message 50 reçu sur espresense/rooms/fagtc2/arduino_ota à 12:13 :
OFF
QoS: 0 - Retain: true
Message 49 reçu sur espresense/rooms/fagtc2/rx_adj_rssi à 12:13 :
0
QoS: 0 - Retain: true
Message 48 reçu sur espresense/rooms/fagtc2/tx_ref_rssi à 12:13 :
-59
QoS: 0 - Retain: true
Message 47 reçu sur espresense/rooms/fagtc2/absorption à 12:13 :
3.50
QoS: 0 - Retain: true
Message 46 reçu sur espresense/rooms/fagtc2/max_distance à 12:13 :
50.00
QoS: 0 - Retain: true
Message 45 reçu sur espresense/rooms/fagtc2/status à 12:13 :
offline
QoS: 0 - Retain: true
Message 44 reçu sur espresense/rooms/fagtc2/button à 12:13 :
OFF
QoS: 0 - Retain: true
```

Telectroboy, 2023-06-26:
I have a view of the free memory.
[image]

Everytime the memory is going low the ESPresense disconnect and disappear. It comes back alone by night.
An idea?

Telectroboy, 2023-06-26:
QoS: 0
Payload: ip: 192.168.1.92
uptime: 47359
firm: esp32
rssi: -57
ver: v3.3.0b0
adverts: 1902090
seen: 353831
reported: 89199
teleFails: 71538
freeHeap: 80864
maxAllocHeap: 31732
memFrag: 60.75880491
scanHighWater: 896

Telectroboy, 2023-06-27:
As an idea :
I tried to remove print.serial to my fork.
https://github.com/Telectroboy/ESPresense

Didn't had time to compile and try...

Telectroboy, 2023-06-28:
And create a build https://github.com/Telectroboy/ESPresense/releases/tag/3.3.0b1.1

OttixSw, 2023-07-16:
You might be running out of memory if the fingerprint linked list gets too long. Try lowering the forget_ms configuration value from 300000 ms (5 minutes) to 60000 (1 minutes) and see what happens. As most BLE devices transmit at least once every 15 seconds this should not cause any problems with prematurely dropped devices. YMMV.
If that works then submit a request to see about lowering the forget_ms value temporarily when the free heap space gets too low. There ought to be some feedback here to automatically free memory as needed.

Alternatively, whitelist only the BLE devices that you care about. That should hopefully prevent them from ending up on the fingerprint linked list in the first place.

Telectroboy, 2023-07-17:
Hello !

Thanks a lot for your help!
I will try that and tell :)

Telectroboy, 2023-07-29:
So It doesn't work. I've tried a lot of things but that's not the solution... Another Idea ?

OttixSw, 2023-07-30:
Something is using up all the memory. The real solution is to figure out what the problem is and either fix it or put in some sort of mitigation when the condition occurs.
You might just restart the ESPresense node when the free memory drops below 48k. It would be interesting to see if the free memory continues dropping after a restart. Alternatively just restart it at 11:00 proactively.
Does the camera continue to work through the MQTT dropout? I've no idea how the camera uses the heap. One possibility is that the heap is being fragmented so heavily that you are losing too much free memory to fragmentation. If you are comfortable coding you could instrument the code looking for memory allocation failures; I've no idea what C++ does on the ESP32 but it should be possible to instrument the underlying malloc() calls and even follow the free chains to report on fragmentation. The question is: how much effort are you willing to put into it?

Telectroboy, 2023-08-01:
Hello,

> You might just restart the ESPresense node when the free memory drops below 48k. It would be interesting to see if the free memory continues dropping after a restart. Alternatively just restart it at 11:00 proactively.

I did it. But Yes the issue comes back in 5 seconds after the reboot of the ESP32.
It only disappear when less Bluetooth signals are in the area (around 300)

I'm not comfortable enough with the code to go so far.
I can give a lot of effort but I miss some knowledge for the moment ! Still looking for a solution.

Version 1.7.2 is OK but lacks a lot of addons and security issues !!!

Telectroboy, 2023-08-09:
Hello,

Does someone have an idea. How to filter only iTag prefix devices?
I'm trying to understand BleFingerprint.cpp file. But... it's quite difficult for me :/

DTTerastar, 2023-11-05:
Unfortunately this is a design issue.  The best fix would be to find more memory.  Have you tried a WROVER module?  They have external PSRAM that should make it handle more devices.  C3 and S3 also have a bit more memory free by default so that would be another option (but not as good as WROVER)

closed 2023-11-05 by DTTerastar