ESPresense/ESPresense #955: ESP 32 ESP-WROOM-32 boards configured but will not show in MQTT

Menz01, 2023-07-04:
### Describe the bug

i purchased 5 of these devices. i have flashed them and put in my Wifi and MQTT server info and saved and rebooted them. no matter what i do they will not show in Home Assistant, i have checked MQTT and none of them will show, the correct information is in them. if i load up MQTT Explorer i can see topics related to them. i can browse to their IP addresses and see all the information so they are connected to my network. i can see them in my router as well. they just dont show in HA

johnnykudahl, 2023-07-19:
Is MQTT discovery enabled and is the discovery topic on the ESP32 set to "homeassistant" ?
https://www.home-assistant.io/integrations/mqtt/#discovery-options

Menz01, 2023-07-19:
I had something misconfigured. Thank you

Menz01, 2023-07-19:
Fixed

closed 2023-07-19 by Menz01