ESPresense/ESPresense #971: LuatOS ESP32C3 doesn't work

RedstoneGithub, 2023-07-13:
### Describe the bug

Like issue #776, I cannot get esp32c3 LuatOS to work. It installs but there is no log output or wifi AP available. I tried flashing through various means and same thing. I believe it needs to flash using DIO mode, which works for esphome.

### Version

3.2.4 and 3.3.0b2

### Active scan enabled

N/A

### Include filter

N/A

### Exclude filter

N/A

### Query filter

N/A

### Firmware flavor

Standard

### Device (be specific)

ESP32C3 LuatOS USB-C 

### Logs

```text
No output at all
```

DTTerastar, 2023-11-05:
It's not easy for us to use dio for just one c3.  I think it would be better to just document it as unsupported.  You can manually flash it just fine and set the board type, it's just a prebaked image won't work.

closed 2023-11-05 by DTTerastar

cmgenius, 2024-04-10:
for what it's worth I added `board_build.flash_mode = dio` to my platformio.ini and had success