ESPresense/ESPresense #975: Feature Request: Retrieve IRK from Apple Watch on Windows in Espresense

Aviv943, 2023-07-15:
### Is your feature request related to a problem? Please describe.

Yes, to the best of my knowledge and after thorough research, it appears there is currently no mechanism in Espresense that allows for the retrieval of the Identity Resolving Key (IRK) from an Apple Watch using a Windows machine.

### Describe the solution you'd like

I would appreciate it if Espresense could be updated to support the extraction of the IRK from an Apple Watch when using a Windows operating system.

### Describe alternatives you've considered

Switching to a macOS system is not a universally applicable solution due to varying device availability among developers.

sotatech, 2023-10-12:
Instructions on how to get the IRK using Windows can be found here: [https://community.home-assistant.io/t/guide-getting-the-irk-for-iphone-and-apple-watch-without-macos-maybe-also-other-devices/623694/3/](https://community.home-assistant.io/t/guide-getting-the-irk-for-iphone-and-apple-watch-without-macos-maybe-also-other-devices/623694/3/)

I've tried it and it works fine. See my post for converting the IRK to ESPresence format there.

DTTerastar, 2023-11-03:
Super cool, if they found a way to make a watch pair, I should be able to do it as well.  Investigating...

DTTerastar, 2023-11-04:
Actually it works with the current Enroll flow just great!  The key insight was that app: https://apps.apple.com/us/app/bluetooth-terminal/id1058693037

If you install that on your watch, hit Enroll and connect to ESPresense it totally captures the IRK just like an iPhone!

DTTerastar, 2023-11-04:
Docs updated: https://espresense.com/beacons/apple#apple-watch-using-bluetooth-terminal-app

closed 2023-11-04 by DTTerastar