ESPresense/ESPresense #997: Home Assistant MQTT integration is deprecates

codyc1515, 2023-08-02:
### Describe the bug

From here https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/

Discovered entities with a name that starts with the device name.

This stops working in version 2024.2.0. Please address before upgrading.

### Logs

```text
Some MQTT entities have an entity name that starts with the device name. This is not expected. To avoid a duplicate name the device name prefix is stripped of the entity name as a work-a-round. Please inform the maintainer of the software application that supplies the affected entities to fix this issue.

List of affected entities:

button.espresense_lounge_restart
switch.espresense_lounge_auto_update
button.espresense_garage_restart
number.espresense_garage_max_distance
number.espresense_lounge_absorption
button.espresense_garage_update
light.espresense_lounge_led_1
button.espresense_lounge_enroll
sensor.espresense_garage_free_mem
switch.espresense_lounge_prerelease
light.espresense_garage_led_1
sensor.espresense_lounge_uptime
switch.espresense_garage_auto_update
sensor.espresense_garage_uptime
number.espresense_garage_absorption
number.espresense_lounge_max_distance
sensor.espresense_lounge_free_mem
switch.espresense_lounge_arduino_ota
switch.espresense_garage_arduino_ota
button.espresense_garage_enroll
switch.espresense_garage_prerelease
button.espresense_lounge_update
```

codyc1515, 2023-08-02:
And also:

Some MQTT entities have an entity name equal to the device name. This is not expected. The entity name is set to null as a work-a-round to avoid a duplicate name. Please inform the maintainer of the software application that supplies the affected entities to fix this issue.

List of affected entities:

binary_sensor.espresense_lounge
binary_sensor.espresense_garage

asafmatan, 2023-08-02:
(Also happens with latest espresense firmware v3.2.4 [I'm using now homeassistant 2023.8 )

daveismith, 2023-08-02:
Also seeing this on Home Assistant 2023.8 with espresence 3.2.4

DAVIZINH0, 2023-08-03:
This is a change on home assistant 2023.8 version with all the device that report data by mqtt integration.

in this post they expain better than me:

[https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099](https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099)

codyc1515, 2023-08-03:
Yes that's right but I believe the issue needs to be fixed by ESPresense (may be wrong here).

DAVIZINH0, 2023-08-03:
> Yes that's right but I believe the issue needs to be fixed by ESPresense (may be wrong here).

yes, i think the same

RMMTSLLP, 2023-08-03:
Facing similar issue since latest upgrade to;

Home Assistant 2023.8.0
Supervisor 2023.07.1
Operating System 10.4
Frontend 20230802.0 - latest

I am running esprescence 3.4.2

Is this a esprescence issue?

Perhaps someone could point me in the right direction to resolve the issue

Regards Roger Morris

[image]
[image]
[image]

jarikai, 2023-08-03:
It is the same problem here after HAOS update 2023.8.0.

codyc1515, 2023-08-03:
> Looks like there will be a fix in HA when September rolls around: https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099

That states "As for anyone using other MQTT addons/discovery providers, please notify the upstream library about the change."

ejpenney, 2023-08-03:
> Looks like there will be a fix in HA when September rolls around

I think this was specifically for Z2M, other MQTT related projects will need to make their updates.

The full quote:

> As it stands, Z2M will not be able to release a version with this fix until September 2023

Z2M is an upstream MQTT provider that needs to make changes for this, (and a PR is currently submitted), but other MQTT providers _also_ need to update their implementations, ESPresense being one of them.

asucrews, 2023-08-04:
Could this also effecting mqtt room part as well? The reason I ask is because I just set up a new node and am tracking a Android phone. And MQTT I see that the topics are updating and the phone's being tracked is showing up in the right room. However, within home assistant, it shows as not home no matter what I do.

ptyers, 2023-08-04:
Given that ESPresence is a fork of mqttroom I suspect same problem is applicable. I think, but cannot be definite, that ESPresence and its progenators are sending messages re entities that need reformatting to remove devicename

austwhite, 2023-08-04:
This is not an issue as much as a breaking change in Home Assistant and the entities will need updating.
Check the MQTT breaking changes under Naming of MQTT Entities
https://www.home-assistant.io/blog/2023/08/02/release-20238/#breaking-changes

This will need the entities updating in ESPresense, and a few other projects

jbouwh, 2023-08-04:
Home Assistant autodiscovery will automatically apply a work-a-round till this is fixed in the firmware.

arne182, 2023-08-06:
Can we just rename them?

RMMTSLLP, 2023-08-09:
If it is possible to rename the entities where should this be done?  From the example is it just as simple as removing the "ESPresense Living Room" bit from the entity settings in my case change "sensor.espresense_lounge_count" to "sensor.count"

broyuken, 2023-08-10:
It's the friendly name that should be changing I think, not the entity name. For example, the friendly name for button.espresense_office_update should change from ESPresense Office Update to Update and the device name should be ESPresense Office

austwhite, 2023-08-12:
I think the title of this bug should reflect the actual issue.  nothing in MQTT is deprecated as such.  It was a naming convention change that means MQTT entities can't contain the device name in the name of the entity.
:)

broyuken, 2023-08-14:
If someone can find the mqtt discovery code I can try to poke around and see if I can fix it then submit a PR.

Jens-Wymeersch, 2023-11-04:
Dear all, I'm on the latest version and still I don't see this warming popping up. When will this be sorted ?

DTTerastar, 2023-11-05:
Released v3.3 just now!

Jens-Wymeersch, 2023-11-05:
@DTTerastar The problem hasn't been fixed, unfortunately. I'm still getting `MQTT entity name starts with the device name in your config` 
Any other way ?

austwhite, 2023-11-05:
At the moment this is still just a warning, but I think we only have about 3 months left.   

With the MQTT entities, did you try completely deleting the entities and letting the device reads itself?

Jens-Wymeersch, 2023-11-05:
@austwhite What do you mean exactly ? I just updated the devices to the latest release and thought this would be sorted.

austwhite, 2023-11-05:
Home Assistant can sometimes need playing with a bit to get things to work .  The MQTT entities may not auto update.

Remove the MQTT entities from home assistant, power cycle your espresence devices so they readd to Home Assistant fresh 
See if that fixes the issue

Jens-Wymeersch, 2023-11-05:
@austwhite I've done this for each of the espresense devices I have but still get the warning in the logs. What can be done else ?

gerard33, 2023-11-05:
For me the warnings in Home Assistant are gone. I already installed the beta with the fix a while ago and think the warnings disappeared after deleting the retained messages with MQTT Explorer.

RMMTSLLP, 2023-11-05:
Hi Guys,  I updated all my devices to beta version 3.3.0b8 when it was updated to resolve this issue.  All was fine no error messages. Nothing else was done only updated the esp devices.
Regards

paul-thomas1964, 2023-11-05:
I’ve updated one of my devices, although it reports as being online, it no longer reports the presence of my iphone.  I’ve restarted the device & HA with no luck.  My other unupdated device is still reporting fine with no warnings.

Jens-Wymeersch, 2023-11-05:
@gerard33 I've been looking into MQTT Explorer, but I don't see any reference to this issue.
I didn't install the beta but the final release. I'm just wondering, the exact steps you took in MQTT Explorer.
@DTTerastar it looks like the beta removed the warming but the final release didn't.

pkscout, 2023-11-05:
I had to delete all the MQTT topics related to ESPresence (except for /espresence/settings and /espresence/devices) and then restart each of my ESPresence devices (after doing the update).  That seemed to rename the sensors.

andrey-yantsen, 2023-11-05:
@paul-thomas1964, could that be due to #1032?

Jens-Wymeersch, 2023-11-05:
@pkscout killed everything in MQTT via MQTT Explorer and in the UI in home assistant. What is the name of the sensor now ? And how does it look like in MQTT Explorer ?

For me the topic is for the living room : espresense/rooms/living_room. Does this change ?

pkscout, 2023-11-05:
@Jens-Wymeersch the name of the sensors are all now sensor.esp_[something].

[image]

In MQTT Explorer the topics all remain the same.  I don't think the topic paths are the thing HA is depreciating.  I think it's sensors that have the name of the device in the name.  Here's an actual topic for that sensor's config:
```
{
    "dev":
        {"ids":["espresense_f8c10c"],
         "cns":[["mac","E0:5A:1B:0C:C1:F8"]],
         "name":"ESPresense greatroom",
         "sa":"greatroom",
         "sw":"v3.3.0",
         "mf":"ESPresense (esp32)",
         "cu":"http://172.16.1.27","mdl":"ESP32-D0WD-V3"
        },
    "~":"espresense/rooms/greatroom",
    "name":"Connectivity",
    "uniq_id":"espresense_f8c10c_connectivity",
    "json_attr_t":"~/telemetry",
    "stat_t":"~/status",
    "dev_cla":"connectivity",
    "pl_on":"online",
    "pl_off":"offline"
}
```

Jens-Wymeersch, 2023-11-05:
@pkscout So if I manually change the name to esp_living_room for the living room, I'll get rid of this problem. I might just do this. No ?

pkscout, 2023-11-05:
I am not an expert on MQTT and Home Assistant sensors.  In theory as long as the sensor has a unique ID, you could manually change every `espresence_` sensor to `esp_` and it will probably work.  I'm just not sure what the implications are down the road if you do that.

paul-thomas1964, 2023-11-05:
> @paul-thomas1964, could that be due to #1032?

Thanks, I’ve made that change and its working again.

Jens-Wymeersch, 2023-11-05:
In the end, it was one device that wasn't set to auto-update that created all the noise. Warning is now totally gone. Deleting MQTT info and restart of all devices did the trick.

DTTerastar, 2023-11-06:
Okay to close this then?

Jens-Wymeersch, 2023-11-06:
For my case it can be.

PR #1015 (2023-08-22): fix: rename mqtt entities to conform to new home assistant reqs
Attempts to fix #997. I _think_ I got them all, however HA still is showing me the below three. For the life of me I cannot find where those would be set or if it's just bad historical data or something.

* status_led
* active_scan
* ota_update
closed 2023-11-06 by DTTerastar