ESPresense/ESPresense #998: Cannot connect to WiFi after power cycle

draftpunk, 2023-08-03:
### Describe the bug

WiFi seems to connect after flashing ESP32 and specifying WiFi, but after a power cycle, it will not reconnect. 

### Version

3.2.4

### Logs

```text
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Pre-Setup Free Mem: 206732
Setup network
Connecting to WiFi SSID 'WoodButSlower'................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................*............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................... failed (status=6).
Starting access point for configuration portal.
```

draftpunk, 2023-08-04:
Just wanted to add that I previously had this same ESP32 board flashed as an ESP Bluetooth proxy device and had no issues connecting to my WiFi.

mashedkeyboard, 2023-08-10:
I'm experiencing this same issue, with the same logs (bar the SSID being different!). Other devices around seem to also occasionally experience WiFi trouble when close by to the ESPresense device failing to connect, but then have no issues when it is not there: I am assuming this is just interference from the autocreated hotspot, but mentioning for completeness.

I am using an ESP32 WROOM32.

kyemacdonald, 2023-11-18:
I have this problem as well.  ESP32 Wroom boards.  Across multiple boards.

kyemacdonald, 2023-11-18:
If you do a full erase and flash the board will connect to the wifi.  But after that it will no longer connect reliably.

If you flash a different espresence version without doing a full erase it still won't connect to wifi.

kyemacdonald, 2023-11-18:
This is the web log from a clean install to hitting the reset device button.

`1 New    | a4c138cfe21d | a4c138cfe21d                                              -93dBm 
0 Reconnect timer
0 Reconnecting to MQTT...
Disconnected from MQTT; reason 0
1 New    | bc7e8b2db02d | bc7e8b2db02d                                              -98dBm 
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
0 Reconnect timer
0 Reconnecting to MQTT...
Disconnected from MQTT; reason 0
0 Reconnect timer
0 Reconnecting to MQTT...
ets Jul 29 2019 12:21:46

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Pre-Setup Free Mem: 206732
Setup network
Connecting to WiFi SSID 'Skyenet-IOT'................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................*............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................... failed (status=6).
Starting access point for configuration portal.`

kyemacdonald, 2023-11-24:
I've narrowed down the problem to the web gui.

If you install espresense and then setup the wifi through the web gui the device will connect once to wifi an never again.  

If however you install through the webgui and then disconnect the esp as soon as you see the finished page, and setup the wifi through the hotspot then the device will work from there on.

Interestingly there is a core panic on the first attempt to connect to the wifi hotspot.  This only happen on the first time you connect to it, subsequent connections work perfectly.  This has occurred across multiple devices. 

`Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40099998  PS      : 0x00060633  A0      : 0x80099df6  A1      : 0x3ffd5180  
A2      : 0x3ffe3c48  A3      : 0xabba1234  A4      : 0x3ffe0454  A5      : 0x00000003  
A6      : 0x00060623  A7      : 0x00060620  A8      : 0x00000012  A9      : 0x00000007  
A10     : 0x3ffe06fc  A11     : 0x3ffe3cbc  A12     : 0x3ffe0478  A13     : 0x3ffe04a0  
A14     : 0x3ffe3c50  A15     : 0x00000013  SAR     : 0x00000013  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xabba1240  LBEG    : 0x40092368  LEND    : 0x4009237e  LCOUNT  : 0xffffffff  

Backtrace: 0x40099995:0x3ffd5180 0x40099df3:0x3ffd51a0 0x40099ef4:0x3ffd51c0 0x4008409a:0x3ffd51e0 0x400840ad:0x3ffd5200 0x400840d6:0x3ffd5220 0x4009a2c9:0x3ffd5240 0x401301e8:0x3ffd5260 0x40130c9e:0x3ffd5280 0x40134b8c:0x3ffd52a0 0x40134d1d:0x3ffd52d0 0x40135439:0x3ffd5300 0x401354a7:0x3ffd5320 0x40131e1d:0x3ffd5360 0x400e0412:0x3ffd5380 0x4012efb4:0x3ffd53a0

ELF file SHA256: 0000000000000000

Rebooting...`

oddboy, 2023-11-26:
same issue here.  I can get it to work once, but after power cycle, wifi will not connect.  tried on several different esp32 boards now.

`ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Pre-Setup Free Mem: 215796
Setup network
Connecting to WiFi SSID '<my ssid>'.........................................................................................
                                                                                                                      .
SSID: 'espresense-living-room'
                              IP: 192.168.4.1
`

chasecee, 2023-12-13:
@kyemacdonald's solution above worked for me.

draftpunk, 2023-12-13:
What this actually turned out to be for me is the USB-to-serial chip (CH340C) is causing a poor WiFi signal. After desoldering pin 16 (vcc) after the board has been flashed and programmed to WiFi, the board connects to WiFi and works great. For more info: https://community.home-assistant.io/t/something-is-very-wrong-with-the-wifi-on-the-esp32-boards-i-ordered-from-aliexpress-is-there-any-way-to-salvage-these-and-fix-the-wifi/477631/12

ecto1a2003, 2024-01-24:
similar issues here, ill have everything up but if an update comes or there's a power cycle everything goes away. i have to try to wet up through the hotspot but that is slow and seems to not respond 49 times out of 50. very frustrating. all my other esp32 boards reconnect perfectly fine.

ESP32 Wroom

OpherV, 2024-01-25:
Confirming the exact same behavior with ESP32 Wroom

ecto1a2003, 2024-02-02:
my issues are unrleated to usbc, i have micro usb powerd ones with the smae issue.

Wildbill-Z, 2024-02-15:
Maybe related to [this](https://github.com/ESPresense/ESPresense/issues/1227)

borodaaaaa, 2024-02-24:
Any updates on the issue ? Espresens devices are not stable at all due to this issue . Getting an available from time to time. Please assist

Wildbill-Z, 2024-02-24:
I went back to V3.2.2 with all devices, as this seems to be the latest version without this problem.
Just reflashed (downgraded) all of them with Web-Installer. No config was deleted, all of them just worked fine again after flashing.

b1t-hunter, 2024-04-18:
I am having the same issue, are there any updates?

talalalisnotavailable, 2024-06-15:
Same issue here - on an M5Stamp Pico - one device has this problem out of 2

carsaig, 2024-06-26:
Same issue here. It's comepletely unusable. I'll ditch it until someone comes up with a fix.

Wildbill-Z, 2024-12-28:
Not solved at all, so should not be closed automatically.

kyemacdonald, 2024-12-28:
The solution is move to Bermuda....

ecto1a2003, 2024-12-28:
The county?

kyemacdonald, 2024-12-28:
@ecto1a2003 

https://github.com/agittins/bermuda/wiki/

Wildbill-Z, 2024-12-28:
Not very helpful (to me), as Bermuda doesn't seem to support mmWave/radar sensor which are connected directly to ESP32 to have a presence detection even without using BLE.

kyemacdonald, 2024-12-28:
It's just esphome, running a Bluetooth proxy. So you can just add mmwave to the esphome configuration.

Wildbill-Z, 2024-12-28:
OK, just flew over the letters at first time. I think I will give it a try with one device to test. Thanks.

ecto1a2003, 2024-12-28:
Going back to the original issue, ive had no problems since i upgraded my wifi from a single asus router to a unifi system. Seems consumer wifi has a limit to how many devices it can handle at once.

Wildbill-Z, 2024-12-28:
Problem in this issue is, that devices won‘t boot up properly when not connected to a serial USB console. If connected to a computer, all is working fine. If connected to a USB wall plug or similiar, they won‘t boot in 9 of 10 tries.
It has nothing to do with which wifi hardware one uses, how many access points there are, and so on.
BTW, I use Unifi for years (Edgerouter and several access points). I had no problem until formware 3.2.2, with 3.2.4 the problem in this issue occured, going back to 3.2.2 and all was working fine again.

ecto1a2003, 2024-12-28:
What usb bricks are you using? Sounds like a power issue there

Wildbill-Z, 2024-12-28:
Huh, several one. It doesn't matter. Even with not so cheap with 3.0A and stable 5.1V the problem occurs with 3.2.4. Read [this linked issue](https://github.com/ESPresense/ESPresense/issues/1227), and you will see, it's wether a power nor a wifi issue. It's just a problem that the code expects an usb serial console to boot properly, which you won't have, if using a usb power supply.

ecto1a2003, 2024-12-28:
Dunno what to tell you. Haven't had this problem in ages, have you tried 3.3.5?

Wildbill-Z, 2024-12-28:
As I wrote, I‘m not having any problem, since I use outdated version 3.2.4. I just wanted to autoclose this issue by stalebot, as it exists since 3.2.4 and isn‘t solved until now. 
I tried all versions abaove 3.2.4 and all have the problem. And I think, i a new release would have a fix for the issue, it would have been closed by developer/maintainer and not stalebot. That‘s the way, Github is working (shopuld work).

Wildbill-Z, 2024-12-30:
> The solution is move to Bermuda....

Tried Bermuda with one ESP32 and DFRobot mmwave and all worked well OOTB (OK, I had to tell the right GPIO 😁 ). So I switched to Bermuda with all of my espresense-devices. In fact, I now have some more devices as there were some Shelly Gen 2 working as bluetooth proxy to.

Thanks to @kyemacdonald

Aziona, 2025-01-11:
I had the same problem, strangely enough with ESP32's that already worked for months providing temperature readings through DS18B20 senssors to my MQTT-server. However, when I put Espresense on them, most of them did not want to (re)connect to wifi anymore. I had none of these problems when using the ESP32-DS18B20-MQTT-code from Random Nerder Tutorials.
I spend several days/weeks working out the problem. As I also read something about faulty designs using the ch340 serial to usb converter, I ordered five new esp32's using the CP2102 USB to serial converter instead of ch340 and ordered a bit more expensive ones at Aliexpress. 
I just put Espresense on all five of them and all five work perfectly also after rebooting them several times. 
The previous ones did either not connect at all or did not reconnect after the first power cycle. 

Also I had both micro-usb as usb-c versions of ESP32. it seemed that the problem was mostly in the usb-c versions. I think that could be explained as the ones that I had almost all used the ch340 converter instead of the CP2102.

Bermuda is not an option for me as I use Domoticz with the Espresense plug-in and I do not know how ESPhome works, only work with the Arduino IDE.

closed 2025-07-25 by stale[bot]