ESPresense/ESPresense #1004: WROOM-32 DHT22

dancingbear84, 2023-08-09:
### Describe the bug

I have installed version 3.2.4 and a DHT22 sensor, when running in ESPresence it returns nan values for the Temp and Humidity.
When I use the same sensor on a different board, same sensor, same cables etc on a board running ESPHome it returns values.
I have tried different pins, I have tried 5v and 3.3v, I have tried different GND pins, I have tested using a multi meter 

### Version

3.2.4

### Device (be specific)

ESP32-WROOM-32

### Logs

```text
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Pre-Setup Free Mem: 206732
Setup network
Connecting to WiFi SSID 'IotWifi'............................................................192.168.x.x
Firmware:     esp32
Version:      v3.2.4
WiFi BSSID:   D2:21:F9:9F:95:09 (channel=6 rssi=-55)
IP address:   192.168.x.x
DNS address:  192.168.x.x
Hostname:     espresense-kitchen
Room:         Kitchen
Mqtt server:  192.168.x.x
Max Distance: 16.00
PIR Sensor:   disabled
Radar Sensor: disabled
I2C Bus 1:    sda=21 scl=22
Scanning I2C for devices on Bus 1...
No I2C devices found

DHT11 Sensor: disabled
DHT22 Sensor: pin 5
DHT Offset:   1.00
Query:        
Include:      
Exclude:      
Known Macs:   
Count Ids:    
Post-Setup Free Mem: 138040

Temp: nan'C Humidity: nan%
Temp: nan'C Humidity: nan%
Temp: nan'C Humidity: nan%
Temp: nan'C Humidity: nan%
Temp: nan'C Humidity: nan%
```

h4irforceone, 2023-09-16:
Try this wiring! 
[image]

dancingbear84, 2023-09-25:
Thanks, I’ve tried that, I have set the pin to G2 and tried on 5v and 3.3v but still the humidity and temperature is unavailable.

dancingbear84, 2023-10-06:
I have also tried a BME280 and an LD2410 and am unable to get any of them to work correctly, they all report unknown in HA, I suspect this must something that I am doing incorrectly, I have seen references to needing to install ESP32 versions with sensor support, but I can not find a version "with sensors" so I suspect this is old, and all versions now have sensor support

kloodhu, 2023-10-17:
Pls answer to this! the sensors should work, or "this is bug, don't waste your time"?

kloodhu, 2023-10-21:
?

DTTerastar, 2023-11-04:
Yes, all versions include sensor now.  Did you ever figure it out?

gupi, 2023-11-06:
Hi, why didn't you use a pullup resistor of 4.7k between signal and +3.3V?

gupi, 2023-11-06:
How frequent is your ESP requesting data from the DHT? The DHT is a quite slow device and it tends to stop working if the next data request is performed when the previous on is not yet finished.

CrazYoshi, 2024-03-28:
Any update about that?
I'm using ESPresence 3.3.2 and I have the same issue using a DHT22. I've tried several PIN with the same result.
On ESPhome the sensor works as expected.

sharkpunch5, 2024-09-23:
i'm getting an issue where the sensor works however reports fine and then goes to NaN after some time, then works fine again - could be to do with polling interval perhaps? not sure how to address

The-May, 2024-10-09:
> Try this wiring! [...]

Tried this one out and it does not seem to work. ESPresense Version: v3.3.5
I connected it like you recommended.

According to the diagram G02 is PIN24 
[image]
Therefore I have set it up like this in the web ui, saved and restarted.

[image]

Mqtt topic is getting published but I never get a value from it:
[image]

if I have done nothing wrong, I can confirm that something is wrong with that esp-type

The-May, 2024-10-18:
I can totally verify OPs issue:

I had a "classic" ESP32-WROOM with micro USB like in the picture
[image]
and could reproduce the issue from @dancingbear84 . I also tried other cables, other GPIO-Ports, used 5v etc etc without any success and Only had Data for maximum 1 Minute on my "classic" esp32.
After that it returned NAN and became unavailable in HA.
Espresence (web ui) itself worked nevertheless though.

I bought myself a esp32 Mini like in the picture
[image]

unfortunately I dont know if it is a CP2102  or CH9102F, but it seems to be much more stable now and I could get reliant data for more than 10 Minutes. 

I used similiar/same wiring from @h4irforceone on both esp32-types:
ESP32 - DHT22:

GPIO4 - Out
3,3v - +
GND - -

Espresense was configured like this on both devices:

[image]

additional information: I used the same sensor in this test.

Toolhead, 2024-11-13:
Experiencing the same.

pmrosa, 2024-11-23:
Any update on this? Bought a bunch of ESP32 and DHT22 for ESpresense but can only use it with sensors in ESPhome.

Toolhead, 2024-11-28:
Take a look at Bermuda BLE triangulation, does the same but in esphome.
Have it running now with the DHT's working!

The-May, 2025-05-09:
> Take a look at Bermuda BLE triangulation, does the same but in esphome. Have it running now with the DHT's working!

Sounds good, but I am not a fan of custom HACS-Integrations that might break from time to time / over time.
I am just really curious why this breaks and what the intended issue is in there. Imma ask in the discord also. I cant fathom what the issue is.