ESPresense/ESPresense #1011: Apple watch series 8 appear not home when sceen is off

xxlin0127, 2023-08-14:
### Describe the bug

I got the irk from keychain and got it working on HA, but the watch always appears not home since I'm not looking at it.I look at the ui, and couldn't find the watch device when the sceen is off, but  the screen is off most of the time. I also tried to add the device id on query, still not working. Anyone has the same problem?

Abulrock, 2023-11-09:
just a quick update: the issue resolved itself, its been a little over 24 hours and not once it failed to report. 
while it wasnt working i added my iphone to be tracked, and i restarted my watch and switched between some focus modes at some point, and for some reason it was just working again, but that's all I did

bakerkj, 2023-11-12:
My iPhone has a similar behavior with 17.1.1

andrey-yantsen, 2024-01-25:
I've been experiencing the same issue for a couple of days, but after I restarted my Watch — the issue was gone 🤷

sgtpepperaut, 2024-05-28:
Iphone 15 works flawless but Watch disconnects rather quickly. only when i raise my wrist it reconnects.  did you guys do anything else but reboot?

[image]

sgtpepperaut, 2024-05-29:
this is used to work perfekt with the watch checking in every 5 seconds regardless if awake or not.
i did observer another signal that is coming in at every 5 seconds that also seems to originate from the watch (confirmed by moving the watch around).... so i wonder what that is? it appears to be another random mac! but its defintly coming from the watch (like clockwork haha).... i wonder what they have changed?

closed 2024-12-10 by stale[bot]