ESPresense/ESPresense #1013: Enroll UI crashes device every time ( 3.3.0b4 )

johntdyer, 2023-08-15:
### Describe the bug

Everytime I try to goto the UI /ui to enroll a device it fails to load the form and in the console logs I see the following

```
Connecting to WiFi SSID 'dyer.iot'........................................................................192.168.121.202
Firmware:     esp32c3
Version:      v3.3.0b4
WiFi BSSID:   78:8A:20:8A:DF:F4 (channel=11 rssi=-59)
IP address:   192.168.121.202
DNS address:  192.168.120.1
Hostname:     espresense-basement-office
Room:         basement-office
Mqtt server:  192.168.120.254:1883
Max Distance: 1.00
PIR Sensor:   disabled
Radar Sensor: disabled
Switch One Sensor:   disabled
Switch Two Sensor: disabled
Button One Sensor:   disabled
Button Two Sensor: disabled
I2C Bus 1:    sda=19 scl=18
Query:        
Include:      irk:
Exclude:      apple:
Known Macs:   
Count Ids:    
Post-Setup Free Mem: 178844

0 Motion | Pir: no Radar: no
0 Advert | iBeacon
0 Switch | Switch One: no Switch Two: no
0 Button | Button One: no Button Two: no
0 Set    | exclude to apple:
0 Set    | include to irk:
0 New    | a4c1387f0ca1 | a4c1387f0ca1                                              -71dBm 
0 New    | 4543c59a9f97 | 4543c59a9f97                                              -53dBm 
0 New    | 2418c6196a96 | 2418c6196a96                                              -83dBm 
0 New    | 22d6247d67f6 | 22d6247d67f6                                              -91dBm 
0 New    | 6d0d549e1ff7 | 6d0d549e1ff7                                              -53dBm 
0 New    | b03829b0fac4 | b03829b0fac4                                              -80dBm 
0 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-108-57121 to {"id":"node:livingroom","name":"livingroom"}
0 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-136-33223 to {"id":"node:kitchen","name":"kitchen"}
0 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-140-12271 to {"id":"node:bedroom","name":"Bedroom"}
0 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-152-28199 to {"id":"node:server_room","name":"server-room"}
0 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-152-61569 to {"id":"node:basement_office","name":"basement-office"}
0 New    | 248b4eccf502 | 248b4eccf502                                              -78dBm 
0 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-176-24422 to {"id":"node:maritzas_office","name":"maritzas-office"}
0 Config | irk:8fad305969973566572740e8401851be to {"id":"johns_iphone_14","name":"johns-iphone-14"}
0 Alias  | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-152-61569 to node:basement_office
0 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-152-61569 to {"id":"node:basement_office","name":"basement-office"}
ArduinoOTA configured and ready
0 New    | bc455b6da113 | bc455b6da113                                              -62dBm 
0 New    | 2673e38a058a | 2673e38a058a                                              -97dBm 
0 New    | 45094ae04a89 | apple:1006:100                                            -88dBm 431dbdf4bc68
0 New    | d66e54aa9c01 | d66e54aa9c01                                              -94dBm 
0 New    | 586096efff15 | johns_iphone_14                                           -82dBm 391f62a159e228
0 New    | 349454fba3be | 349454fba3be                                              -94dBm 
0 New    | 551c3ab44edf | apple:1007:11-12                                          -92dBm 081fd78d319578
0 New    | d4adfc787ec9 | d4adfc787ec9                                              -95dBm 
0 New    | 543d64967200 | apple:1006:10                                             -96dBm 431dbdf4bc68
0 New    | c869cd3666a7 | apple:1005:9-12                                           -92dBm 0214898c39
0 New    | d4d4daef2f8e | node:bedroom                                              -76dBm 
0 New    | 6d42fb705c69 | irk:6e079d58e180a46bea4619cb3d3d7b29                      -66dBm 2298068a4f
0 New    | d81f128aab8a | d81f128aab8a                                              -97dBm 
0 New    | f7468d58bdc5 | name:n024q                                                -92dBm 
0 New    | de110a4aa230 | de110a4aa230                                              -68dBm 
0 New    | d0c1fe1355a5 | d0c1fe1355a5                                              -52dBm 
0 New    | db4711e7f4cb | db4711e7f4cb                                              -82dBm 
0 New    | 5e00b1e92b7a | apple:1006:10-12                                          -98dBm 0319c0836c58
[... 56 lines cut ...]

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382128
SPIWP:0xee
mode:DIO, clock div:1
```

### Version

v3.3.0b4

### Include filter

irk:

### Exclude filter

apple:

### Device (be specific)

M5 Stamp

johntdyer, 2023-08-15:
Same issue on 3.2.4 FWIW

```

0 Close  | eae497133257 | eae497133257                                              -38dBm
0 New    | eae497133257 | eae497133257                                              -38dBm 
0 New    | 551c3ab44edf | apple:1007:11-12                                          -63dBm 081fd78d319978
0 New    | ca4766a45fbb | ca4766a45fbb                                              -55dBm 
0 New    | ededfa77728d | sonos:ededfa77728d                                        -84dBm 
0 Close  | fe07c36352f2 | fe07c36352f2                                              -30dBm
0 New    | fe07c36352f2 | fe07c36352f2                                              -30dBm 
0 New    | c25c5e26f39d | apple:findmy                                              -71dBm 
0 Close  | 75f6527536a9 | johns_iphone_14                                           -33dBm
0 Close  | d0373e9f184a | d0373e9f184a                                              -36dBm
0 New    | d0373e9f184a | d0373e9f184a                                              -36dBm 
0 New    | d4d4daef2f8e | node:bedroom                                              -45dBm 
0 Close  | e3dc7db85471 | e3dc7db85471                                              -36dBm
0 New    | e3dc7db85471 | e3dc7db85471                                              -36dBm 
0 New    | e41baad73034 | e41baad73034                                              -75dBm 
0 New    | d66e54aa9c01 | d66e54aa9c01                                              -64dBm 
0 New    | f7468d58bdc5 | name:n024q                                                -67dBm 
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42015ee4  RA      : 0x42015f38  SP      : 0x3fcb3d10  GP      : 0x3fc8fc00  
TP      : 0x3fc937ec  T0      : 0x4038bd2e  T1      : 0xfffffff0  T2      : 0xffffffff  
S0/FP   : 0x3fcb4c78  S1      : 0x3fcc4614  A0      : 0x00000001  A1      : 0xefd102cd  
A2      : 0x3fcc27b0  A3      : 0x00000001  A4      : 0x3fc9b89c  A5      : 0x600c2000  
A6      : 0x65676173  A7      : 0x73656d72  S2      : 0x3fc9ca54  S3      : 0x3fcc4624  
S4      : 0x3fcc46c0  S5      : 0x00000001  S6      : 0x0000000a  S7      : 0x00000004  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x736e6f69  T4      : 0x736e6574  T5      : 0x78452d74  T6      : 0x656b636f  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0xefd102dc  
MHARTID : 0x00000000  

Stack memory:
3fcb3d10: 0x00000000 0x00000000 0x00000000 0x040181a4 0x0000000a 0x00000001 0x3fcc4648 0x00000002
3fcb3d30: 0x3fcc460e 0x00000002 0x3fcc4614 0x42015fba 0x3fcc460e 0x00000002 0x3fcc4648 0x4204afc6
3fcb3d50: 0x0000000a 0x00000001 0x3fcc4648 0x00000002 0x3fcc460e 0x00000002 0x3fcc4614 0x420160b8
3fcb3d70: 0x3fcc2888 0x3c137000 0x3fcc4614 0x42015e7e 0x00000001 0x3fc93704 0x42014270 0x420143e0
3fcb3d90: 0x00000002 0x3fcc28f4 0x3fcc28e0 0x00000001 0x00000000 0x3fcc43c8 0x3fcc2888 0x42016288
3fcb3db0: 0x00000000 0x3fcc43c8 0x3fcc2888 0x420fd660 0x3fcc4614 0x3fcc2888 0x3fcc4416 0x000001fa
3fcb3dd0: 0x00000002 0x00000005 0x3fc9c000 0x3fc9bcb8 0x00000001 0x3fcc2888 0x3fca9118 0x420fd704
3fcb3df0: 0x00000001 0x3fcc2888 0x3fca9118 0x420fddd0 0x00000000 0x3fca9118 0x00000000 0x00000000
3fcb3e10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb3e30: 0x00000000 0x00000000 0x00000000 0x4038bcda 0x00000000 0x00000000 0x00000000 0x00000000
3fcb3e50: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154
3fcb3e70: 0x3fcb3c00 0x000012b6 0x3fc96914 0x3fc96914 0x3fcb3e70 0x3fc9690c 0x00000016 0x3fcb1dc0
3fcb3e90: 0x3fcb1dc0 0x3fcb3e70 0x00000000 0x00000003 0x3fcb1e6c 0x6e797361 0x63745f63 0x3b2a0070
3fcb3eb0: 0x006c6b64 0x00000000 0x3fcb3e60 0x00000003 0x00000000 0x3fcc2330 0x42057034 0x00000000
3fcb3ed0: 0x3fc9d02c 0x3fc9d094 0x3fc9d0fc 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000
3fcb3ef0: 0x00000000 0x42109c8a 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb3f10: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb3f30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb3f50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb3f70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb3f90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb3fb0: 0x00000000 0x00000000 0x00000000 0x00000000 0x3f000000 0x00000018 0x3fc9ca54 0x3fc9ca54
3fcb3fd0: 0x3fcb3fc0 0x00000000 0x00000000 0x00000000 0x0000001c 0x00000000 0x3fc9ca54 0x00000000
3fcb3ff0: 0x00000000 0x00000000 0xff000000 0x3fcb3fdc 0x00000040 0x3fc945ec 0x00000000 0x00000000
3fcb4010: 0x3fcb4048 0x3fcb405c 0x00000000 0x00000000 0x00000000 0x0218ffff 0x00000000 0x3fcb3fc0
3fcb4030: 0x00000067 0x3fc9ca54 0x3fc9ca54 0x3fcb4000 0x3fcb3f00 0x00000010 0x3c12abe4 0x3fc9ca54
3fcb4050: 0x00000118 0x3fcb4004 0x00000010 0x3c12ac00 0x3fc9ca54 0x00000000 0x3fcb4004 0x00000028
3fcb4070: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x0000ffff 0x00000058
3fcb4090: 0xb33fffff 0x3fcb4068 0x0000000c 0x3fcb40ac 0x3fc9ca54 0x3fcb4094 0x00000054 0x3fcb40ac
3fcb40b0: 0x3fcb40ac 0x3fcb40ac 0x3fcb40ac 0x00000000 0x3fcb40c4 0xffffffff 0x3fcb40c4 0x3fcb40c4
[... cut ...]
0 New    | eae497133257 | eae497133257                                              -65dBm 
0 Close  | fe07c36352f2 | fe07c36352f2                                              -29dBm
0 New    | fe07c36352f2 | fe07c36352f2                                              -29dBm 
0 New    | f7468d58bdc5 | name:n024q                                                -68dBm 
0 New    | 74e435ceefbc | apple:1006:10-12                                          -70dBm 0f19e2145648
0 New    | fdf272ed994d | fdf272ed994d                                              -62dBm 

```

gunnarbeutner, 2023-09-24:
I think I ran into the same problem with my own ESP32C3 devices here: https://github.com/ESPresense/ESPAsyncWebServer/pull/1

FlyGoat, 2023-10-15:
Can confirm everything works now by cherry-picking https://github.com/me-no-dev/ESPAsyncWebServer/pull/952 and https://github.com/me-no-dev/ESPAsyncWebServer/pull/953 into ESPPresence's ESPAsyncWebServer tree.

@DTTerastar is it possible to make a release with this change?

gunnarbeutner, 2023-10-29:
@FlyGoat Have your ESP32C3 nodes been _really_ stable though? I'm monitoring the `uptime` attribute in the telemetry messages for mine (about 20 nodes) and noticed that they still crash regularly. I ran into two more issues (not related to the UI crashes):

* Stack overflow in the BLE task (happens about a total of 5-10 times per day here): #1069 
* Race condition because multiple threads are using the BleFingerprint array at the same time (exceedingly rare, maybe one crash every 2-3 days). This probably happens more frequently based on how many BLE advertisements the sensor nodes see (about 37 per second here). I'll post another comment with the issue ID once I've cleaned up the fix a bit.

With those two fixes my nodes have been considerably more stable. Previously they'd barely survive an hour or two without crashing:

[image]

FlyGoat, 2023-10-29:
@gunnarbeutner Just check my logs form past week, for #1069 I had met once, for race condition it never happens on my side. I guess that's because I don't have many devices broadcasting on BLE.

DTTerastar, 2023-11-02:
> Can confirm everything works now by cherry-picking [me-no-dev/ESPAsyncWebServer#952](https://github.com/me-no-dev/ESPAsyncWebServer/pull/952) and [me-no-dev/ESPAsyncWebServer#953](https://github.com/me-no-dev/ESPAsyncWebServer/pull/953) into ESPPresence's ESPAsyncWebServer tree.
> 
> @DTTerastar is it possible to make a release with this change?

Yup, I merged both of those in and am releasing a beta now

DTTerastar, 2023-11-02:
Well I tested it and the websocket changes completely broke it.  I'll release the version with just https://github.com/ESPresense/ESPresense/pull/1069 and then work on the other

DTTerastar, 2023-11-03:
There is a PR with a fix to ESPAsyncWebServer.  I decided to just use esphomes version because it's had more work put into it then mine or the official version.  If you have companion you can install it like this:

[image]

Or for USB flash use this:

https://espresense.com/alpha

gunnarbeutner, 2023-11-03:
I've updated my nodes and it seems to be working fine here:

```
$ websocat ws://192.168.3.80/ws
{"state":{"enrolling":false},"room":"486d06"}
```

DTTerastar, 2023-11-03:
The official beta would crash on that?

gunnarbeutner, 2023-11-03:
Dunno about the beta, but I've just rebuilt the master branch (i.e. without 369f8d9) and yeah, even just connecting to it with `websocat` causes it to crash:

```
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Setting breakpoint at 0x4202d006 and returning...
```

Here's the HTTP request `websocat` sends - in case you want to reproduce it (though I suspect this _might_ only crash on RISCV-based processors):

```
GET /ws HTTP/1.1
Host: 192.168.3.80
Connection: Upgrade
Upgrade: websocket
Sec-WebSocket-Version: 13
Sec-WebSocket-Key: GqkDV1i2gmA1EO15M9IvdQ==
```

DTTerastar, 2023-11-03:
Okay, that's good enough for me.  Merging now...

DTTerastar, 2023-11-03:
Fixed in [v3.3.0b10](https://github.com/ESPresense/ESPresense/releases/tag/v3.3.0b10)

closed 2023-11-03 by DTTerastar