ESPresense/ESPresense #1021: M5STACK ATOM Matrix Addressable LED Text

colinhatfield, 2023-08-23:
### Is your feature request related to a problem? Please describe.

I would like to display the value (Number) of an attached sensor on the matrix display

https://github.com/ESPresense/ESPresense/assets/48314728/13d32b54-773d-4f92-b794-ce4be54e6e58

### Describe the solution you'd like

With ESPHome, I currently use the following to retrieve a value via mqtt and have the temperature scroll across the display:
[Example.yaml.txt](https://github.com/ESPresense/ESPresense/files/12422562/Example.yaml.txt)

Ideally I would like to have a drop-down with options from the sensors I've got configured, such as:
Humidity
Lux
Temperature

And the option to set the text colour would be a nice bonus

### Describe alternatives you've considered

I could use a ATOM Lite to get the sensor data and then use ESPHome on the ATOM Matrix to get the value via MQTT and display it

### Additional context

The device I'm using
https://shop.m5stack.com/collections/m5-controllers/products/atom-matrix-esp32-development-kit

colinhatfield, 2023-08-23:
This is my first feature request so hopefully it has the detail you need. Also I just wanted to say how much I love the project and appreciate the hard work you've already done!

DTTerastar, 2023-11-04:
That's a cool idea!  I'd love it as well.