ESPresense/ESPresense #1022: Crash when login to device

davlaw, 2023-08-24:
### Describe the bug

When you log into the device it causes it to reboot.

And while using the serial console, when I downloaded the logs it rebooted again..

This was a bigger problem when I had pasted a url into the update field, and it downloaded on each reboot till I was able to clear it.

 

### Version

3.3.0b5

### Active scan enabled

yes

### Include filter

iBeacon:

### Firmware flavor

standard 

### Device (be specific)

esp32

### Logs

```text
1 Del    | 7a41d5440e58 | apple:1007:11-7                                           -93dBm 7a1f7ffeb43578[redacted]
1 C# -1  | dd340206dda8 | iBeacon:426c7565-4368-6172-6d42-6561636f6e73-3838-4949    -97dBm (9.55m) 30001ms
1 C# +1  | dd340206dda8 | iBeacon:426c7565-4368-6172-6d42-6561636f6e73-3838-4949    -94dBm (10.70m) 6ms
heap_caps_malloc was called but failed to allocate 20125 bytes with 0x1800 capabilities. 

abort() was called at PC 0x401c390e on core 1

Backtrace: 0x40083ca5:0x3ffe6240 0x4009672d:0x3ffe6260 0x4009a20d:0x3ffe6280 0x401c390e:0x3ffe6300 0x401a9283:0x3ffe6320 0x401a91ea:0x3ffe6340 0x401a92e5:0x3ffe6360 0x40118501:0x3ffe6380 0x40118549:0x3ffe63a0 0x400e7640:0x3ffe63c0 0x400e7668:0x3ffe63e0 0x401dde96:0x3ffe6400 0x400eaf64:0x3ffe6420 0x400eafb5:0x3ffe64b0 0x400e890d:0x3ffe64d0 0x400e710d:0x3ffe6520 0x400e71dd:0x3ffe6570 0x400e7395:0x3ffe65c0 0x401c400d:0x3ffe65e0 0x401c408d:0x3ffe6610 0x401c469e:0x3ffe6630

ELF file SHA256: 0000000000000000

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
Pre-Setup Free Mem: 212552
Setup network
Connecting to WiFi SSID 'toast'...............................................................................................................................................................192.168.4.10
Firmware:     esp32
Version:      v3.3.0b5
WiFi BSSID:   A8:5E:45:ED:F2:90 (channel=1 rssi=-41)
IP address:   192.168.4.10
DNS address:  192.168.5.1
Hostname:     espresense-ec0798
Room:         ec0798
Mqtt server:  mqtt2.localdomain:1883
Max Distance: 0.00
PIR Sensor:   disabled
Radar Sensor: disabled
Switch One: disabled
Switch Two: disabled
Button One Sensor:   disabled
Button Two Sensor: disabled
I2C Bus 1:    sda=21 scl=22
Query:        
Include:      iBeacon:
Exclude:      
Known Macs:   
Count Ids:    iBeacon:
Post-Setup Free Mem: 130256

ArduinoOTA configured and ready
1 Motion | Pir: no Radar: no
1 Advert | iBeacon
1 Switch | Switch One: no Switch Two: no
1 Button | Button One: no Button Two: no
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-172-1433 to {"id":"node:ac0599","name":"ac0599"}
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-132-25616 to {"id":"node:846410","name":"846410"}
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-236-1944 to {"id":"node:ec0798","name":"ec0798"}
1 Alias  | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-236-1944 to node:ec0798
1 New    | fb0ab10a72cf | md:0059:8                                                 -91dBm 
1 New    | 487e48368ff1 | 487e48368ff1                                              -45dBm 
1 Config | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-236-1944 to {"id":"node:ec0798","name":"ec0798"}
1 New    | b8bc5bb44d65 | b8bc5bb44d65                                              -79dBm 
1 New    | 487e48647f74 | 487e48647f74                                              -59dBm 
1 New    | 4dff7b0f47a1 | apple:1007:11-5                                           -92dBm 031fc8bd3d1c38
1 New    | 613df7dc1cc9 | apple:1006:10-8                                           -94dBm 2d1a736373c1
1 New    | 487e48631bf4 | 487e48631bf4                                              -57dBm 
1 New    | 641cae77389b | 641cae77389b                                              -89dBm 
1 New    | 487e48698a4c | 487e48698a4c                                              -51dBm 
1 New    | a4c1386b770f | a4c1386b770f                                              -77dBm 
1 New    | b83a9d8063ff | b83a9d8063ff                                              -91dBm 
1 New    | 400c0c605860 | apple:1005:9-12                                           -94dBm 031cc756a4
1 New    | cf6c4bd9125f | name:n6cm6                                                -92dBm 
1 New    | a4c138ea241a | a4c138ea241a                                              -51dBm 
1 New    | c5196ff43e79 | md:0059:8                                                 -87dBm 
1 New    | d8a35cf1304b | d8a35cf1304b                                              -97dBm 
1 New    | 4e336dceb5de | sd:0xfe9f                                                 -90dBm 
1 New    | 78e36d106486 | node:846410                                               -83dBm 
1 New    | d112be1ff597 | mifit:d112be1ff597                                        -61dBm 
1 New    | 6533ba2744d1 | apple:1006:10-7                                           -94dBm 331d04a95b58
1 New    | a4c13829a8a2 | a4c13829a8a2                                              -47dBm 
1 New    | 519ec0b13d7b | apple:1002:6-9                                            -93dBm 0100
```

davlaw, 2023-08-24:
its a esp-wroom-32 chipset

board id mini32 v1.0.0

jpettitt, 2023-08-25:
+1 also seeing crashes

davlaw, 2023-08-25:
boot-loops as others have seen, when I was able to finally log into one it had the updated url for b6

Just ended up re-flashing, but back to the original issue when I log in to the web interface it reboots

DTTerastar, 2023-11-05:
This was due to low memory and is fixed in v3.3.0 released yesterday.

closed 2023-11-05 by DTTerastar