ESPresense/ESPresense #1025: v3.3.0b6 crashes all instances

Wheemer, 2023-08-25:
### Describe the bug

Every one of my espresence are offline after what I can only assume was the auto update to v3.3.0b6

They are all in reboot loops from what I can see.

`Showing logs:
[12:20:43]Pre-Setup Free Mem: 212696
[12:20:43]Setup network
[12:21:23]Connecting to WiFi SSID 'Reid'.....................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................192.168.1.195
[12:21:23]Firmware:     esp32
[12:21:23]Version:      v3.3.0b5
[12:21:23]WiFi BSSID:   08:62:66:91:E6:20 (channel=10 rssi=-32)
[12:21:23]IP address:   192.168.1.195
[12:21:23]DNS address:  192.168.1.1
[12:21:23]Hostname:     espresense-garage
[12:21:23]Room:         Garage 
[12:21:23]Mqtt server:  192.168.1.40:1883
[12:21:23]Max Distance: 16.00
[12:21:23]PIR Sensor:   disabled
[12:21:23]Radar Sensor: disabled
[12:21:23]Switch One: disabled
[12:21:23]Switch Two: disabled
[12:21:23]Button One Sensor:   disabled
[12:21:23]Button Two Sensor: disabled
[12:21:23]DHT11 Sensor: disabled
[12:21:23]DHT22 Sensor: pin 27
[12:21:23]DHT Offset:   0.00
[12:21:23]Query:        
[12:21:23]Include:      apple:1005:9-12 irk:8a89096351d3a6091596435d01e72da1 irk:3f3f4f129c754eef9b82b52273adc2e0 iBeacon:e5ca1ade-f007-ba11-0000-000000000000-52-22879 iBeacon:e5ca1ade-f007-ba11-0000-000000000000-8-43007 iBeacon:e5ca1ade-f007-ba11-0000-000000000000-40-13115 iBeacon:e5ca1ade-f007-ba11-0000-000000000000-216-47668 iBeacon:e5ca1ade-f007-ba11-0000-000000000000-64-59414 node:foyer node:steveroom node:garage node:diningroom node:krisroom node:basement node:momroom node:livingroom
[12:21:23]Exclude:      
[12:21:23]Known Macs:   
[12:21:23]Count Ids:    apple:1005:9-12 irk:8a89096351d3a6091596435d01e72da1 irk:3f3f4f129c754eef9b82b52273adc2e0 iBeacon:e5ca1ade-f007-ba11-0000-000000000000-52-22879 iBeacon:e5ca1ade-f007-ba11-0000-000000000000-8-43007 iBeacon:e5ca1ade-f007-ba11-0000-000000000000-40-13115 iBeacon:e5ca1ade-f007-ba11-0000-000000000000-216-47668 iBeacon:e5ca1ade-f007-ba11-0000-000000000000-64-59414 node:foyer node:steveroom node:garage node:diningroom node:krisroom node:basement node:momroom node:livingroom
[12:21:23]Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception). 
[12:21:23]Debug exception reason: Stack canary watchpoint triggered (loopTask) 
[12:21:23]Core  1 register dump:
[12:21:23]PC      : 0x40096bd3  PS      : 0x00060f36  A0      : 0x80099e20  A1      : 0x3ffcef00  
[12:21:23]A2      : 0x3ffb6524  A3      : 0xb33fffff  A4      : 0x0000cdcd  A5      : 0x0000abab  
[12:21:23]A6      : 0x00060f23  A7      : 0x00060f20  A8      : 0xb33fffff  A9      : 0xffffffff  
[12:21:23]A10     : 0x00000001  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x00000080  
[12:21:23]A14     : 0x003fffff  A15     : 0x3ffe2858  SAR     : 0x00000010  EXCCAUSE: 0x00000001  
[12:21:23]EXCVADDR: 0x00000000  LBEG    : 0x40092534  LEND    : 0x4009253f  LCOUNT  : 0x00000000  
[12:21:23]
[12:21:23]
[12:21:23]Backtrace: 0x40096bd0:0x3ffcef00 0x40099e1d:0x3ffcef40 0x40099f49:0x3ffcef60 0x400840fe:0x3ffcef80 0x40084111:0x3ffcefa0 0x4008413a:0x3ffcefc0 0x4009a325:0x3ffcefe0 0x401e9fae:0x3ffcf000 0x4017c21b:0x3ffcf020 0x4017c249:0x3ffcf040 0x4017c2b9:0x3ffcf060 0x4017ad6b:0x3ffcf080 0x4017ae13:0x3ffcf0a0 0x4017ae4d:0x3ffcf0c0 0x40177004:0x3ffcf0e0 0x4017155e:0x3ffcf1e0 0x40171656:0x3ffcf250 0x401716a2:0x3ffcf300 0x40171825:0x3ffcf380 0x4016d1e9:0x3ffcf3a0 0x4016d33e:0x3ffcf550 0x40114a06:0x3ffcf5b0 0x40114429:0x3ffcf830 0x4011449e:0x3ffcf870 0x4011452e:0x3ffcf8e0 0x401ddcf5:0x3ffcf900 0x4011546e:0x3ffcf920 0x401164e3:0x3ffcf950 0x401166a7:0x3ffcf9c0 0x400d7194:0x3ffcf9e0 0x400d7323:0x3ffcfa20 0x400de837:0x3ffcfb20 0x400de906:0x3ffcfc20 0x400e0639:0x3ffcfc40 0x401193d6:0x3ffcfc70
[12:21:23]
[12:21:23]
[12:21:23]
[12:21:23]
[12:21:23]ELF file SHA256: 0000000000000000
[12:21:23]
[12:21:23]Rebooting...`

DanielBaulig, 2023-08-25:
Seems to be the same issue: https://github.com/ESPresense/ESPresense/issues/1026

DanielBaulig, 2023-08-25:
Note that the firmware version reported by the serial console dump is v3.3.0b5 for both of our reports. It is curious though that it started around the same time that v3.3.0b6 was released. Maybe the version string wasn't updated in v3.3.0b6?

Wheemer, 2023-08-25:
Flashing manually and b6 boots fine.

jpettitt, 2023-08-25:
The OTA update attempt from b5 -> b6 put all my nodes into a boot loop all my modes.  Manually re-flashing to b6 fixed it.

tbclark3, 2023-09-01:
Same experience as jpettitt.  After manually re-flashing to b6, everything is normal again.

DTTerastar, 2023-09-03:
Yeah, It was too low a stack for the main thread, sorry, but it's fixed in b6.

DTTerastar, 2023-11-03:
https://github.com/ESPresense/ESPresense-companion has a UI to ota flash to fix stuff like this in the future!

closed 2023-11-03 by DTTerastar