ESPresense/ESPresense #1027: Change in Home Assistant naming guidelines - MQTT entity name starts with the device name in your config

heyitsyang, 2023-08-28:
### Describe the bug

I haven't seen this posted as an issue.  Apologies if a duplicate

Home assistant 2023.8 has started enforcing (via warnings in logs so far) that MQTT  entity names must comply with their naming convention.  This causes ESPresense warnings to spam the log file.

Related reading:
- https://developers.home-assistant.io/blog/2023-057-21-change-naming-mqtt-entities/
- https://community.home-assistant.io/t/psa-mqtt-name-changes-in-2023-8/598099
- https://developers.home-assistant.io/blog/2022/07/10/entity_naming/#:~:text=Device%2C%20Area%2C%20and%20Entity%20names,the%20has_entity_name%20property%20to%20True%20.

### Version

3.2.4

### Logs

```text
Logger: homeassistant.components.mqtt.mixins
Source: components/mqtt/mixins.py:294
Integration: MQTT (documentation, issues)
First occurred: 5:06:36 PM (95 occurrences)
Last logged: 5:06:38 PM

    MQTT entity name starts with the device name in your config {'device': {'identifiers': ['espresense_605be6'], 'connections': [['mac', '94:B9:7E:E6:5B:60']], 'name': 'ESPresense Office', 'suggested_area': 'Office', 'sw_version': 'v3.2.4', 'manufacturer': 'ESPresense (esp32)', 'configuration_url': 'http://10.27.0.103', 'model': 'ESP32-D0WDQ6'}, 'name': 'ESPresense Office Arduino OTA', 'unique_id': 'espresense_605be6_arduino_ota', 'availability_topic': 'espresense/rooms/office/status', 'state_topic': 'espresense/rooms/office/arduino_ota', 'command_topic': 'espresense/rooms/office/arduino_ota/set', 'entity_category': <EntityCategory.CONFIG: 'config'>, 'payload_off': 'OFF', 'payload_not_available': 'offline', 'payload_on': 'ON', 'retain': False, 'encoding': 'utf-8', 'enabled_by_default': True, 'payload_available': 'online', 'optimistic': False, 'qos': 0, 'availability_mode': 'latest'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Arduino OTA'
    MQTT entity name starts with the device name in your config {'device': {'identifiers': ['espresense_605be6'], 'connections': [['mac', '94:B9:7E:E6:5B:60']], 'name': 'ESPresense Office', 'suggested_area': 'Office', 'sw_version': 'v3.2.4', 'manufacturer': 'ESPresense (esp32)', 'configuration_url': 'http://10.27.0.103', 'model': 'ESP32-D0WDQ6'}, 'name': 'ESPresense Office Prerelease', 'unique_id': 'espresense_605be6_prerelease', 'availability_topic': 'espresense/rooms/office/status', 'state_topic': 'espresense/rooms/office/prerelease', 'command_topic': 'espresense/rooms/office/prerelease/set', 'entity_category': <EntityCategory.CONFIG: 'config'>, 'payload_off': 'OFF', 'payload_not_available': 'offline', 'payload_on': 'ON', 'retain': False, 'encoding': 'utf-8', 'enabled_by_default': True, 'payload_available': 'online', 'optimistic': False, 'qos': 0, 'availability_mode': 'latest'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Prerelease'
    MQTT entity name starts with the device name in your config {'device': {'identifiers': ['espresense_b41a62'], 'connections': [['mac', '7C:9E:BD:62:1A:B4']], 'name': 'ESPresense GuestBdrm', 'suggested_area': 'GuestBdrm', 'sw_version': 'v3.2.4', 'manufacturer': 'ESPresense (esp32)', 'configuration_url': 'http://10.27.0.100', 'model': 'ESP32-D0WDQ6'}, 'name': 'ESPresense GuestBdrm Auto Update', 'unique_id': 'espresense_b41a62_auto_update', 'availability_topic': 'espresense/rooms/guestbdrm/status', 'state_topic': 'espresense/rooms/guestbdrm/auto_update', 'command_topic': 'espresense/rooms/guestbdrm/auto_update/set', 'entity_category': <EntityCategory.CONFIG: 'config'>, 'payload_off': 'OFF', 'payload_not_available': 'offline', 'payload_on': 'ON', 'retain': False, 'encoding': 'utf-8', 'enabled_by_default': True, 'payload_available': 'online', 'optimistic': False, 'qos': 0, 'availability_mode': 'latest'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Auto Update'
    MQTT entity name starts with the device name in your config {'device': {'identifiers': ['espresense_b41a62'], 'connections': [['mac', '7C:9E:BD:62:1A:B4']], 'name': 'ESPresense GuestBdrm', 'suggested_area': 'GuestBdrm', 'sw_version': 'v3.2.4', 'manufacturer': 'ESPresense (esp32)', 'configuration_url': 'http://10.27.0.100', 'model': 'ESP32-D0WDQ6'}, 'name': 'ESPresense GuestBdrm Arduino OTA', 'unique_id': 'espresense_b41a62_arduino_ota', 'availability_topic': 'espresense/rooms/guestbdrm/status', 'state_topic': 'espresense/rooms/guestbdrm/arduino_ota', 'command_topic': 'espresense/rooms/guestbdrm/arduino_ota/set', 'entity_category': <EntityCategory.CONFIG: 'config'>, 'payload_off': 'OFF', 'payload_not_available': 'offline', 'payload_on': 'ON', 'retain': False, 'encoding': 'utf-8', 'enabled_by_default': True, 'payload_available': 'online', 'optimistic': False, 'qos': 0, 'availability_mode': 'latest'}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Arduino OTA'
    MQTT entity name starts with the device name in your config....
```

mandreko, 2023-09-06:
I believe this is a duplicate of https://github.com/ESPresense/ESPresense/issues/997

heyitsyang, 2023-09-06:
Yep, a duplicate.  Thanks.

closed 2023-09-06 by heyitsyang