ESPresense/ESPresense #1029: Can't compile source due to read only variable being changed

blacknell, 2023-08-29:
### Describe the bug

When compiling ESPresense (latest release source) I'm getitng compile errors. However, they seem to be on included libraries.

```
Compiling .pio/build/esp32/lib2cf/NimBLE-Arduino/NimBLEServer.cpp.o
.pio/libdeps/esp32/AsyncWiFiSettings/src/AsyncWiFiSettings.cpp: In lambda function:
.pio/libdeps/esp32/AsyncWiFiSettings/src/AsyncWiFiSettings.cpp:410:68: error: assignment of read-only variable 'num_networks'
             if (num_networks < 0) num_networks = WiFi.scanNetworks();
                                                                    ^
.pio/libdeps/esp32/AsyncWiFiSettings/src/AsyncWiFiSettings.cpp: In lambda function:
.pio/libdeps/esp32/AsyncWiFiSettings/src/AsyncWiFiSettings.cpp:523:42: error: assignment of read-only variable 'num_networks'
         num_networks = WiFi.scanNetworks();
                                          ^
.pio/libdeps/esp32/AsyncWiFiSettings/src/AsyncWiFiSettings.cpp: In member function 'void AsyncWiFiSettingsClass::portal()':
.pio/libdeps/esp32/AsyncWiFiSettings/src/AsyncWiFiSettings.cpp:569:40: error: assignment of read-only variable 'desired'
             desired = onPortalWaitLoop();
                                        ^
.pio/libdeps/esp32/AsyncWiFiSettings/src/AsyncWiFiSettings.cpp: At global scope:
.pio/libdeps/esp32/AsyncWiFiSettings/src/AsyncWiFiSettings.cpp:40:12: warning: 'String {anonymous}::pwgen()' defined but not used [-Wunused-function]
     String pwgen() {
            ^~~~~
```

`num_networks` is indeed declared `static int const num_networks = -1;` in the class method scope.

So I don't understand how this would ever compile

I just cloned the source, opened with PlatformIO and clicked build on ESP32

travnewmatic, 2023-09-01:
i am also running into this

>I just cloned the source, opened with PlatformIO and clicked build on ESP32

same experience on my end

Version
No response

Active scan enabled
No response

Include filter
No response

Exclude filter
No response

Query filter
No response

Firmware flavor
No response

Device (be specific)
No response

Logs
No response

Screenshots
No response

travnewmatic, 2023-09-01:
It was introduced after this tag https://github.com/ESPresense/ESPresense/releases/tag/v3.3.0b4

v3.3.0b4 does compile successfully without any modification

blacknell, 2023-09-01:
My issues are with release not dev

blacknell, 2023-09-20:
this seems to have been fixed