ESPresense/ESPresense #1031: IRK setup

nmcrae85, 2023-09-01:
### Describe the bug

Am i doing something wrong when trying to add the device to HA yaml file with something like this? I used the UI to get the IRK as per example below (not the irk is changed for the purpose of the post).

Just seeing not home...

  - platform: mqtt_room
    device_id: "irk:e46c37a106468d1376c58d33881ee47a"
    name: 'Me BLE'
    state_topic: 'espresense/devices'
    timeout: 5
    away_timeout: 20 # number of seconds after which the enitity will get status not_home

goedh452, 2023-09-08:
The state_topic should be "espresense/rooms". At least, that works for me.

nickelnine, 2023-09-14:
Also, setting 'devce_id' to the nickname was the trick for me.

hdprice, 2023-10-26:
> Has that resolved for you? I have the same problem, my IRK is displayed in MQTT Explorer and configured that way:
> 
> > platform: mqtt_room
> > device_id: "irk:xxxxxxxxxxxxxxxxxxxxxxxxx"
> > name: '(BLE) iPhone Marco'
> > state_topic: 'espresense/rooms'
> > timeout: 5
> > away_timeout: 120
> 
> Apple Watch and girlfriend's iPhone (same config) works. But my iPhone is always shown as not_home :/
> 
> > 
> 
> platform: mqtt_room device_id: "irk:xxxxxxxxxxxxxxxxxxxxxxxxxx" name: '(BLE) iPhone Marion ' state_topic: 'espresense/rooms' timeout: 5 away_timeout: 120
> 
> Earlier it had worked.

I had this same problem. it all worked fine with just one device being tracked but after adding a second, that only showed as "not home". I fixed it by changing the state topic to "espresense/devices/irk:xxxxxx" in each of the sensor declarations for the watches in sensors.yaml.

closed 2023-11-04 by DTTerastar

meetbryce, 2024-01-11:
I was able to get it working by doing what @hdprice suggested!

slashinpdx, 2024-09-24:
Was able to get IRK of iPhone 16 Pro using the method from [here](https://github.com/fryefryefrye/Decoding-Random-Bluetooth-Address/tree/master?tab=readme-ov-file) using a spare wroom32 and Bluetooth Terminal app (from app store).

clipse2004, 2024-09-24:
I'm giving it up soon. Until now, I had to delete the entities and have them recreated by HA, which would be fine in itself. But as of yesterday, no matter what I do, I can't create the entities anymore. 
Homeassistant created me of 4 pieces always only one though same config etc.

dxmnkd316, 2024-09-24:
The entities are created through MQTT though right?  Not ESPresense.  ESPresense just creates topics and your broker handles the rest I thoguht.  You also need to make sure topics are retained.  That's really important or they won't survive.

Not to ask a dumb question, but are you using the actual IRK and not XXXXXXX?

clipse2004, 2024-09-26:
> The entities are created through MQTT though right? Not ESPresense. ESPresense just creates topics and your broker handles the rest I thoguht. You also need to make sure topics are retained. That's [...]

Hi, well the espresense published the message in mqtt and in homeassistant the sensor is created for it.
No matter how I do it, homeassistant only creates one sensor. Oh, and to your question, of course, instead of the XXXXX, there's the right number, but I'm doing it now with the enroll function. 
Then the config looks like this: 

` 
- platform: mqtt_room
   device_id: "esp:marcos-iphone"
   unique_id: marcos_iphone
   name: 'Marcos iPhone'
   state_topic: 'espresense/devices/esp:marcos-iphone' 
   timeout: 10
   away_timeout: 120
`
My problem is, I control the heating by being in the room. It sucks if the sensors don't work.