ESPresense/ESPresense #1035: All my devices are reporting as not_home since updating to v3.3.0b8

DanielBaulig, 2023-09-05:
### Describe the bug

All my devices (all Apple IRK devices) are reporting as not_home since the base stations updated to v3.3.0b8. It seems like stuff is still being written to espresense MQTT topics, but Home Assistant just reports all the devices as not_home.

### Version

v3.3.0b8

### Active scan enabled

n/a

### Include filter

n/a

### Exclude filter

n/a

### Query filter

n/a

DanielBaulig, 2023-09-05:
Seems like there was a breaking change in how the reporting to Home Assistant works:
https://github.com/ESPresense/ESPresense/releases/tag/v3.3.0b8

This here describes the new format: 
https://espresense.com/home_assistant#beacons-configurationyaml

DanielBaulig, 2023-09-05:
Updated the state_topics and it appears to be working again.

closed 2023-09-05 by DanielBaulig

harphere, 2023-11-20:
I was pulling my remaining hair out trying to figure out why HA was not receiving updates from MQTT. Thanks for posting this.