ESPresense/ESPresense #1036: Firmware install failing with "Failed to download manifest"

wtangofoxtrot, 2023-09-06:
### Describe the bug

Attempting to flash an esp32 using https://espresense.com/firmware however the version tab does not show any available versions and when I click connect it gives the error "Failed to download manifest". 

### Version

N/A

### Active scan enabled

N/A

### Include filter

N/A

### Exclude filter

N/A

### Query filter

N/A

### Firmware flavor

N/A

### Device (be specific)

N/A

### Logs

```text
N/A
```

### Screenshots

[image]

mralanlee, 2023-09-19:
I'm getting the same thing, it looks like Github is rate limiting the request.

```json
{"message":"API rate limit exceeded for <ip_here>. (But here's the good news: Authenticated requests get a higher rate limit. Check out the documentation for more details.)","documentation_url":"https://docs.github.com/rest/overview/resources-in-the-rest-api#rate-limiting"}
```

daviddesmet, 2023-10-12:
The same issue over here, I'm seen a 500 in the console log

[image]

[image]

goldstar19, 2023-10-20:
Same issue I am having.

twojciac, 2023-10-20:
I'm having the same problem as well.  I was able to load the initial ESPHome firmware just fine using web.esphome.io and I'm still having problems using the ESPresence web loader on the same ESP32.

DTTerastar, 2023-10-31:
Dang it, I thought I had cloudflare caching so this wouldn't happen...