ESPresense/ESPresense #1041: The /ui page now works, but navigating to /ui/#/devices still causes a core dump

tcreswick, 2023-09-11:
### Describe the bug

Previously, any access to the /ui page would result in a core dump on my M5Stamp C3 Mate.

Now, the /ui page works (including settings and fingerprints), however the /devices URL causes a core dump:

Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

```
Core  0 register dump:
MEPC    : 0x42037c12  RA      : 0x42037c70  SP      : 0x3fcb2850  GP      : 0x3fc8ee00  
TP      : 0x3fc8bad8  T0      : 0x40389cd4  T1      : 0xfffffff0  T2      : 0xffffffff  
S0/FP   : 0x3fcc439c  S1      : 0x00000002  A0      : 0x00000001  A1      : 0xb7b1dd91  
A2      : 0x3fcc0338  A3      : 0x00000001  A4      : 0x3fc9a43c  A5      : 0x600c2000  
A6      : 0x42037920  A7      : 0x00000003  S2      : 0x3fcc4dd5  S3      : 0x3fcbf8e0  
S4      : 0x00000002  S5      : 0x3c13d000  S6      : 0x3fcc43e4  S7      : 0x3c122000  
S8      : 0x3c127000  S9      : 0x3fcc4448  S10     : 0x3fcc43ac  S11     : 0x3fc9b314  
T3      : 0x00000000  T4      : 0x73726564  T5      : 0x6165482d  T6      : 0x776f6c6c  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0xb7b1dda0  
MHARTID : 0x00000000  

Stack memory:
3fcb2850: 0xffffffff 0x00000004 0x00000001 0x600c0000 0xfa000000 0x00000003 0xffffffff 0x00000000
3fcb2870: 0x00000000 0x00000000 0x00000000 0x0300f000 0x00000000 0x00000000 0x00000000 0x03cc4334
3fcb2890: 0x00000000 0x00000000 0x00000000 0x006f6c6c 0x00000000 0x00000000 0x00000000 0x0438a132
3fcb28b0: 0x3fc9a000 0x3fca3ae8 0x3fc918bc 0x3fcc4334 0x3fcc4320 0x00000001 0x00000000 0x4200f000
3fcb28d0: 0x3c122000 0x3fcc4ba0 0x05266000 0x3c12017c 0x3fc9a000 0x3fca3ae8 0x3fcc42c8 0x420114e4
3fcb28f0: 0x42011368 0x40389c7e 0x00000000 0x3fcc4390 0x3fc8bad8 0x3fca3ae8 0x3fcc439c 0x3fcc42c8
3fcb2910: 0x3fcc4bee 0x000001e9 0x3fcc381c 0x00000006 0x0000013b 0x00000000 0x00000000 0x00000000
3fcb2930: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2950: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40389c7e
3fcb2970: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb2990: 0xa5a5a5a5 0x00000154 0x3fcb27c0 0x3fc9b314 0x3fc952f4 0x3fc952f4 0x3fcb2998 0x3fc952ec
3fcb29b0: 0x00000016 0x3fcb08e8 0x3fcb08e8 0x3fcb2998 0x00000000 0x00000003 0x3fcb0994 0x6e797361
3fcb29d0: 0x63745f63 0x6e790070 0x00745f63 0x00000000 0x3fcb2990 0x00000003 0x00000000 0x3fcc3938
3fcb29f0: 0x42056df4 0x00000000 0x3fc9bc2c 0x3fc9bc94 0x3fc9bcfc 0x00000000 0x00000000 0x00000001
3fcb2a10: 0x00000000 0x00000000 0x00000000 0x4210390e 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2a30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2a50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2a70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2a90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2ab0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2ad0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3f000000 0x00000018
3fcb2af0: 0x3fc9b314 0x3fc9b314 0x3fcb2ae8 0x00000000 0x00000000 0x00000000 0x0000001c 0x00000000
3fcb2b10: 0x3fc9b314 0x00000000 0x00000000 0x00000000 0xff000000 0x3fcb2b04 0x00000040 0x3fc919bc
3fcb2b30: 0x00000000 0x00000000 0x3fcb2b70 0x3fcb2b80 0x00000000 0x00000000 0x00000000 0x0218ffff
3fcb2b50: 0x00000000 0x3fcb2ae8 0x00000067 0x3fc9b314 0x3fc9b314 0x3fcb2b00 0x3fcb2b00 0x0000000c
3fcb2b70: 0x3c126b70 0x00000118 0x3fcb2b2c 0x0000000c 0x3c126b8c 0x00000000 0x3fcb2b2c 0x00000020
3fcb2b90: 0x00000000 0x00000000 0x00000000 0x3fcb0000 0x00000000 0x00000000 0x00000001 0x3fcb2b58
3fcb2bb0: 0x0000000c 0x3fcb3a80 0x00000000 0x3fcb2bac 0x0000000c 0x3fcb2bd4 0x3fc9b314 0x3fcb2bbc
3fcb2bd0: 0x00000054 0x3fcb2bd4 0x3fcb2bd4 0x3fcb2bd4 0x3fcb2bd4 0x00000000 0x3fcb2bec 0xffffffff
3fcb2bf0: 0x3fcb2bec 0x3fcb2bec 0x00000000 0x3fcb2c00 0xffffffff 0x3fcb2c00 0x3fcb2c00 0x00000000
3fcb2c10: 0x00000001 0x00000000 0x0000ffff 0x00000000 0xb33fffff 0x00000000 0x0000000c 0x00000003
3fcb2c30: 0x3fcb2bc4 0x00000000 0x0000004c 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 0000000000000000

Rebooting...
```

### Version

All versions including v3.3.0b8

### Logs

```text
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42037c12  RA      : 0x42037c70  SP      : 0x3fcb2850  GP      : 0x3fc8ee00  
TP      : 0x3fc8bad8  T0      : 0x40389cd4  T1      : 0xfffffff0  T2      : 0xffffffff  
S0/FP   : 0x3fcc439c  S1      : 0x00000002  A0      : 0x00000001  A1      : 0xb7b1dd91  
A2      : 0x3fcc0338  A3      : 0x00000001  A4      : 0x3fc9a43c  A5      : 0x600c2000  
A6      : 0x42037920  A7      : 0x00000003  S2      : 0x3fcc4dd5  S3      : 0x3fcbf8e0  
S4      : 0x00000002  S5      : 0x3c13d000  S6      : 0x3fcc43e4  S7      : 0x3c122000  
S8      : 0x3c127000  S9      : 0x3fcc4448  S10     : 0x3fcc43ac  S11     : 0x3fc9b314  
T3      : 0x00000000  T4      : 0x73726564  T5      : 0x6165482d  T6      : 0x776f6c6c  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0xb7b1dda0  
MHARTID : 0x00000000  

Stack memory:
3fcb2850: 0xffffffff 0x00000004 0x00000001 0x600c0000 0xfa000000 0x00000003 0xffffffff 0x00000000
3fcb2870: 0x00000000 0x00000000 0x00000000 0x0300f000 0x00000000 0x00000000 0x00000000 0x03cc4334
3fcb2890: 0x00000000 0x00000000 0x00000000 0x006f6c6c 0x00000000 0x00000000 0x00000000 0x0438a132
3fcb28b0: 0x3fc9a000 0x3fca3ae8 0x3fc918bc 0x3fcc4334 0x3fcc4320 0x00000001 0x00000000 0x4200f000
3fcb28d0: 0x3c122000 0x3fcc4ba0 0x05266000 0x3c12017c 0x3fc9a000 0x3fca3ae8 0x3fcc42c8 0x420114e4
3fcb28f0: 0x42011368 0x40389c7e 0x00000000 0x3fcc4390 0x3fc8bad8 0x3fca3ae8 0x3fcc439c 0x3fcc42c8
3fcb2910: 0x3fcc4bee 0x000001e9 0x3fcc381c 0x00000006 0x0000013b 0x00000000 0x00000000 0x00000000
3fcb2930: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2950: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40389c7e
3fcb2970: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb2990: 0xa5a5a5a5 0x00000154 0x3fcb27c0 0x3fc9b314 0x3fc952f4 0x3fc952f4 0x3fcb2998 0x3fc952ec
3fcb29b0: 0x00000016 0x3fcb08e8 0x3fcb08e8 0x3fcb2998 0x00000000 0x00000003 0x3fcb0994 0x6e797361
3fcb29d0: 0x63745f63 0x6e790070 0x00745f63 0x00000000 0x3fcb2990 0x00000003 0x00000000 0x3fcc3938
3fcb29f0: 0x42056df4 0x00000000 0x3fc9bc2c 0x3fc9bc94 0x3fc9bcfc 0x00000000 0x00000000 0x00000001
3fcb2a10: 0x00000000 0x00000000 0x00000000 0x4210390e 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2a30: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2a50: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2a70: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2a90: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2ab0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2ad0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x3f000000 0x00000018
3fcb2af0: 0x3fc9b314 0x3fc9b314 0x3fcb2ae8 0x00000000 0x00000000 0x00000000 0x0000001c 0x00000000
3fcb2b10: 0x3fc9b314 0x00000000 0x00000000 0x00000000 0xff000000 0x3fcb2b04 0x00000040 0x3fc919bc
3fcb2b30: 0x00000000 0x00000000 0x3fcb2b70 0x3fcb2b80 0x00000000 0x00000000 0x00000000 0x0218ffff
3fcb2b50: 0x00000000 0x3fcb2ae8 0x00000067 0x3fc9b314 0x3fc9b314 0x3fcb2b00 0x3fcb2b00 0x0000000c
3fcb2b70: 0x3c126b70 0x00000118 0x3fcb2b2c 0x0000000c 0x3c126b8c 0x00000000 0x3fcb2b2c 0x00000020
3fcb2b90: 0x00000000 0x00000000 0x00000000 0x3fcb0000 0x00000000 0x00000000 0x00000001 0x3fcb2b58
3fcb2bb0: 0x0000000c 0x3fcb3a80 0x00000000 0x3fcb2bac 0x0000000c 0x3fcb2bd4 0x3fc9b314 0x3fcb2bbc
3fcb2bd0: 0x00000054 0x3fcb2bd4 0x3fcb2bd4 0x3fcb2bd4 0x3fcb2bd4 0x00000000 0x3fcb2bec 0xffffffff
3fcb2bf0: 0x3fcb2bec 0x3fcb2bec 0x00000000 0x3fcb2c00 0xffffffff 0x3fcb2c00 0x3fcb2c00 0x00000000
3fcb2c10: 0x00000001 0x00000000 0x0000ffff 0x00000000 0xb33fffff 0x00000000 0x0000000c 0x00000003
3fcb2c30: 0x3fcb2bc4 0x00000000 0x0000004c 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 0000000000000000

Rebooting...
```

gunnarbeutner, 2023-09-24:
I believe this is the same bug I encountered in https://github.com/ESPresense/ESPAsyncWebServer/pull/1.

DTTerastar, 2023-11-03:
This is fixed in b10.

closed 2023-11-03 by DTTerastar