ESPresense/ESPresense #1051: MQTT entity name starts with the device name in your config

hellcry37, 2023-09-19:
### Describe the bug

New breaking changes for MQTT in Home Assistant 2023.8

### Version

v3.2.4

### Active scan enabled

-

### Include filter

-

### Exclude filter

-

### Query filter

-

### Firmware flavor

not important

### Device (be specific)

no

### Logs

```text
Logger: homeassistant.components.mqtt.mixins
Source: components/mqtt/mixins.py:296
Integration: MQTT (documentation, issues)
First occurred: 15:15:00 (44 occurrences)
Last logged: 15:15:00

MQTT entity name starts with the device name in your config {'device': {'identifiers': ['espresense_549480'], 'connections': [['mac', '78:21:84:80:94:54']], 'name': 'ESPresense Kid Room', 'suggested_area': 'Kid Room', 'sw_version': 'v3.2.4', 'manufacturer': 'ESPresense (esp32)', 'configuration_url': 'http://192.168.1.38', 'model': 'ESP32-D0WDQ6'}, 'name': 'ESPresense Kid Room Update', 'unique_id': 'espresense_549480_update', 'availability_topic': 'espresense/rooms/kid_room/status', 'command_topic': 'espresense/rooms/kid_room/update/set', 'entity_category': <EntityCategory.DIAGNOSTIC: 'diagnostic'>, 'payload_available': 'online', 'availability_mode': 'latest', 'encoding': 'utf-8', 'payload_press': 'PRESS', 'retain': False, 'enabled_by_default': True, 'payload_not_available': 'offline', 'qos': 0}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Update'
MQTT entity name starts with the device name in your config {'device': {'identifiers': ['espresense_549480'], 'connections': [['mac', '78:21:84:80:94:54']], 'name': 'ESPresense Kid Room', 'suggested_area': 'Kid Room', 'sw_version': 'v3.2.4', 'manufacturer': 'ESPresense (esp32)', 'configuration_url': 'http://192.168.1.38', 'model': 'ESP32-D0WDQ6'}, 'name': 'ESPresense Kid Room Enroll', 'unique_id': 'espresense_549480_enroll', 'availability_topic': 'espresense/rooms/kid_room/status', 'command_topic': 'espresense/rooms/kid_room/enroll/set', 'entity_category': <EntityCategory.CONFIG: 'config'>, 'payload_available': 'online', 'availability_mode': 'latest', 'encoding': 'utf-8', 'payload_press': 'PRESS', 'retain': False, 'enabled_by_default': True, 'payload_not_available': 'offline', 'qos': 0}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Enroll'
MQTT entity name starts with the device name in your config {'device': {'identifiers': ['espresense_784e7c'], 'connections': [['mac', '78:21:84:7C:4E:78']], 'name': 'ESPresense Master Bedroom', 'suggested_area': 'Master Bedroom', 'sw_version': 'v3.2.4', 'manufacturer': 'ESPresense (esp32)', 'configuration_url': 'http://192.168.1.39', 'model': 'ESP32-D0WDQ6'}, 'name': 'ESPresense Master Bedroom Restart', 'unique_id': 'espresense_784e7c_restart', 'availability_topic': 'espresense/rooms/master_bedroom/status', 'command_topic': 'espresense/rooms/master_bedroom/restart/set', 'entity_category': <EntityCategory.DIAGNOSTIC: 'diagnostic'>, 'payload_available': 'online', 'availability_mode': 'latest', 'encoding': 'utf-8', 'payload_press': 'PRESS', 'retain': False, 'enabled_by_default': True, 'payload_not_available': 'offline', 'qos': 0}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Restart'
MQTT entity name starts with the device name in your config {'device': {'identifiers': ['espresense_784e7c'], 'connections': [['mac', '78:21:84:7C:4E:78']], 'name': 'ESPresense Master Bedroom', 'suggested_area': 'Master Bedroom', 'sw_version': 'v3.2.4', 'manufacturer': 'ESPresense (esp32)', 'configuration_url': 'http://192.168.1.39', 'model': 'ESP32-D0WDQ6'}, 'name': 'ESPresense Master Bedroom Update', 'unique_id': 'espresense_784e7c_update', 'availability_topic': 'espresense/rooms/master_bedroom/status', 'command_topic': 'espresense/rooms/master_bedroom/update/set', 'entity_category': <EntityCategory.DIAGNOSTIC: 'diagnostic'>, 'payload_available': 'online', 'availability_mode': 'latest', 'encoding': 'utf-8', 'payload_press': 'PRESS', 'retain': False, 'enabled_by_default': True, 'payload_not_available': 'offline', 'qos': 0}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Update'
MQTT entity name starts with the device name in your config {'device': {'identifiers': ['espresense_784e7c'], 'connections': [['mac', '78:21:84:7C:4E:78']], 'name': 'ESPresense Master Bedroom', 'suggested_area': 'Master Bedroom', 'sw_version': 'v3.2.4', 'manufacturer': 'ESPresense (esp32)', 'configuration_url': 'http://192.168.1.39', 'model': 'ESP32-D0WDQ6'}, 'name': 'ESPresense Master Bedroom Enroll', 'unique_id': 'espresense_784e7c_enroll', 'availability_topic': 'espresense/rooms/master_bedroom/status', 'command_topic': 'espresense/rooms/master_bedroom/enroll/set', 'entity_category': <EntityCategory.CONFIG: 'config'>, 'payload_available': 'online', 'availability_mode': 'latest', 'encoding': 'utf-8', 'payload_press': 'PRESS', 'retain': False, 'enabled_by_default': True, 'payload_not_available': 'offline', 'qos': 0}, this is not expected. Please correct your configuration. The device name prefix will be stripped off the entity name and becomes 'Enroll'
```

goedh452, 2023-09-21:
Has been fixed with #997. Waiting for the release.

hellcry37, 2023-09-21:
idd waiting, thanks

hellcry37, 2023-10-19:
this is fixed in beta, I've tested. Thank you!

closed 2023-10-19 by hellcry37

pickonedev, 2023-10-19:
I hope the release will go out soon, I hate this warning...