ESPresense/ESPresense #1054: Add device name to settings page

CoMPaTech, 2023-09-21:
### Is your feature request related to a problem? Please describe.

With the recent beta's one has to open the additional UI for the detailed settings, `/ui/#/settings` which has no identifiers set as to which device you are on. 

### Describe the solution you'd like

With multiple devices it would be nice to have the device name somewhere in the left pane

colinhatfield, 2023-11-03:
+1
Would be nice to see it in the page title

MaZZly, 2024-01-21:
+1 Was just about to file an issue about this, it would be nice if all pages on the `/ui/`-paths included the Room name in the page title.

e.g. `ESPresense - Livingroom` or `ESPresense | Livingroom`

baudneo, 2024-04-16:
Please, this is a **nightmare** when calibrating several devices. 

`ESPresense | VERSION | ROOM_NAME ` As a floating footer or even just under the ESPresense logo.

CoMPaTech, 2024-08-23:
@DTTerastar I'd guess #1352 solves this, right?

DTTerastar, 2025-03-20:
Yep, the betas all have the room name in the title

closed 2025-03-20 by DTTerastar