ESPresense/ESPresense #1063: Access device page on ui trigger crash all the time

FlyGoat, 2023-10-15:
### Version

v3.3.0b8

### Device (be specific)

ESP32-C3-MINI-1-N4 

### Logs

```text
Disconnected from MQTT; reason 0
0 New    | f5b57ab1149b | f5b57ab1149b                                              -86dBm 
0 New    | e0cac94b158c | e0cac94b158c                                              -97dBm 
Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.

Core  0 register dump:
MEPC    : 0x42037c12  RA      : 0x42037c70  SP      : 0x3fcb2760  GP      : 0x3fc8ee00  
TP      : 0x3fc8b9e8  T0      : 0x40389cd4  T1      : 0xfffffff0  T2      : 0xffffffff  
S0/FP   : 0x3fcc7b38  S1      : 0x00000002  A0      : 0x00000001  A1      : 0x90b3cbaa  
A2      : 0x3fcc4ed8  A3      : 0x00000001  A4      : 0x3fc9a43c  A5      : 0x600c2000  
A6      : 0x2e302e30  A7      : 0x2e383131  S2      : 0x3fcc7944  S3      : 0x3fca3a78  
S4      : 0x00000002  S5      : 0x3c13d000  S6      : 0x3fcc7b80  S7      : 0x3c122000  
S8      : 0x3c127000  S9      : 0x3fcc7be4  S10     : 0x3fcc7b48  S11     : 0x3fc9b314  
T3      : 0x72684320  T4      : 0x296f6b63  T5      : 0x65472065  T6      : 0x6b696c20  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0x90b3cbb9  
MHARTID : 0x00000000  

Stack memory:
3fcb2760: 0x3c122000 0x00000000 0x05266000 0x3c12017c 0x3fcc2238 0x3c13d000 0x3fcc22cc 0x42016ea0
3fcb2780: 0x00000000 0x00000000 0x00000000 0x030ac401 0x00000000 0x00000000 0x00000000 0x0300f000
3fcb27a0: 0x00000000 0x00000000 0x00000000 0x0038cf78 0x00000000 0x00000000 0x00000000 0x0638a132
3fcb27c0: 0x3fc9a000 0x3fcc230c 0x3fc918bc 0x3fcc22a4 0x3fcc2290 0x00000001 0x00000000 0x4200f000
3fcb27e0: 0x3c122000 0x3fcc76e0 0x05266000 0x3c12017c 0x3fc9a000 0x3fcc2da0 0x3fcc2238 0x420114e4
3fcb2800: 0x42011368 0x40389c7e 0x00000000 0x3fcc2b38 0x3fc8b9e8 0x3fcc2da0 0x3fcc7b38 0x3fcc2238
3fcb2820: 0x3fcc772e 0x00000218 0x3fcc28d4 0x0000000d 0x00000159 0x00000000 0x00000000 0x00000000
3fcb2840: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2860: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x40389c7e
3fcb2880: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fcb28a0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000154 0x3fcb2630 0x3fc9b314 0x3fc952f4
3fcb28c0: 0x3fc952f4 0x3fcb28b4 0x3fc952ec 0x00000016 0x3fcb0804 0x3fcb0804 0x3fcb28b4 0x00000000
3fcb28e0: 0x00000003 0x3fcb08b0 0x6e797361 0x63745f63 0x00000070 0x00000000 0x00000000 0x3fcb28a0
3fcb2900: 0x00000003 0x00000000 0x3fcc288c 0x42056df4 0x00000000 0x3fc9bc2c 0x3fc9bc94 0x3fc9bcfc
3fcb2920: 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x4210390e 0x00000000
3fcb2940: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2960: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2980: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb29a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb29c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb29e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcb2a00: 0x00000000 0x3f000000 0x00000018 0x3fc9b314 0x3fc9b314 0x3fcb2a04 0x00000000 0x00000000
3fcb2a20: 0x00000000 0x0000001c 0x00000000 0x3fc9b314 0x00000000 0x00000000 0x00000000 0xff000000
3fcb2a40: 0x3fcb2a20 0x00000040 0x3fc919bc 0x00000000 0x00000000 0x3fcb2a8c 0x3fcb2a9c 0x00000000
3fcb2a60: 0x00000000 0x00000000 0x0218ffff 0x00000000 0x3fcb2a04 0x00000067 0x3fc9b314 0x3fc9b314
3fcb2a80: 0x3fcb2a00 0x3fcb2a00 0x0000000c 0x3c126b70 0x00000118 0x3fcb2a48 0x0000000c 0x3c126b8c
3fcb2aa0: 0x00000000 0x3fcb2a48 0x00000020 0x00000000 0x00000000 0x00000000 0x3fcb0000 0x00000000
3fcb2ac0: 0x00000000 0x00000001 0x3fcb2a58 0x0000000c 0x3fcb399c 0x00000000 0x3fcb2ac8 0x0000000c
3fcb2ae0: 0x3fcb2af0 0x3fc9b314 0x3fcb2ad8 0x00000054 0x3fcb2af0 0x3fcb2af0 0x3fcb2af0 0x3fcb2af0
3fcb2b00: 0x00000000 0x3fcb2b08 0xffffffff 0x3fcb2b08 0x3fcb2b08 0x00000000 0x3fcb2b1c 0xffffffff
3fcb2b20: 0x3fcb2b1c 0x3fcb2b1c 0x00000000 0x00000001 0x00000000 0x0000ffff 0x00000000 0xb33fffff
3fcb2b40: 0x00000000 0x0000000c 0x00000003 0x3fcb2ae0 0x00000000 0x0000004c 0x00000000 0x00000000

ELF file SHA256: 0000000000000000

```
```

FlyGoat, 2023-10-15:
Detailed backtrace:

```
11:10:13.605 > Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
11:10:13.662 >
11:10:13.662 > Core  0 register dump:
11:10:13.665 > MEPC    : 0x42018418  RA      : 0x4201846c  SP      : 0x3fcb2ea0  GP      : 0x3fc8ee00  
11:10:13.721 > => 0x42018418: String::buffer() const at C:/Users/flygoat/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:339 (discriminator 2)
11:10:13.721 >       (inlined by) String::c_str() const at C:/Users/flygoat/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.h:248 (discriminator 2)
11:10:13.721 >       (inlined by) AsyncWebServerRequest::_removeNotInterestingHeaders() at .pio/libdeps/esp32c3/ESP Async WebServer/src/WebRequest.cpp:184 (discriminator 2)
11:10:13.721 > => 0x4201846c: LinkedList<AsyncWebHeader*, LinkedListNode>::Iterator::operator++() at .pio/libdeps/esp32c3/ESP Async WebServer/src/StringArray.h:53 (discriminator 3)
11:10:13.721 >       (inlined by) AsyncWebServerRequest::_removeNotInterestingHeaders() at .pio/libdeps/esp32c3/ESP Async WebServer/src/WebRequest.cpp:183 (discriminator 3)
11:10:13.721 > TP      : 0x3fc8bb04  T0      : 0x40389d46  T1      : 0xfffffff0  T2      : 0xffffffff  
11:10:13.752 > => 0x40389d46: vPortExitCritical at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port/riscv/port.c:259
11:10:13.752 > S0/FP   : 0x3fcc5298  S1      : 0x3fcc6de0  A0      : 0x00000001  A1      : 0xaaca3123     
11:10:13.752 > A2      : 0x3fcc6fc8  A3      : 0x00000001  A4      : 0x3fc9a88c  A5      : 0x600c2000     
11:10:13.752 > A6      : 0x2e302e30  A7      : 0x2e383131  S2      : 0x3fc9ba44  S3      : 0x3fcc6df0     
11:10:13.752 > S4      : 0x3fcc6e8c  S5      : 0x00000001  S6      : 0x0000000a  S7      : 0x00000004     
11:10:13.752 > S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000     
11:10:13.752 > T3      : 0x72684320  T4      : 0x296f6b63  T5      : 0x65472065  T6      : 0x6b696c20  
11:10:13.794 > MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000005  MTVAL   : 0xaaca3132     
11:10:13.794 > => 0x40380001: _vector_table at ??:?
11:10:13.794 > MHARTID : 0x00000000
11:10:13.794 >
11:10:13.794 > Stack memory:
11:10:13.794 > 3fcb2ea0: 0x00000000 0x00000000 0x00000000 0x0401a6d8 0x0000000a 0x00000001 0x3fcc6e14 0x00000002
11:10:13.794 > 3fcb2ec0: 0x3fcc6c60 0x00000002 0x3fcc6de0 0x420184ee 0x3fcc6c60 0x00000002 0x3fcc6e14 0x42050124
11:10:13.999 > => 0x420184ee: AsyncWebServerRequest::_parseLine() at .pio/libdeps/esp32c3/ESP Async WebServer/src/WebRequest.cpp:572
11:10:13.999 >       (inlined by) AsyncWebServerRequest::_parseLine() at .pio/libdeps/esp32c3/ESP Async WebServer/src/WebRequest.cpp:554
11:10:13.999 > => 0x42050124: String::concat(char const*) at C:/Users/flygoat/.platformio/packages/framework-arduinoespressif32/cores/esp32/WString.cpp:342
11:10:13.999 > 3fcb2ee0: 0x0000000a 0x00000001 0x3fcc6e14 0x00000002 0x3fcc6c60 0x00000002 0x3fcc6de0 0x420185ec
11:10:13.999 > => 0x420185ec: AsyncWebServerRequest::_onData(void*, unsigned int) at .pio/libdeps/esp32c3/ESP Async WebServer/src/WebRequest.cpp:124
11:10:13.999 > 3fcb2f00: 0x3fcc7034 0x3c14c000 0x3fcc6de0 0x420183b2 0x00000001 0x3fc9215c 0x420167a4 0x42016914
11:10:13.999 > => 0x420183b2: AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*) at .pio/libdeps/esp32c3/ESP Async WebServer/src/WebRequest.cpp:78
11:10:13.999 > => 0x420167a4: std::_Function_base::_Base_manager<AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*)#9}>::_M_manager(std::_Any_data&, std::_Function_base::_Base_manager<AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*)#9}> const&, std::_Manager_operation) at c:\users\flygoat\.platformio\packages\toolchain-riscv32-esp\riscv32-esp-elf\include\c++\8.4.0\bits/std_function.h:196
11:10:13.999 > => 0x42016914: std::_Function_handler<void (void*, AsyncClient*), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*)#9}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&) at c:\users\flygoat\.platformio\packages\toolchain-riscv32-esp\riscv32-esp-elf\include\c++\8.4.0\bits/std_function.h:295
11:10:13.999 > 3fcb2f20: 0x00000002 0x3fcc70a0 0x3fcc708c 0x00000001 0x00000000 0x3fcc69fc 0x3fcc7034 0x420187bc
11:10:13.999 > => 0x420187bc: std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#8}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at c:\users\flygoat\.platformio\packages\toolchain-riscv32-esp\riscv32-esp-elf\include\c++\8.4.0\bits/std_function.h:299
11:10:13.999 > 3fcb2f40: 0x00000000 0x3fcc69fc 0x3fcc7034 0x42104228 0x3fcc6de0 0x3fcc7034 0x3fcc6a4a 0x00000218
11:10:13.999 > => 0x42104228: AsyncClient::_recv(tcp_pcb*, pbuf*, signed char) at .pio/libdeps/esp32c3/AsyncTCP@src-39de97abf7348c44d4dda815b8aab0ae/src/AsyncTCP.cpp:970
11:10:13.999 > 3fcb2f60: 0x00000002 0x00000005 0x3fc9b000 0x3fc9aca8 0x00000001 0x3fcc7034 0x3fcc0578 0x421042cc
[... cut ...]
11:10:14.178 > 3fcb3200: 0x3fcb0000 0x00000000 0x00000000 0x00000001 0x3fcb3158 0x0000000c 0x3fcb40e4 0x00000000
11:10:14.178 > 3fcb3220: 0x3fcb3210 0x0000000c 0x3fcb3238 0x3fc9ba44 0x3fcb3220 0x00000054 0x3fcb3238 0x3fcb3238
11:10:14.178 > 3fcb3240: 0x3fcb3238 0x3fcb3238 0x00000000 0x3fcb3250 0xffffffff 0x3fcb3250 0x3fcb3250 0x00000000
11:10:14.178 > 3fcb3260: 0x3fcb3264 0xffffffff 0x3fcb3264 0x3fcb3264 0x00000000 0x00000001 0x00000000 0x4400ffff
11:10:14.256 > => 0x4400ffff: ?? ??:0
11:10:14.256 > 3fcb3280: 0x00000000 0xb33fffff 0x00000000 0x0000000c 0x00000003 0x3fcb3228 0x00000000 0x0000004c
```

FlyGoat, 2023-10-15:
It's a duplication of #1013

FlyGoat, 2023-10-15:
Can confirm everything works now by cherry-picking https://github.com/me-no-dev/ESPAsyncWebServer/pull/952 and https://github.com/me-no-dev/ESPAsyncWebServer/pull/953 into ESPPresence's ESPAsyncWebServer tree.

DTTerastar, 2023-11-03:
Fixed in [v3.3.0b10](https://github.com/ESPresense/ESPresense/releases/tag/v3.3.0b10)

closed 2023-11-03 by DTTerastar