ESPresense/ESPresense #1066: HA Integration: Exclude distance attribute from recorder

GoNzCiD, 2023-10-22:
### Is your feature request related to a problem? Please describe.

Since the sensor is reporting the distance a lot of times in a short period of time, the HA recorder is writing down a new database entry.
In a long period of time, the database contains a lot of duplicates entries with the same `status` (home/not_home per ESPresence device) because of the distance.

In the end, a lot of size used in the database.

### Describe the solution you'd like

Not to write down duplicates entries in the HA database. Maybe it could be an optional choice.

### Describe alternatives you've considered

Regarding the [documentation](https://developers.home-assistant.io/docs/core/entity/#excluding-state-attributes-from-recorder-history), there is an alternative to send attributes to HA but flag it to exclude for the recorder.

DTTerastar, 2023-11-03:
This just sends mqtt messages, to resolve this (which I agree is annoying) you'd need to take that up w/ home assistant itself.  It really could use some debouncing / deduping to write less junk to the db.

closed 2023-11-03 by DTTerastar

GoNzCiD, 2023-11-04:
Oh! My bad @DTTerastar, this repo is not a HA built-in integration... 🤦. It uses the MQTT discovery & push messages, I know.

Then I can see two different actions:
  * For the device's distance should be in the `mqtt_room`, I guess: https://github.com/home-assistant/core/blob/dev/homeassistant/components/mqtt_room/sensor.py
  * For this repo, I guess it's needed to review the discovery topics published, so HA will ignore these values.
  
For instance, I can see this discovery topic `homeassistant/binary_sensor/espresense_XXXXXX/connectivity/config`, even if I disable the telemetry topic via UI and reboot   
[image]

This topic contains the definition for discovery, and I can see:
```
{
    "dev": {
        "ids": [
            "espresense_XXXXXX"
        ],
        "cns": [...
        ],
        "name": "ESPresense Test",
        "sa": "Test",
        ...
    },
    "~": "espresense/rooms/test",
    "name": "ESPresense Test",
    "uniq_id": "espresense_XXXXXX_connectivity",
    "json_attr_t": "~/telemetry",
    "stat_t": "~/status",
    "dev_cla": "connectivity",
    "pl_on": "online",
    "pl_off": "offline"
}
```
Pay attention to the `json_attr_t`, it points to the entire telemetry which has an uptime, memory, RSSI strength.... so a lot of attributes that **will change in each MQTT telemetry topic published**. Maybe [here](https://github.com/ESPresense/ESPresense/blob/master/src/mqtt.cpp#L44)

The expected result from my point of view is that once unchecked the `Send to telemetry topic` in the UI below the discovery, this telemetry attributes will not be published in the discovery topic. So it's safe to remove the device from HA, restart the ESPresense device and no more attributes rubbish in the HA database once the discovery reads it again.

*Not sure if there are other topics like this*
*Maybe the best solution is to exclude only the dynamic attributes from the topic, not the entire topic as the fastest/easiest one.*

Do you think it's doable?

DTTerastar, 2023-11-04:
Discovery is retained so you need to disable the Send to discovery topic (default: ☑) delete it in mqtt (via mqtt explorer) and then it shouldn't come back on reboot

DAVIZINH0, 2024-03-01:
sorry for coment in a closed isue but this is very interesting for me. @GoNzCiD 
i have the same problem. My 6 espresense are the 10% of the recorder size. is crazy. :-)

Do you solve this? or find some idea to fix this?, the distance is not inportant for me too

GoNzCiD, 2024-03-01:
I haven't solved it. But I make some cleanup task to erase content older than X in the database as a workaround.

DAVIZINH0, 2024-03-01:
> I haven't solved it @DAVIZINH0. But I make some cleanup task to erase content older than X in the database as a workaround.

same solution in my case. 
It will be great if can configure some in espresense that report only the room and only when the room change.
I open a disscussion with this topic i you want to see it
[https://github.com/ESPresense/ESPresense/discussions/1244](url)