ESPresense/ESPresense #1081: Led not keep last status

18rrs, 2023-11-04:
### Describe the bug

I use a led 1 as PWM (MQTT control) and each time i restart or power outage, the led turns on after boot. I thing the lod status must be the las know of default off

### Version

v3.2.4

### Device (be specific)

esp32c

Kuifje88, 2024-07-17:
I have a m5 Atom lite and use pin 33 to control a LED ring as a nightlight. I have the same problem, when the esp32 restarts, the light turns on.

DTTerastar, 2026-11-06:
@18rrs @Kuifje88 this is fixed again on the current firmware: an LED set to `control: mqtt` now remembers whether it was on or off across a reboot or power cut. The earlier fix (#2081) was for the old Arduino firmware that has since been replaced, so it was rewritten in #2646.

Could one of you try it before it's merged? https://espresense.com/alpha/?pr=2646

1. Flash that build on one node (it's v5, so settings, MQTT topics and HA entities differ from v4; you can go back by reflashing with the normal installer).
2. Turn the LED off from Home Assistant and wait about 15 seconds (the state is saved every 15 s).
3. Restart the node or pull the power. The LED should stay off, and HA should show it off.

Let us know if it does, or if it still comes back on.

closed 2026-11-07 by DTTerastar