ESPresense/ESPresense #1090: Update from 3.3.0 to 3.3.1 exits with UNKNOWN ERROR CODE

N3rdix, 2023-11-06:
### Describe the bug

This night one of my ESP32 updated first to 3.3.0 and then later it tried to auto-update to 3.3.1 and didn't come back online.
The UI is available a few seconds before it dies, so I checked the logs in the console when connected via USB (see below). Once it runs into this error it can even not be pinged any longer, but still display some new discoveries in the log via USB...

Besides re-flashing is there anything I can do? If this is an unexpected bug I am happy to provide more information.

### Version

3.3.1

### Include filter

some irks

### Query filter

some irks

### Firmware flavor

esp32

### Logs

```text
.............................................................10.10.30.11
Firmware:     esp32
Version:      v3.3.0
WiFi BSSID:   B2:8B:A9:1E:03:AE (channel=11 rssi=-39)
IP address:   10.10.30.11
DNS address:  192.168.1.2
Hostname:     espresense-hobbyraum
Room:         Hobbyraum
Mqtt server:  10.10.10.10:1883
Max Distance: 16.00
PIR Sensor:   disabled
Radar Sensor: disabled
Switch One:   disabled
Switch Two:   disabled
Button One:   disabled
Button Two:   disabled
I2C Bus 1:    sda=21 scl=22
Query:        
Include:      irk:123123123123123123                      
Exclude:      
Known Macs:   
Count Ids:    
1 Update | started
1 Update | 0%
1 Update | 0%
1 Update | 0%
1 Update | 0%
1 Update | 0%
1 Update | 1%
1 Update | 1%
1 Update | 1%
1 Update | 1%
1 Update | 2%
1 Update | 2%
1 Update | 2%
1 Update | 3%
1 Update | 3%
1 Update | 3%
1 Update | 3%
1 Update | 4%
1 Update | 4%
1 Update | 4%
1 Update | 5%
1 Update | 5%
1 Update | 5%
1 Update | 5%
1 Update | 6%
1 Update | 6%
1 Update | 6%
1 Update | 6%
1 Update | 7%
1 Update | 7%
1 Update | 7%
1 Update | 8%
1 Update | 8%
1 Update | 8%
1 Update | 8%
1 Update | 9%
1 Update | 9%
[ 12033][E][ssl_client.cpp:37] _handle_error(): [data_to_read():361]: (-76) UNKNOWN ERROR CODE (004C)
Update.writeStream failed! (Stream Read Timeout)
1 Update | finished
Http Update Failed (Error=6): Update error: Stream Read Timeout
Post-Setup Free Mem: 146260

1 Motion | Pir: no Radar: no
1 Advert | iBeacon
1 New    | 3c610579ae86 | 3c610579ae86                                              -77dBm 
1 New    | ef5cda1543f5 | ef5cda1543f5                                              -73dBm 
1 Switch | Switch One: no Switch Two: no
1 Close  | 1695ac3ec2cd | msft:cdp:0922                                             -33dBm
1 New    | 1695ac3ec2cd | msft:cdp:0922                                             -33dBm 
1 New    | 1c90ff1fc4bc | 1c90ff1fc4bc                                              -71dBm 
1 Button | Button One: no Button Two: no
1 New    | 678941902ed0 | irk:123123123123123123                      -69dBm 3a1ee9c42d15
1 New    | 7ec2ab26ac74 | irk:123123123123123123                      -70dBm 2198187f9c
1 New    | de5964536371 | de5964536371                                              -71dBm 
1 New    | 94e6863b58de | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-220-22587    -71dBm 
1 New    | 94e6863bb1fa | iBeacon:e5ca1ade-f007-ba11-0000-000000000000-248-45371    -85dBm 
1 New    | c1e81a463583 | c1e81a463583                                              -67dBm 
1 Close  | 070b91270c54 | msft:cdp:0922                                             -33dBm
1 New    | 070b91270c54 | msft:cdp:0922                                             -33dBm 
1 New    | 60c1d87dd8f5 | 60c1d87dd8f5                                              -87dBm 
1 New    | 7ceb1853ea97 | 7ceb1853ea97                                              -85dBm 
1 New    | 49158bcb517b | 49158bcb517b                                              -85dBm
```

DTTerastar, 2023-11-06:
SSL is REALLY hard for these little things to do, so sometimes you need to retry.  It should work after a few tries.  If not you can use the firmware updating in ESPresense-Companion

N3rdix, 2023-11-06:
Indeed it succeeded after around 10 retries, thanks for the hint 👍