ESPresense/ESPresense #1091: ESP32 based presence detection in HA doesn't work after V3.2.4

itnassol, 2023-11-06:
### Describe the bug

I have a dozen or so of these around the house to say who is in what room, been working brilliantly for months, last week the office and the living room stopped reporting. after extensive checks I noticed that the 2 not reporting the living room and the office had updated to V3.3.1, I reloaded the office with V3.2.4 and it started working straight away. Any ideas or do we just need to make sure we are loading the older version for  ESP32 based presence detection?

### Version

3.3.1

closed 2023-11-06 by DTTerastar

itnassol, 2023-11-07:
Thank you for that, works perfectly.

fisch55, 2023-11-08:
Hi 
got a Problem after Updating to 3.3.1.

just make the new format in HA like this.
In ESP ist shows me, my devices correctly....

But in HA it shows the device as unknown....

platform: mqtt_room
device_id: "irk:1b0xxxxxxxxxxx34"
name: Iphone Tom
unique_id: 'Iphone Tom'
state_topic: "espresense/devices/irk:1b0xxxxxxxxxxx34"
timeout: 10
away_timeout: 120

Did I something wrong?
Thx for helping me...

itnassol, 2023-11-08:
Mine are working perfectly and I have this set up in the configuration.yaml

sensor:
  - platform: mqtt_room
    name: 'Where is Andrews Phone'
    device_id: "iBeacon:xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx_100_1"
    state_topic: 'espresense/rooms'
    timeout: 5
    away_timeout: 60

fisch55, 2023-11-08:
I will change it for  Version 4. you set roomtopics set on, right? 
With version 4 it will stop working….

itnassol, 2023-11-08:
I'm going be honest with you, I am very new to all of this, just thought if you had a setup that worked you might be able to copy some of it. Good luck.

fisch55, 2023-11-08:
Thx before 3.3.1 everything works fine …. But with 3.3.1 you have to change the format in yaml…. But for me it does not work.

itnassol, 2023-11-08:
Did you check this check box in the ESP setup?
[image]

fisch55, 2023-11-08:
Yes, input it off…. Because I use the new format in the yaml…

michaelblight, 2023-11-09:
> It will does not work for me.... I don't now, what's going wrong...

It initially wasn't working for me. I did a copy/paste of `espresense/devices/` and I think it may have had some weird characters in there. When I retyped it and restarted HA, all was well. Bit of a pain that I now have to have two sets of very similar looking strings in my secrets file.

86lava, 2023-11-09:
Checking the box restores the function, but it sounds like this fix won't work starting with v4. A better fix is to revise the state_topic in the sensor yaml as follows:  
`

    #### BEFORE ####
    - platform: mqtt_room
      device_id: "ipad" # the name used to ENROLL the iOS device in ESPresense
      name: "6th Gen iPad"
      state_topic: 'espresense/rooms'
      timeout: 10
      away_timeout: 120
   
    #### sensor yaml AFTER Send to rooms topic was deprecated ####
    - platform: mqtt_room
      device_id: "ipad" # the name used to ENROLL the iOS device in ESPresense
      name: "6th Gen iPad"
      state_topic: 'espresense/devices/ipad'
      timeout: 10
      away_timeout: 120`

itnassol, 2023-11-10:
I have to say I am a little confused as to what to do next, I know I am getting old, lol. But I have tried every combination.

As Follows...

With check box off,

I have tried ..

  - platform: mqtt_room
    name: 'Where is Andrew'
    device_id: "iBeacon:bbbxxxxx-100-1"
    state_topic: 'espresense/devices/iBeacon:bbbxxxxx-100-1'
    timeout: 5
    away_timeout: 60

Didn't work.

Tried

  - platform: mqtt_room
    name: 'Where is Andrew'
    device_id: "iBeacon:bbb1b2ec-26e6-465f-afb3-a0e1fc667b8e-100-1"
    state_topic: 'espresense/devices'
    timeout: 5
    away_timeout: 60

Still nothing.

I know it is being depreciated in v4, so would like to get things sorted now, but not sure of the new yaml format.

DTTerastar, 2023-11-10:
device_id: "iBeacon:bbbxxxxx-100-1"
state_topic: 'espresense/devices/iBeacon:bbbxxxxx-100-1' 

should be correct.  Try re-doing but copy and paste out of MQTT Explorer.  Also make sure you didn't get dreaded smart quotes: https://chat.openai.com/share/793a91e7-0441-4082-b2b2-3ed6d77ab8b2

itnassol, 2023-11-13:
Hi Darrell, Thanks for that, I am not sure where to fine the MQTT Explorer? But I am making those changes now and trying again.

itnassol, 2023-11-13:
Scrap that, found it installed and found the correct ibeacon copied it to the config.yaml.

itnassol, 2023-11-13:
OK, so it is now all working without the check box checked. I found the issue.

I was taking the iBeacon from the companion app, which was the correct number, iBeacon:aaaxxxbbbxx but it was the end of the number that was the problem, on the companion app it ends -100_1 (underscore 1) on MQTT Explorer it ends -100-1 (Dash 1) once I had corrected them all they all now work beautifully.

Thank you for all your help and guidance, maybe what I found can help someone else.