ESPresense/ESPresense #1092: Upgraded from 3.2.4 to 3.3.1 esp32 will no longer boot unless plugged into usb

bakerkj, 2023-11-06:
### Describe the bug

I have been running 3.2.4 on numerous esp32s successfully for months. I tried upgrading two of them to 3.3.1 (one through the espresense addon to home assistant, one directly through the espesense UI). 

Neither of them started up successfully after upgrade (no wifi; no dhcp, no mqtt). 

I then plugged them into my computer on USB and without any other changes they boot successfully (wifi, dhcp, and mqtt all successfully). They are running 3.3.1!

When I unplug them from my laptop again and return them to 5V power they no longer connect to wifi, dhcp, mqtt)

Please advise.

Thanks!

### Version

3.3.1

### Firmware flavor

Standard

### Device (be specific)

D1 Mini NodeMCU ESP32 ESP-WROOM-32

m0vva, 2023-11-09:
I have the same problem with all my ESP32 DEVKITV1 NodeMCU ESP-WROOM-32 boards - all new from Amazon and work fine when connected to my laptop but are mostly useable on regular 5V micro-USB wall-wart type SMPS.

bakerkj, 2023-11-11:
I purchased a few new boards they work fine with 3.3.1. I then upgraded some of my other NodeMCU boards and all but one worked fine.

closed 2023-11-11 by bakerkj