ESPresense/ESPresense #1101: Incorrect formula in the one_euro_filter class

gunnarbeutner, 2023-11-10:
This was independently found by @edgar-bonet in https://github.com/haimoz/SoftFilters/issues/1:

One of the paper's authors also has a repo where this is already fixed: https://github.com/casiez/OneEuroFilter/blob/main/CppUsingTemplates/1efilter.cc

I'm assuming the comment `03/08/23: fixed implementation` refers to this bug.

DTTerastar, 2023-11-12:
I just replaced SoftFilters with a custom implementation in v3.3.2b1 please test!

DTTerastar, 2023-11-12:
I always knew something was wrong w/ SoftFilters version...

gunnarbeutner, 2023-11-13:
I've had another look at the old code and I think there are three problems:

* The bug in the one_euro_filter class (i.e. `hatxprev` vs. `xprev`).
* I've just realized that the timestamps that were passed to the filter class are using the incorrect unit (milliseconds vs. seconds).
* The filter parameters were chosen to somewhat match the incorrect timestamp unit, but that only partially fixes/hides the problem:

I've gathered a couple of RSSI values and corresponding "true" distances (using UWB and ARKit with camera assistance on an iPhone, no filtering applied) - https://github.com/gunnarbeutner/validate-1euro:

**Scenario 1: Walking to the kitchen and back**

[image]

(The BLE distance measurements decrease before I actually walked back. This is probably because I turned around which caused a sudden drop in signal attenuation.)

As you can see the broken/old filter _does_ smooth the distance values... but it behaves quite strangely when I walk back to the node.

Also, good news: When I comment out the `removeSpike` call the filtered distances match the 1€ algorithm.

The new algorithm does add a bit of jitter, but seems to converge a lot quicker. I'll do some more testing to see how this affects the companion app's location data.

**Scenario 2: Stationary device (nearby)**

[image]

**Scenario 3: Stationary device (further away)**

[image]

**Scenario 4: Rotating while holding the phone**

[image]

DTTerastar, 2023-11-13:
I also am trying a Kalman filter in the kalman branch.  Not sure if it's better or not.  I like how quickly 1ueuro can catch up.  I'm thinking we need a 3d kalman in companion to fix the floor switching

gunnarbeutner, 2023-11-13:
Shouldn't those constructor params be the other way around?

```c++
FilteredDistance(25, 0.1)
```

With the params the way they were the filter output is virtually identical to the input data:

[image]

Swapping the parameters results in a more reasonable curve. Like you said though - 1 euro converges quite a bit faster:

```c++
FilteredDistance(0.1, 25)
```

[image]

DTTerastar, 2023-11-13:
I only spent a little bit of time testing it.  I decided going out the door w/ 1euro made more sense since it minimized change.  I used the same class signature so I can make it configurable in the future.  You can send those graphs to chat gpt and it'll suggest new parameters to try...

DTTerastar, 2023-11-13:
Oh, and I was seeing some strange incidents where the distance went out of control...

gunnarbeutner, 2023-11-14:
Oh yeah, I noticed that as well. Try this:

```patch
     // Update covariance
-    covariance[0][0] += deltaTime * (covariance[1][0] + covariance[0][1]) + processNoise;
-    covariance[0][1] += deltaTime * covariance[1][1];
-    covariance[1][0] += deltaTime * covariance[1][1];
+    covariance[0][0] += deltaTime * (deltaTime * covariance[1][1] + covariance[1][0]) + processNoise;
+    covariance[0][1] += deltaTime * covariance[1][1] + processNoise;
+    covariance[1][0] += deltaTime * covariance[1][1] + processNoise;
+    covariance[1][1] += processNoise;
```

That _should_ match the formula more closely:

[image]

The problem you described still happens for very small `processNoise` values (e.g. 0.001) - which shouldn't really be an issue in practice. I haven't looked into that but I _guess_ it's probably due to floating point rounding errors.

DTTerastar, 2023-11-18:
In my testing slowly updating tiles work good on kalman, but my apple watch works much worse on it.  I do want to try adding back in the spike detection still.

gunnarbeutner, 2023-11-30:
I've spent the past couple of days reading research papers after denoising algorithms. I ended up half-way through into an implementation of a denoising auto-encoder when I think I realized something. All those papers - like our current filter implement - seem to ignore one or more fundamental physical properties of the signal we're trying to filter. By staring at RSSI graphs for way too long I've come up with this:

1. I believe we have to take into account the logarithmic nature of the RSSI values before trying to apply further filtering. For example the Kalman filter tries to model linear movement but we're feeding it with logarithmic dBm values. Sure, they're converted to meters, but e.g. 20 dBm "noise" has a different meaning depending on the underlying true RSSI value, i.e. the filter observes significantly different levels of noise depending on how far we're from the node. We need to get rid of that noise first.

2. Assuming stationary transmitters and receivers and given example measurements of -70, -65, -89 and -75 surely the value that's the best approximation is -65 (i.e. the maximum). The filter implementations I've seen so far try to average the values. Why is it -65? Well:

    a) The received signal strength can only ever be worse than the true signal strength. Attenuation and multipathing doesn't improve the signal strength beyond the true signal strength.

    b) Constructive interference from other radio sources can be ignored because it'd likely corrupt the signal beyond recognition, i.e. we wouldn't even get a measurement from the BLE stack.

[image]

So, to filter measurements for a stationary sender we just `max(measurements)` for an appropriate time window. Tell me I'm wrong or I'm missing something here... but I don't think I'm wrong on this:

[image]

3. We can model movement of the sender using the following algorithm:

    a) Calculate the mean of the time window - ignoring the last N measurements. This is our predicted distance "a while ago".

    b) If the latest measurement is greater than the mean (plus maybe the standard deviation times a tunable factor?) we _might_ be moving. Let's save that information and change to "movement" mode for a couple of seconds if the last M measurements indicated potential movement. This filters out transient spikes quite nicely.

    c) If we're in movement mode return `max(smaller_window)`, otherwise we return `max(window)`.

The sizes of the two windows can be tuned to improve convergence time when we're moving _away_ from the sensor. When we're moving towards the sensor the predicted distance converges instantly.

[image]

Surely I must be missing something here. It can't be that simple, can it?

You can find the filter I've built at https://github.com/gunnarbeutner/ESPresense/commit/9967b8bab3ed6cc5f1787ce780ee4d38234e8cc1 and I think it has four interesting properties:

1. When the sender is moving towards the node it converges instantly.

2. The predicted distances are closer to what they really are, at least as far as the RSSI values are concerned. The previous filters vastly overestimated the true distance. This might require re-tuning reference RSSI values/absorption, etc.

3. When the sender is moving away from the node it converges _almost_ as quickly as the 1€ filter. To counteract this I've thought about biasing the companion app by increasing the weight for nodes which indicate that the sender is approaching them.

4. There is virtually no noise when the sender is stationary.

Also, in a somewhat unrelated note: I'm currently testing a different weighting algorithm for the companion app. The discrete weights based on the node order don't really work quite well when you have lots of nodes with similar distance measurements. So I've just plugged the gaussian PDF for the filtered distances right into the optimization function:

```patch
diff --git a/Locators/NelderMeadMultilateralizer.cs b/Locators/NelderMeadMultilateralizer.cs
index aa19f76..6d58c40 100644
--- a/Locators/NelderMeadMultilateralizer.cs
+++ b/Locators/NelderMeadMultilateralizer.cs
@@ -58,6 +58,9 @@ public class NelderMeadMultilateralizer : ILocate
             {
                 var lowerBound = Vector<double>.Build.DenseOfArray(new[] { _floor.Bounds[0].X, _floor.Bounds[0].Y, _floor.Bounds[0].Z, 0.5 });
                 var upperBound = Vector<double>.Build.DenseOfArray(new[] { _floor.Bounds[1].X, _floor.Bounds[1].Y, _floor.Bounds[1].Z, 1.5 });
+                var minDistance = nodes.Min(a => a.Distance!);
+                var maxDistance = nodes.Max(a => a.Distance!);
+                var _sigma = 0.02;
                 var obj = ObjectiveFunction.Value(
                     x =>
                     {
@@ -66,7 +69,7 @@ public class NelderMeadMultilateralizer : ILocate
                             .PointwiseMaximum(0)
                             .L2Norm();
                         return (distanceFromBoundingBox > 0 ? Math.Pow(5, 1 + distanceFromBoundingBox) : 0) + Math.Pow(5 * (1 - x[3]), 2) + nodes
-                            .Select((dn, i) => new { err = Error(x, dn), weight = _state?.Weighting?.Get(i, nodes.Length) ?? 1.0 })
+                            .Select((dn, i) => new { err = Error(x, dn), weight = 1d / Math.Sqrt(_sigma * 2d * Math.PI) * Math.Exp(-(Math.Pow((dn.Distance - minDistance) / (maxDistance - minDistance), 2d) / (2d * Math.Pow(_sigma, 2d)))) })
                             .Average(a => a.weight * Math.Pow(a.err, 2));
                     });
```

gunnarbeutner, 2023-11-30:
Also, I wonder how well the RSSI dispersion pattern is correlated with specific points in space. Maybe we can use that fingerprint to distinguish the angle of the transmitter towards the node:

```mermaid
graph TD
subgraph Z[" "]
direction LR
nodeA --> nodeB --> WALL --> transmitter --> nodeC
end
```

Surely the fingerprint observed by nodeB would be different when the transmitter is between nodeA and B as compared with when it's between nodeB and nodeC.

DTTerastar, 2023-11-30:
I have attempted the min filter as well.  But I honestly don't fully remember the results.  The fact it's fast to notice you approaching is a VERY nice property.  If you look at the latest beta I'm adding a variance value to the reporting.  It can be used to figure out how trustworthy the value really is.  I'm working on a Maximum Likelihood Estimation locator that will take into account the variance to "trust" low variance results more.  While doing that I notice my node beaconing it REALLY inconsistent w/ way too high a variance.  I think this is part of the error we are chasing since optimization is doing everything based on shotty data.

DTTerastar, 2023-11-30:
I've given you write access so should be able to PR in directly.  That way your PRs will get built and be an option to install via the github artifacts.

gunnarbeutner, 2023-12-01:
Awesome, thank you! About the variance: That sounds useful - I'll add that to my filter and play around a bit with it a bit. This might be even more useful when the underlying RSSI values are more stable, i.e. as an indicator whether the devices are moving. Or maybe... we previously had a speed value which wasn't terribly useful because it was affected by the high variance, but maybe this could work: remember the filtered distance from 5 seconds ago and calculate the speed based on that.

The high variance in your case is probably because the old filter is bouncing between the two or more RSSI "bands" which I'm seeing for most of my measurements (e.g. around -88 and -97 in this example):

[image]

I think this is because the BLE stack is sometimes reporting the signal strength for beacons received directly and sometimes for reflections (which have lost a bit more signal strength due to attenuation).

Also, it's kind of obvious in hindsight, but I think it's _possible_ to use node pairs to "scan" for people even if they're not carrying any devices. The RSSI values are for two of my nodes and the line between them transects a few rooms. The red rectangle matches perfectly with my motion sensor.

DTTerastar, 2023-12-01:
BLE broadcasts using 3 different frequencies so if you're seeing 2 or 3 bands it's probably the effect of the different frequencies.  I've tried unsuccessfully to get the advertised freq so we could setup 3 separate filters.  I still think we could get it if we used a lower level lib instead of NimBLE.

closed 2024-06-15 by stale[bot]