ESPresense/ESPresense #1113: Fingerprints should show all IDs, not just included ones

rwjack, 2023-11-11:
### Is your feature request related to a problem? Please describe.

https://github.com/ESPresense/ESPresense/issues/1064

Referenced above, if fingerprints are for debugging purposes, wouldn't it be clear to show all IDs there, instead of just the included ones?

DTTerastar, 2023-11-12:
[v3.3.1](https://github.com/ESPresense/ESPresense/releases/tag/v3.3.1) just added just this!

closed 2023-11-12 by DTTerastar

rwjack, 2023-11-13:
@DTTerastar I am running v3.3.1. 

When I go to fingerprints all I see is the single beacon that is included on the Settings>Filtering page.

DTTerastar, 2023-11-13:
[image]

is the default, change it to blank (no filter)

[image]

rwjack, 2023-11-14:
Ohh, I see now. That seemed invisible to me.

I'm not an UX expert, but perhaps those tabs should be rotated?

Show the table headers first, then the data?

DTTerastar, 2023-11-14:
Svelte table works like that... I do agree it's not the most intuitive....